ros-visualization/rviz #834: rviz does not save which tf frames were selected for viewing

jmtatsch, 2015-01-07:
also, when resetting the clock it forgets the current selection as well.

wjwwood, 2015-01-07:
@jmtatsch can you give more details, like a step by step example on how to reproduce the problem, what exactly you mean by selected (boxes checked versus the selection tool), and pictures or a short video would be helpful in demonstrating the issue.

Also, what OS and ROS version are you using?

What is the exact version of rviz that you are using?

Without more information, I would not expect anyone could help you with this issue.

jmtatsch, 2015-01-08:
OS: OS X 10.10.1
ROS indigo
rviz version 1.11.3 compiled against OGRE version 1.9.0
[image]

wjwwood, 2015-01-28:
Thanks for the updated information, I'll keep it here on the list and we'll get to it when we can.

ipa-lth, 2015-04-21:
I just recognized the same issue.
The relevant information (which frame is checked) is stored in the .rviz file, however. It seems like the tf plugin is just not parsing (or using) anything after "All enabled".

If you want to reproduce the problem:
1. Uncheck some frames in the tf plugin
2. Save you configuration
3. Load the previous saved configuration again. (All frames will be checked again.) 

System:
Ubuntu 14.04
ROS Indigo
RViz version 1.11.7 (indigo). Compiled against OGRE version 1.8.1 (Byatis).

zweistein, 2015-10-13:
Same issue on: Ubuntu 14.04; ROS Indigo, Rviz 1.11.8 and OGRE 1.8.1

wjwwood, 2015-12-10:
Closing with https://github.com/ros-visualization/rviz/pull/946, please comment if this is still and issue.

closed 2015-12-10 by wjwwood