ros-visualization/rviz #836: Ambient color is rendered much too light colored for many COLLADA files

skohlbr, 2015-01-14:
While making a URDF model I just experienced this bug again. Essentially, when a COLLADA file has zeros in it's ambient color tag, for some reason rviz renders it much too light colored. Doing a search of replace and replacing

```xml
  <color sid="ambient">0 0 0 1</color>
```
with
```xml
  <color sid="ambient">0.1 0.1 0.1 1</color>
```
in the .dae file "fixes" the problem.

This problem has been around for years. I think it had been fixed 3 years ago but quickly came back for some reason. Some history in https://code.ros.org/trac/ros-pkg/ticket/5237. Screenshots posted there by @wjwwood show exactly the symptoms I'm seeing now on rviz from .debs/ROS Indigo/AMD64. Don't have time to fix myself currently, but wanted to have it tracked at least.

Also described in http://answers.ros.org/question/27734/mesh-rendering-in-rviz/

wjwwood, 2015-01-16:
More info, if this is the same issue, then it is likely a patch would be for the `collada_parser` package which is in the `robot_model` repository: https://github.com/ros/robot_model/tree/indigo-devel/collada_parser

wjwwood, 2015-01-17:
Actually, this has to be a different code path... Something must of changed since the original ticket, since rviz doesn't even use `collada_parser`, which is what they originally patched.

wjwwood, 2015-01-17:
I've found it. This is the problem, ambient defaults to 0.5, 0.5, 0.5, 1.0 if the mesh defines them as 0, 0, 0 (r, g, b):

https://github.com/ros-visualization/rviz/blob/8a010bbda36d425129dc06887eddcab032459319/src/rviz/mesh_loader.cpp#L455

https://github.com/ros-visualization/rviz/blob/8a010bbda36d425129dc06887eddcab032459319/src/rviz/mesh_loader.cpp#L483-490

After changing the default to 0, 0, 0, 1 and removing the `if (clr.r > 0 && clr.g > 0 && clr.b > 0)` condition I get more sane results. The old pull request to `collada_parser` that you linked to and I commented on 3 years ago they set it to 0.0001, 0.0001, 0.0001 for some reason, but it seems to me like 0, 0, 0 works totally fine.

I'll post some pictures and open a pull request later tonight.

skohlbr, 2015-01-17:
Thanks for looking into this and finding the issue :)

wjwwood, 2015-01-19:
I've open a pull request: https://github.com/ros-visualization/rviz/pull/837

closed 2015-02-06 by wjwwood