ros-visualization/rviz #839: Linking error Undefined symbols for architecture x86_64:  "rviz::MapDisplay::mapUpdated()

jmtatsch, 2015-01-23:
When trying to compile rviz with OS X Yosemite and ROS Indigo I run into linking issues:

````cd /Users/tatsch/ros_catkin_ws/build_isolated/rviz && /opt/ros/indigo/env.sh make -j4 -l4                   tatsch@JRetinaMacbookPro
[ 59%] Built target rviz
[ 59%] Compiling generated code for rviz_sip Python bindings...
[ 59%] Built target executable
[ 60%] Built target rviz_image_view
[ 61%] Meta target for rviz_sip Python bindings...
[ 61%] Built target librviz_sip
Linking CXX shared library /Users/tatsch/ros_catkin_ws/devel_isolated/rviz/lib/libdefault_plugin.dylib
Undefined symbols for architecture x86_64:
  "rviz::MapDisplay::mapUpdated()", referenced from:
      rviz::MapDisplay::incomingMap(boost::shared_ptr<nav_msgs::OccupancyGrid_<std::__1::allocator<void> > const> const&) in map_display.cpp.o
      rviz::MapDisplay::incomingUpdate(boost::shared_ptr<map_msgs::OccupancyGridUpdate_<std::__1::allocator<void> > const> const&) in map_display.cpp.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [/Users/tatsch/ros_catkin_ws/devel_isolated/rviz/lib/libdefault_plugin.dylib] Error 1
make[1]: *** [src/rviz/default_plugin/CMakeFiles/default_plugin.dir/all] Error 2
make: *** [all] Error 2````

Curiously in map_display.cpp there never has been a mapUpdated(). Any hints?

wjwwood, 2015-01-28:
Sorry, I haven't seen this before, I'd have to try and reproduce it on my machine. My first suggestion would be to make sure you're building against the latest version of the dependencies.

jmtatsch, 2015-01-28:
Sure, I always have the latest dependencies. In case that this is somehow related with sip, I am using sip 4.16.5 from brew.

wjwwood, 2015-01-31:
@jmtatsch Sorry, I was unable to reproduce this on my Mac which is 10.9.5, but I can try again when I upgrade (it's on my todo list). I have some theories, but I'll need to upgrade before I can follow up on them.