ros-visualization/rviz #840: Update Ogre Point Cloud even possible?

gabyx, 2015-01-26:
Hello,

there is an addPoints function in the point_cloud.h  file, is there an update function too, because
the addPoints copies the points internally and then copies them into the vertex buffer.

Is this Ogre Point Cloud even updatable?
Thanks for an answer

gabyx, 2015-01-26:
Could we add this to the update function:
```C++
void OgrePointCloud::_updateRenderQueue(Ogre::RenderQueue* queue)
{

  Ogre::AxisAlignedBox aabb;
  bounding_box_.setNull();

  uint32_t current_point = 0;
  uint32_t vpp = getVerticesPerPoint();

  V_OgrePointCloudRenderable::iterator it = renderables_.begin();
  V_OgrePointCloudRenderable::iterator end = renderables_.end();
  for (; it != end; ++it)
  {
    // update points in vertex buffer
    vbuf = it->getBuffer();
    op = it->getRenderOperation();
    aabb.setNull();

    // Lock buffer
    vdata = vbuf->lock(Ogre::HardwareBuffer::HBL_NO_OVERWRITE);
    fptr = (float*)((uint8_t*)vdata);
    for(uint32_t i = 0; i < op->vertexData->vertexCount; ++i, ++current_point){

        const Point& p = points[current_point];

        uint32_t color;

        if (color_by_index_)
        {
          // convert to ColourValue, so we can then convert to the rendersystem-specific color type
          color = (current_point + point_count_ + 1);
          Ogre::ColourValue c;
          c.a = 1.0f;
          c.r = ((color >> 16) & 0xff) / 255.0f;
          c.g = ((color >> 8) & 0xff) / 255.0f;
          c.b = (color & 0xff) / 255.0f;
          root->convertColourValue(c, &color);
        }
        else
        {
          root->convertColourValue( p.color, &color );
        }

        aabb.merge(p.position);

        float x = p.position.x;
        float y = p.position.y;
        float z = p.position.z;

        for (uint32_t j = 0; j < vpp; ++j)
        {
          *fptr++ = x;
          *fptr++ = y;
          *fptr++ = z;

          if (!current_mode_supports_geometry_shader_)
          {
            // skip static vertices
              fptr += 3;
//            *fptr++ = vertices[(j*3)];
//            *fptr++ = vertices[(j*3) + 1];
//            *fptr++ = vertices[(j*3) + 2];
          }

          uint32_t* iptr = (uint32_t*)fptr;
          *iptr = color;
          ++fptr;

          ASSERTMSG((uint8_t*)fptr <= (uint8_t*)vdata + rend->getBuffer()->getNumVertices() * vertex_size,"");
        }
    }
    vbuf->unlock();

    // set new AABB
    it->setBoundingBox(aabb);
    bounding_box_.merge(aabb);

    // add to queue
    queue->addRenderable((*it).get());
  }
}

```

wjwwood, 2015-01-26:
@gabyx I'm not sure. I neither wrote this part of rviz nor have I used it recently. I'll have to dig into what you're talking about to understand it better.

It would be helpful if you provided a diff or a pull request with your suggestion so we could more easily discuss it.

gabyx, 2015-01-26:
Hi :-) jeah, unfortunately have little time for a pull request ( I do only use the point cloud ogre implementation from your library) :-)

But the only functions I changed is the update function, which now copies the points from the internal buffer to the GPU for each frame! (the points can be changed by the function)

```C++
OgrePointCloud::Point & OgrePointCloud::getPoint( uint32_t idx){
        return points_[idx];
}
```

**Update FUnction**
```C++
void OgrePointCloud::_updateRenderQueue(Ogre::RenderQueue* queue)
{
  Ogre::Root* root;
  if(!color_by_index_){
    root = Ogre::Root::getSingletonPtr();
  }
  Ogre::AxisAlignedBox aabb;
  bounding_box_.setNull();

  Ogre::HardwareVertexBufferSharedPtr vbuf;
  void* vdata = 0;
  Ogre::RenderOperation* op = 0;
  float* fptr = 0;
  uint32_t current_point = 0;
  uint32_t vpp = getVerticesPerPoint();

  V_OgrePointCloudRenderable::iterator it = renderables_.begin();
  V_OgrePointCloudRenderable::iterator end = renderables_.end();
  for (; it != end; ++it)
  {
    // update points in vertex buffer
    vbuf = (*it)->getBuffer();
    op = (*it)->getRenderOperation();
    aabb.setNull();

    // Lock buffer
    vdata = vbuf->lock(Ogre::HardwareBuffer::HBL_NO_OVERWRITE);
    fptr = (float*)((uint8_t*)vdata);
    uint32_t size = points_.size();
    for(uint32_t current_point = 0; current_point < size; ++current_point){

        const Point& p = points_[current_point];

        aabb.merge(p.position);

        float x = p.position.x;
        float y = p.position.y;
        float z = p.position.z;

        for (uint32_t j = 0; j < vpp; ++j)
        {
          *fptr++ = x;
          *fptr++ = y;
          *fptr++ = z;

          if (!current_mode_supports_geometry_shader_)
          {
            // skip static vertices
              fptr += 3;
//            *fptr++ = vertices[(j*3)];
//            *fptr++ = vertices[(j*3) + 1];
//            *fptr++ = vertices[(j*3) + 2];
          }

            // if color by index, do not write buffer, otherwise write color
            if (!color_by_index_)
            {
              uint32_t color;
              root->convertColourValue( p.color, &color );
              uint32_t* iptr = (uint32_t*)fptr;
              *iptr = color;
            }
            ++fptr;
        }

    }
    vbuf->unlock();

    // set new AABB
    (*it)->setBoundingBox(aabb);
    bounding_box_.merge(aabb);

    // add to queue
    queue->addRenderable((*it).get());
  }
}
```

wjwwood, 2015-01-26:
I also have limited time, we can leave this here for now, but I won't have time to investigate it further.

Thanks for letting us know.