ros-visualization/rviz #843: Multiple visuals on a link get the first material defined in the urdf for all subsequent geometries.

freyzor, 2015-02-08:
When defining a urdf link with multiple visual elements with different materials only the first material gets applied to all visual elements. 

By viewing the code robot_link.cpp we can see how this happens as the color gets cached in a default value and never checked again while the default is set.

```c++
void RobotLink::createEntityForGeometryElement(const urdf::LinkConstPtr& link, const urdf::Geometry& geom, const urdf::Pose& origin, Ogre::SceneNode* scene_node, Ogre::Entity*& entity)
{
    // **** SNIP code removed for clarity *****
    if (default_material_name_.empty())
    {
      default_material_ = getMaterialForLink(link);

      static int count = 0;
      std::stringstream ss;
      ss << default_material_->getName() << count++ << "Robot";
      std::string cloned_name = ss.str();

      default_material_ = default_material_->clone(cloned_name);
      default_material_name_ = default_material_->getName();
    }
    // **** SNIP rest of code removed for clarity *****
```

The method should always try to find a material for every element processed and use a default of none is found.

Also calling the urdf element processed 'link' is very confusing the methods is actually processing a <visual> or a <collision> element both belonging to a parent <link>.  Seems like a relic from a older and flatter urdf structure.

wjwwood, 2015-02-10:
For posterity: https://github.com/ros-visualization/rviz/blob/09bc0e0b0526c984199e6734221bbdeeaead95a3/src/rviz/robot/robot_link.cpp#L610-L621

wjwwood, 2015-02-10:
@freyzor Thanks for the report, I believe I am following what you are saying, but it isn't immediately obvious to me how to fix it. It sounds like you have a pretty good handle on what should happen, I don't suppose you'd like to try your hand at making a pull request to address it?

Unfortunately I'll likely not have time to address this myself in the short term. I've just run out of cycles to spend on rviz for the moment. For now I'll mark it as untargeted.

freyzor, 2015-02-10:
I would have taken a stab at it if I had a proper environment setup and am pretty new to ROS.  I have been working through a virtualbox and software render at abyssmal framerates :P   I caved and have now done a proper install. I will take a look if still unresolved when I have got my ros mojo going properly.