ros-visualization/rviz #844: RViZ selection tool - points not in FixedFrame

ZdenekM, 2015-02-12:
I'm using RViZ selection tool to get some points out of Kinect cloud. I thought that those point coordinates will be transformed into current fixed frame (everything in scene should be transformed into fixed frame, right?). But apparently they are in Kinect frame. Is this correct? I believe it might be bug. I'm using ROS Hydro, RViZ from debs.

wjwwood, 2015-02-13:
@ZdenekM I'm not sure what the original intention was, whether this is expected behavior or a bug.

Maybe @hersh, @acornacorn, or @dgossow can say what the original intention was.

Regardless of that, it would probably be a good feature to let the user select in which frame the coordinates are.

hersh, 2015-02-13:
The point-selection feature was written before @hersh or @acornacorn or @dgossow worked on rviz.  It was from Josh Faust.

My understanding of it was simply that the selection properties from the point cloud let you see some details of the points you selected.  As far as I know, he just made it report the raw point cloud data.

It does seem like a nice feature to add, to make it selectable between original coordinates and rviz fixed-frame coordinates.

ZdenekM, 2015-02-16:
Ok, thanks for info. Ability to choose points in fixed-frame coordinates will be especially useful when selecting points from multiple sources. With current behaviour you can get selected points but you can't know which point is from which cloud and which frame.

senselessDev, 2019-03-11:
I am working on an own version of a [point cloud selection publisher](https://github.com/tu-rbo/turbo-ros-pkg.git) and do not want to subscribe to every pcl message topic. Instead, I am trying to recreate the point cloud only with the information from the `rviz::PropertyTreeModel`. 
Unfortunately, I don't know in which frame the selected point coordinates are given.
Is there a possibility to gain this information somehow?

Besides, wouldn't it make sense to at least state in the GUI which frame the coordinates are in?
For example, add the frame_id here:
https://github.com/ros-visualization/rviz/blob/64f474baa40f650289a2a13e1b52d5b6b26243b9/src/rviz/default_plugin/point_cloud_common.cpp#L171-L172