ros-visualization/rviz #850: Crash of STL based robot model on IFC6410

mendimik, 2015-02-22:
Using motoman_mh5_support package, rviz crashes on attempt of adding the robot model view, with bus error.

strace  https://gist.github.com/tfoote/8c7c925d2868e26babf3
gdb backtrace  https://gist.github.com/tfoote/4a9db2b2d7480a08b27d

Any STL file brings to same result

wjwwood, 2015-02-23:
For people who may not know, the IFC6410 is a quad-core ARM single board computer: http://www.inforcelive.com/index.php?route=product/product&product_id=53

This might have something to do with ARM as I cannot reproduce this on a normal amd64 desktop machine.

@ahendrix wasn't there a patch to assimp for ARM? Maybe that could be the source of the difference?

rkojcev, 2015-02-24:
I also have the same issue on Jetson TK1 dev board. However when loading robot models constructed with Collada files (*.dae) rviz runs fine....

trainman419, 2015-02-25:
My best guess is that this crash is related to this part of the stacktrace:

```
Program received signal SIGBUS, Bus error.
0x40327984 in ogre_tools::STLLoader::load(unsigned char*) () from /opt/ros/indigo/lib/librviz.so
(gdb) bt full
#0  0x40327984 in ogre_tools::STLLoader::load(unsigned char*) () from /opt/ros/indigo/lib/librviz.so
No symbol table info available.
#1  0xbef0bf68 in ?? ()
```

Which suggests that this is a bug in OGRE's STL loader on arm.

Two things to try which can help confirm this:

 * Try to load a different robot model which contains STL meshes, to confirm that this affects most or all STL files.
 * Try to use the ORGE STL loading functions by themselves, outside of rviz and ROS. This will likely mean finding an OGRE sample that loads meshes, or writing your own.

tfoote, 2015-02-26:
Note ogre_tools are actually inside the rviz sources: http://docs.ros.org/indigo/api/rviz/html/c++/stl__loader_8cpp_source.html

wjwwood, 2015-03-02:
The segfault is happening in rviz's code:

https://github.com/ros-visualization/rviz/blob/indigo-devel/src/rviz/ogre_helpers/stl_loader.cpp

Though those load functions are a bit scary, I don't see anything that jumps out at me. Without a debug build of rviz and a backtrace from that, I can't narrow it down to a particular line of code so I think there is little chance I can do anything about this.

tfoote, 2015-03-18:
Here's a minimal package which can reproduce the problem: https://www.dropbox.com/s/p3hcpeko4w3euc1/stlcrash.tgz?dl=0

jbohren, 2015-04-03:
Rviz only reads binary STL files, which the mh5 models are, and the stl loader in rviz is pretty hacky.

You guys are having issues on the IFC6410 and Jetson TK1, but not on x86 platforms.

Is this just endianness problem?

mehtajaghvi, 2015-05-23:
I am having issue with Rviz on Jetson-TK1 running on ubuntu 14.04 , ros indigo.
I enhanced the cpu performance to maximun but still the same issue. It did run once on Jetson but I cannot run it again. I have also set the GTK_IM_MODULE and tried deleting the ~/.rviz but nothing works. Can anyone suggest anything please?

Abbyls, 2016-10-16:
@mehtajaghvi
Hi mehtajaghvi,
i met the same problem as you, but i used the tx1 instead, have you solved it? Can you give me some advice as i confused some days!

shinichiba, 2021-12-09:
I met same problem on Rpi3 install Xubuntu 16.04 . When use mesh to load stl file in rviz , system dump and exit.