ros-visualization/rviz #853: Collada (.dae) meshes with unit != "meter" are broken

andreasBihlmaier, 2015-02-25:
If a URDF contains a collada (.dae) mesh which has <unit> with non meter units (e.g. centimeter):

    --- collada_unit_meter.dae      2015-02-16 11:14:28.128423513 +0100
    +++ collada_unit_centimeter.dae 2015-02-16 11:16:06.542388857 +0100
    @@ -7,7 +7,7 @@
         </contributor>
         <created>2015-02-16T11:14:28</created>
         <modified>2015-02-16T11:14:28</modified>
    -    <unit name="meter" meter="1"/>
    +    <unit name="centimeter" meter="0.01"/>
         <up_axis>Z_UP</up_axis>
       </asset>
       <library_images/>
    </COLLADA>

no mesh at all is displayed in rviz RobotModel. It seems the mesh is also ignored for MoveIt! collision checking.

Test files can be created by simply exporting the default cube as .dae in Blender. Once without modification and once with modified . These files can also be found here: https://gist.github.com/andreasBihlmaier/576b5bc3837c568a6730

Issue is already confirmed by others:
http://answers.ros.org/question/203188/collada-dae-meshes-with-unit-meter-broken/

NikolausDemmel, 2015-02-25:
Here is another set of test meshes that seemed to recently (~ a year ago) work fine (https://github.com/ros-visualization/rviz/issues/751), but currently do not work for me on Ubuntu 14.04.2: https://github.com/garaemon/rviz_collada_marker

wjwwood, 2015-02-25:
Thanks for the report, we'll try to have a look into it as soon as possible.

chapulina, 2015-04-02:
We tried to reproduce the issue and it turned out that the meshes in centimeters were just coming up too small to be seen. You can check [this](https://github.com/wjwwood/rviz_issue_853) example repository, if you `roslaunch test.launch`, you'll see 4 unit cubes: white is in meters, red in decimeters, green in centimeters and blue in inches. If you can't see the centimeters one, just scroll into the origin.

Tested on Ubuntu 14.04.1, RViz 1.11.7 (indigo).

This was tested using markers, we haven't tested URDFs yet.

chapulina, 2015-04-02:
Just checked that it also worked with URDF on Indigo. Here's a screenshot: 

[image]

Can you please try out our repository and try to reproduce your issue with it? (make sure you've zoomed in, because the box is very small).

andreasBihlmaier, 2015-04-07:
@chapulina thank you for providing the test cases.
Unfortunately, the bug persists for me:
[image]
[image]

Tested on two hosts with very different GPUs, both running:

    > uname -a
    Linux myhost 3.16.0-33-generic #44~14.04.1-Ubuntu SMP Fri Mar 13 10:33:29 UTC 2015 x86_64 x86_64 x86_64 GNU/Linux

    > apt-cache showpkg ros-indigo-rviz
    Package: ros-indigo-rviz
    Versions: 
    1.11.7-0trusty-20150307-2257-+0000 (/var/lib/apt/lists/packages.ros.org_ros_ubuntu_dists_trusty_main_binary-amd64_Packages) (/var/lib/dpkg/status)
     Description Language: 
                      File: /var/lib/apt/lists/packages.ros.org_ros_ubuntu_dists_trusty_main_binary-amd64_Packages
                      MD5: dc362b1b78b867ed720836e8e80709fe

andreasBihlmaier, 2021-04-02:
Just found this old issue and did some tests with latest melodic rviz (1.13.16-1bionic.20210304.161522). Unfortunately the bug persists.
In the collada mesh
`<unit name="foo" meter="X"/>`
works as expected for all values of X>=1, but nothing shows up for all values X<1.0.

rhaschke, 2021-04-02:
Sorry, but I cannot reproduce the issue. The ros package mentioned in https://github.com/ros-visualization/rviz/issues/853#issuecomment-89043684 works for both, Melodic and Noetic. Are you sure you are using the default libassimp (4.1.0 for Melodic, 5.0.1 for Noetic)? Meshes are all loaded by libassimp. I also verified that this issue isn't related to the language settings.

andreasBihlmaier, 2021-04-02:
This is very strange:
[image]

The only thing showing up is the "meter_marker" 
[image]

I'm currently updating my systems at home. Will retry on another system and also this one with focal+noetic tomorrow.

andreasBihlmaier, 2021-04-02:
libassimp seems to be system default:
```
desknew ~/catkin_ws/src/rviz_issue_853> dpkg -S /usr/lib/x86_64-linux-gnu/libassimp.so.4.1.0
libassimp4:amd64: /usr/lib/x86_64-linux-gnu/libassimp.so.4.1.0
desknew ~/catkin_ws/src/rviz_issue_853> apt-cache showpkg libassimp4
Package: libassimp4
Versions: 
4.1.0~dfsg-3 (/var/lib/apt/lists/archive.ubuntu.com_ubuntu_dists_bionic_universe_binary-amd64_Packages) (/var/lib/dpkg/status)
 Description Language: 
                 File: /var/lib/apt/lists/archive.ubuntu.com_ubuntu_dists_bionic_universe_binary-amd64_Packages
                  MD5: 19a3b9c95112bea7e3f3134dbd0cf8eb
 Description Language: 
                 File: /var/lib/apt/lists/archive.ubuntu.com_ubuntu_dists_bionic_universe_binary-i386_Packages
                  MD5: 19a3b9c95112bea7e3f3134dbd0cf8eb
 Description Language: en
                 File: /var/lib/apt/lists/archive.ubuntu.com_ubuntu_dists_bionic_universe_i18n_Translation-en
                  MD5: 19a3b9c95112bea7e3f3134dbd0cf8eb

Reverse Depends: 
  libassimp-dev,libassimp4 4.1.0~dfsg+1~
  ros-eloquent-rviz-rendering,libassimp4
  ros-dashing-rviz-rendering,libassimp4
  ros-crystal-rviz-rendering,libassimp4
  ros-crystal-mrpt2,libassimp4
  ros-bouncy-rviz-rendering,libassimp4
  ros-melodic-rviz,libassimp4
  ros-melodic-robot-self-filter,libassimp4
  ros-melodic-pr2-navigation-self-filter,libassimp4
  ros-melodic-mrpt2,libassimp4
  ros-melodic-mrpt1,libassimp4
  ros-melodic-hpp-fcl,libassimp4
  ros-melodic-geometric-shapes,libassimp4
  ros-melodic-euscollada,libassimp4
  ros-melodic-collada-urdf,libassimp4
  ros-melodic-cob-obstacle-distance,libassimp4
  libassimp4:i386,libassimp4 4.1.0~dfsg-3
  libassimp4:i386,libassimp4 4.1.0~dfsg-3
  python3-pyassimp,libassimp4 4.1.0~dfsg+1~
  python3-pyassimp,libassimp4 4.1.0~dfsg-3
  python-pyassimp,libassimp4 4.1.0~dfsg+1~
  python-pyassimp,libassimp4 4.1.0~dfsg-3
  librviz2d,libassimp4
  libmrpt-opengl1.5,libassimp4
  libkido0.1,libassimp4
  libkido-utils0.1,libassimp4
  libkido-gui-osg0.1,libassimp4
  libgeometric-shapes2d,libassimp4
  libcollada-urdf0d,libassimp4
  libassimp-dev,libassimp4 4.1.0~dfsg-3
  assimp-utils,libassimp4
  doomsday-common,libassimp4
  collada-urdf-tools,libassimp4
Dependencies: 
4.1.0~dfsg-3 - libc6 (2 2.14) libgcc1 (2 1:3.0) libminizip1 (2 1.1) libstdc++6 (2 5.2) zlib1g (2 1:1.1.4) libassimp4:i386 (35 4.1.0~dfsg-3) libassimp4:i386 (38 4.1.0~dfsg-3) 
Provides: 
4.1.0~dfsg-3 - 
Reverse Provides: 
```

rhaschke, 2021-04-03:
Maybe you have another version of libassimp installed as well? What's the result of
`$ ldd /opt/ros/melodic/lib/librviz.so | grep assimp`

andreasBihlmaier, 2021-04-03:
Nope, resolves to lib above:
```
desknew ~/catkin_ws/src/rviz_issue_853> ldd /opt/ros/melodic/lib/librviz.so | grep assimp
        libassimp.so.4 => /usr/lib/x86_64-linux-gnu/libassimp.so.4 (0x00007f70fbb18000)
desknew ~/catkin_ws/src/rviz_issue_853> ll /usr/lib/x86_64-linux-gnu/libassimp.so.4
lrwxrwxrwx 1 root root 18 Dez 21  2017 /usr/lib/x86_64-linux-gnu/libassimp.so.4 -> libassimp.so.4.1.0
```

andreasBihlmaier, 2021-04-03:
I had the same issue on a fresh install of focal + noetic.

The problem is that my systems had
```
desknew ~> env | egrep '^(LANG|LC)'
LANG=en_US.UTF-8
LANGUAGE=
LC_ADDRESS=de_DE.UTF-8
LC_IDENTIFICATION=de_DE.UTF-8
LC_MEASUREMENT=de_DE.UTF-8
LC_MONETARY=de_DE.UTF-8
LC_NAME=de_DE.UTF-8
LC_NUMERIC=de_DE.UTF-8
LC_PAPER=de_DE.UTF-8
LC_TELEPHONE=de_DE.UTF-8
LC_TIME=de_DE.UTF-8
```
from the default Ubuntu installer.

After `unset LC_NUMERIC` everything works as expected:
[image]

rhaschke, 2021-04-05:
Hm. I suspected the language settings from the very first moment. Looks like, I missed testing the correct variable...