ros-visualization/rviz #854: Alpha transparency of mesh markers is broken

hersh, 2015-02-26:
Back in the old days when I worked on rviz, I made mesh markers obey the "alpha" field of their color, like all the other marker types do.  I made a test program that shows a bunch of combinations.  Recently (on HEAD) I was trying to show a semi-transparent mesh marker and it was shown opaque.  I re-ran my test program (src/test/mesh_marker_test.cpp) and I see that it no longer behaves as I intended.

Here is the new image:
[image]

You can see the spheres in the image show the alphas as intended.

I'll try to narrow this down to a particular revision.

trainman419, 2015-02-26:
It looks like this was introduced in #752 

It looks like your old program and @brindza 's changes are both achieving the same results, through different APIs.

Thoughts @brindza ?

hersh, 2015-02-26:
Damn, it looks like I "+1"ed the original change.  I guess I didn't run my test program at the time. :(

hersh, 2015-02-26:
Perhaps the issue is that the colorized meshes in #752 and the pr2-base meshes I show here use different features of the collada file type.  The test program mesh_marker_test.cpp should really include something like the rgb axes shown in #752 which showed the problem there.

NikolausDemmel, 2015-10-17:
Just browsing through the issue list. I guess this can be closed, given #893 is released?

NikolausDemmel, 2015-10-17:
(Maybe the port forward to jade is still missing, at least there is a label #893 that says so)

wjwwood, 2015-10-21:
@NikolausDemmel I must have accidentally put that tag on there. Jade and Indigo are released from the same branch.

wjwwood, 2015-10-21:
I'll let @hersh decide if this can be closed based on #893

LemonPi, 2022-07-22:
This is still an issue

rhaschke, 2022-07-28:
The default use case (`mesh_use_embedded_materials = False`) works as expected:
[image]

@LemonPi, if `mesh_use_embedded_materials = True` doesn't work as expected, please open a new issue and provide a corresponding mesh file comprising embedded materials.

closed 2022-07-28 by rhaschke