ros-visualization/rviz #863: rviz indigo-devel OS X 10.10.2 cannot load rviz/MoveCamera when started from rqt_gui

jmtatsch, 2015-03-18:
I wanted to add a nice rviz display to my rqt dashboard, but it refuses to load:

```
The class required for this tool, 'rviz/MoveCamera', could not be loaded.
Error:
MultiLibraryClassLoader: Could not create class of type rviz::MoveTool
```

Does anyone else have this issue or knows how to solve it.

wjwwood, 2015-03-18:
It's not just `rviz/MoveCamera`.

It looks like rviz is not on your path.

jmtatsch, 2015-03-21:
How do you mean not on my path?
```
echo $PATH
/usr/local/bin:/Users/tatsch/ros_catkin_ws/devel/bin:/opt/ros/indigo/bin:/usr/local/bin:/usr/local/sbin:/usr/bin:/bin:/usr/sbin:/sbin:/usr/texbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/opt/X11/bin:/usr/texbin
```

wjwwood, 2015-03-24:
I should have been more specific, I think rviz might not be on the `ROS_PACKAGE_PATH`. But after looking at it, I think it might be something else. Based on the output you've posted, I'm not sure I can debug it. I may need to reproduce the problem myself.

jmtatsch, 2015-03-24:
```
echo $ROS_PACKAGE_PATH                                                                                                                                                                
/Users/tatsch/ros_catkin_ws/src:/opt/ros/indigo/share:/opt/ros/indigo/stacks
```

```
which rviz                                                                                                                                                                            
/opt/ros/indigo/bin/rviz
```

I tried adding /opt/ros/indigo/bin/ to the package path but nothing changed...

wjwwood, 2015-03-24:
Those class that cannot be initialized are in `librviz.so` or `librviz.dylib` not in the executable.

But like I said, it's not clear to me now why they are not loaded. I would avoid using rviz in rqt for now, as there are a few issues about rviz embedded in rqt:

https://github.com/ros-visualization/rqt_robot_plugins/issues/81
(others I can't find right now)

We've been wanting to spend some time on this for a while now, but we haven't gotten around to it.