ros-visualization/rviz #865: Delete all rviz markers

JessChL, 2015-03-19:
On the tutorial ( http://wiki.ros.org/rviz/Tutorials/Ma... ) and docs for the visualisation markers, there's supposed to be an option for action "DELETEALL" which would remove all markers. I've tried this and it has not deleted any markers. Is this function actually in effect? Is there anything else I need to specify apart from marker.action?

wjwwood, 2015-03-19:
Can you tryout the pull request here: https://github.com/ros-visualization/rviz/pull/866

Or give me a set of instructions (example script, bag files, or whatever) to reproduce the problem.

JessChL, 2015-03-19:
I didn't do a source install and the use case is simply that I have a bunch of markers, I try to clear them by adding another one with action = DELETEALL and it doesn't do anything.

I ended up using rviz_visual_tools::RvizVisualTools and calling deleteAllMarkers() - did the trick for me.

wjwwood, 2015-03-19:
Ok, well in that case it's probably not a bug in rviz, since that mechanism is how `rviz_visual_tools` works. I still think the change in #866 is worth while. I'll close this for now.

JessChL, 2015-03-19:
Btw, just so someone is aware of it, the DELETE action doesn't work either. We "hacked" it by changing the alpha to 0 if we only want a specific marker gone.

wjwwood, 2015-03-19:
@JessChL I don't understand, you said that `DELETEALL` works with `rviz_visual_tools` right?

wjwwood, 2015-03-19:
> the use case is simply that I have a bunch of markers, I try to clear them by adding another one with action = DELETEALL and it doesn't do anything.

Can you provide a simple C++ or Python example of this?

JessChL, 2015-03-19:
Sorry, I should have made it clearer.

This works:
------------------------------------------------------------------------------------
rviz_visual_tools::RvizVisualTools rviz_interface("dummy_wall","/visualization_marker")
rviz_interface.deleteAllMarkers();
------------------------------------------------------------------------------------

This DOESN'T work:
------------------------------------------------------------------------------------
ros::Publisher marker_pub;
marker_pub = n.advertise<visualization_msgs::Marker>("visualization_marker", 1);
visualization_msgs::Marker text_marker;
text_marker.header.frame_id = "wall";
text_marker.ns = "points";
text_marker.type = text_marker.TEXT_VIEW_FACING;
text_marker.action = text_marker.DELETE; // this doesn't work, neither does .DELETEALL
text_marker.color.a = 0.0; // this is a hack to make marker invisible
text_marker.lifetime = ros::Duration();
marker_pub.publish(text_marker);
--------------------------------------------------------------

wjwwood, 2015-03-19:
Ok, well it should work, I'm going to reopen this until we can figure out why you're not able to get it to work. AFAICT, the `rviz_visual_tools` isn't doing anything special.

rethink-rlinsalata, 2015-03-26:
FYI, this may be reading too much details into your example code, but did you try publishing the delete / deleteall in a loop?  I had a problem where I published a DELETEALL at the start of my program in an attempt to "reset existing clients", but the first message never got through (in fact, I often couldn't even echo it on the topic).  I fixed it by adding a loop beforehand to wait for the num_connections > 0 (or timeout).  Something like (in python):

```python
    while not marker_pub.get_num_connections():
        rospy.sleep(1.0)
    marker_pub.publish(text_marker)
```
I added a timeout and a is_shutdown() check, but you get the idea.  I also added on a `SubscriberListener` to my Publisher for new client connections, so that I could send a DELETEALL message as the first message to all clients.

One other issue I ran into is that it seems rviz's MarkerArray requires a valid tf2 transform in your msg's Header - i.e. if the frame_id in your DELETEALL message is empty or the transforms aren't currently being published for it, the message is dropped without the DELETEALL taking effect.  I would get warnings in my terminal like this:

```
[ WARN] [1427379027.945965162]: MessageFilter [target=world ]: Discarding message from [unknown] due to empty frame_id.  This message will only print once.
[ WARN] [1427379027.946080497]: Invalid argument passed to canTransform argument source_frame in tf2 frame_ids cannot be empty
```

JessChL, 2015-03-27:
The frame_id is definitely set and the messages are definitely published and received because setting alpha to 0.0 make it completely transparent but the DELETE or DELETEALL is not in effect because the marker is still in rviz's list.

PR #866 (2015-04-15): Potential fix for DELETEALL marker action
See:

- http://answers.ros.org/question/205322/delete-all-rviz-markers/
- https://github.com/ros-visualization/rviz/issues/865

Fixes #865
closed 2015-04-15 by wjwwood