ros-visualization/rviz #879: Angular torque marker is dettached with negative z-axis torque

130s, 2015-04-20:
Originally reported in https://github.com/tork-a/dynpick_driver/issues/16.

[image]
[image]

To reproduce,
```
term-1$ rosrun rviz rviz   (then open Wrench plugin on RViz, choose "`/force`" topic)

term-2$ rosrun tf static_transform_publisher 0.0.0 0.0 0.0 0.0 0.0 2.0 /base /sensor 100

term-3$ rostopic pub -r 10 /force geometry_msgs/WrenchStamped '{ header: { seq: 1, stamp: 11, frame_id: "base" }, wrench: { force: { x: 0.5, y: 0.5, z: 1.0 }, torque: { x: 0.5, y: 0.5, z: -1.1 } } }'
```

```
$ dpkg -p ros-indigo-rviz|grep Vers
Version: 1.11.7-0trusty-20150328-2254-+0000
```
But another user reported that this happens on Hydro too.

goretkin, 2015-05-02:
`stamp: now` works better than `stamp: 11` for reproducing the issue. am submitting a PR soon

wjwwood, 2015-05-19:
@130s can you test https://github.com/ros-visualization/rviz/pull/883 to see if it solves your issue?

130s, 2015-05-20:
Will do this weekend.

wjwwood, 2015-08-05:
Closing in favor of https://github.com/ros-visualization/rviz/pull/883

commit 607271d7 (2015-08-05): fix #879, torque marker
commit b82749da (2015-08-05): Merge pull request #883 from goretkin/fix-879

Fix #879
commit 5b4e8172 (2015-08-05): fix #879, torque marker
commit c56f989c (2015-08-05): fix #879, torque marker
closed 2015-08-05 by wjwwood