ros-visualization/rviz #882: Segfault within YamlConfigReader::readYamlNode

mwerlberger, 2015-04-28:
After updating rviz to version 1.11.7 on my Arch Linux system I get a segfault when starting rviz.

gdb output:
```
(gdb) run
Starting program: /opt/ros/indigo/bin/rviz 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/usr/lib/libthread_db.so.1".
[ INFO] [1430203170.862147809]: rviz version 1.11.7
[ INFO] [1430203170.862213114]: compiled against OGRE version 1.9.0 (Ghadamon)
[New Thread 0x7fffde0a3700 (LWP 11205)]
[New Thread 0x7fffde8a4700 (LWP 11200)]
[New Thread 0x7fffdf0a5700 (LWP 11199)]
[New Thread 0x7fffdf8a6700 (LWP 11198)]

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff6678acb in std::basic_string<char, std::char_traits<char>, std::allocator<char> >::basic_string(std::string const&) ()
   from /usr/lib/libstdc++.so.6
(gdb) where
#0  0x00007ffff6678acb in std::basic_string<char, std::char_traits<char>, std::allocator<char> >::basic_string(std::string const&) ()
   from /usr/lib/libstdc++.so.6
#1  0x00007ffff7b5af7e in rviz::YamlConfigReader::readYamlNode(rviz::Config&, YAML::Node const&) () from /opt/ros/indigo/lib/librviz.so
#2  0x00007ffff7b5d462 in rviz::YamlConfigReader::readStream(rviz::Config&, std::istream&, QString const&) ()
   from /opt/ros/indigo/lib/librviz.so
#3  0x00007ffff7b5e65e in rviz::YamlConfigReader::readFile(rviz::Config&, QString const&) () from /opt/ros/indigo/lib/librviz.so
#4  0x00007ffff7b4472c in rviz::VisualizationFrame::loadPersistentSettings() () from /opt/ros/indigo/lib/librviz.so
#5  0x00007ffff7b47351 in rviz::VisualizationFrame::initialize(QString const&) () from /opt/ros/indigo/lib/librviz.so
#6  0x00007ffff7b505d8 in rviz::VisualizerApp::init(int, char**) () from /opt/ros/indigo/lib/librviz.so
#7  0x0000000000400c7e in main ()
```

The installed package:
```
✗ yaourt -Qs ros-indigo-rviz
local/ros-indigo-rviz 1.11.7-2
```

Should I dig deeper or is this a known issue? To ensure not having library inconsistencies, I updated all first level dependencies that are needed by rviz.

Best,
 Manuel

hersh, 2015-04-28:
I've never used arch linux, but I wrote that code in rviz and it doesn't seem like a known issue to me. :)  Digging deeper would definitely be appreciated.

hersh, 2015-04-28:
This may be related to the yaml-cpp API change at version 0.5.  Since you are using arch linux, perhaps the yaml-cpp version detection didn't work the same as on ubuntu?

wjwwood, 2015-04-28:
I don't think it would have compiled if the yaml version detection had not worked, but it is probably the case that the yaml-cpp 0.5 code path is more poorly tested since OS X and Ubuntu are still using 0.3. For a long time the 0.5 version of yaml-cpp did not have a way to provide backwards compatibility though `#ifdef`'s.

mwerlberger, 2015-04-29:
I've yaml-cpp 0.5.2 installed. I'll check if this causes the issue. I'm currently re-installing all ros dependencies because there has been a boost update and I want to avoid that this causes any additional hickups. I'll then check if yaml-cpp causes the segfault and otherwise compile rviz with debug flags so we can check the full stack trace. I'll keep you updated....

mwerlberger, 2015-04-29:
Downgrading to yaml-cpp 0.3 did the trick. Is there the intention to make this work with 0.5.2? Otherwise I will contact the maintainer of the rviz package in arch for having the  dependencies corrected....

trainman419, 2015-04-29:
I'm not too familiar with how arch distributes packages; are you installing pre-compiled binaries or automatically building from source?

The source build of rviz should work with version 0.3 and 0.5 of yaml-cpp.

Given the massive changes in yaml-cpp between 0.3 and 0.5, I would expect a binary version of rviz that was built against one version to fail to load and/or explode wildly when run with the other version.

mwerlberger, 2015-04-29:
It compiles the packages from source. For package details you can check [here](https://aur.archlinux.org/packages/ros-indigo-rviz). It has a dependency to [yaml-cpp](https://www.archlinux.org/packages/community/x86_64/yaml-cpp/) which installs 0.5.2 from source. There is a package [yaml-cpp03](https://aur.archlinux.org/packages/yaml-cpp0.3) which is in conflict with 0.5.2 and installs version 0.3. After reinstalling rviz then it links correctly against yaml-cpp 0.3 and it works without segfaulting.

@trainman419 what do you mean with working with 0.3 and 0.5. For me everything comiles fine with 0.5 but as mentioned it segfaults then. Do you have it working (and starting up) with 0.5? If so I can compile rviz with debug flags if you like and we can try to fix this with more information about where and why it segfaults....

wjwwood, 2015-04-29:
From what I've read here, it sounds like yaml-cpp 0.5's code path in rviz has a bug. If you have a chance, please build rviz with yaml-cpp 0.5 and in debug mode so you can get a reasonable back trace (one with line numbers) from `gdb`. Thanks.

mwerlberger, 2015-04-30:
I need to have a working system today but I will create and send the logs during the weekend.

closed 2016-08-25 by mwerlberger