ros-visualization/rviz #884: change rviz log level

goretkin, 2015-05-04:
currently I don't know how to see rviz debug messages (without turning on the logger level to DEBUG for every ROS node).

for reference:
http://answers.ros.org/question/208335/change-rviz-log-level/?comment=208425#comment-208425

trainman419, 2015-05-04:
As a workaround, it should be possible the change the rviz log level at runtime with rqt_logger_level.

goretkin, 2015-05-04:
I believe i tried that but there are no loggers in the anonymous rviz node

wjwwood, 2015-05-04:
@goretkin That seems like it might be a different issue. I'll try to reproduce what you're describing, but I know I've set rviz to debug after launch using the `rqt_logger_level` tool. What I have not been able to do in the past is start rviz in debug mode, which is a general problem with ros nodes. There is a config file, but it needs the node name and since the rviz node always changes at startup, it's not really useful.

wjwwood, 2015-08-05:
Ok, I've added a command line option to turn debug on with rviz for convenience: https://github.com/ros-visualization/rviz/pull/904

closed 2015-08-05 by wjwwood