ros-visualization/rviz #885: Pointclouds with RGBA fields cannot be colored using RGB

calderpg, 2015-05-05:
Pointclouds created using PCL::PointXYXRGBL and some other PCL types have a field named "rgba", rather than the "rgb" value expected by Rviz. (For alignment issues, it's always rgba, except for the apparently special case of PCL::PointXYZRGB which reports it as "rgb"). Without the "rgb" field, Rviz does not provide a RGB color transformer, so you cannot display the real colors of the points in Rviz.

As far as I can tell, the problematic check is here:

https://github.com/ros-visualization/rviz/blob/indigo-devel/src/rviz/default_plugin/point_cloud_common.cpp#191

Instead of `name == "rgb"` this should be `name == "rgb" || name == "rgba"`.

Obviously, this doesn't actually add support for drawing transparency, but rendering the RGB color for all poinclouds that support it is very useful.

wjwwood, 2015-05-05:
Thanks for the report. If you think the above solution would work, please make a pull request and perhaps provide a bag file so it can easily be tested.

Otherwise you can wait for someone else to get to it, but it might take a while.

wjwwood, 2015-08-05:
Closed by #885

closed 2015-08-05 by wjwwood