ros-visualization/rviz #888: tf display tool does not handle static transforms correctly

davetcoleman, 2015-05-10:
Seems to me static tf2 transforms should always display in Rviz, but instead they fade out as if they were non-static. Resetting the tf display makes the transform show again until the "Frame Timeout" is reached. 

According to @tfoote 

> The rviz display needs to be updated to know the difference between regular and static transforms. For efficiency static_transforms are now published as latched topics. The gray out logic in rviz's display does not take that into account. This should be filed against the rviz tf plugin.

wjwwood, 2015-05-18:
Thanks for the report, I'll try to have a look at this soon. @tfoote do you know what versions of ROS this will affect?  Did we do static tf transforms starting in hydro or indigo?

tfoote, 2015-05-18:
Static transforms started in hydro

closed 2015-08-05 by wjwwood

eric-schleicher, 2018-03-21:
would we expect this behavior to be any different if it's called from within python?  I'm using a (single)  tf2_ros static transform broadcaster to handle periodic updates for a series of infrequently changing transformations in my system.  the rqt TF tree display correctly, and other code seems to work fine, however rviz refuses to show the markers for more than just a few seconds after update, and then generally only for the most recenty submitted transform.

When i have the same static transforms in a launch file, it does not behave in this fashion (it works as expected in rviz).   would this be a new issue at this point; given this is many years old now?

commit ea0b402d (2015-08-05): Merge pull request #899 from terakoji/issue-888

Fix #888: TF plugin handles static transform of tf2.