ros-visualization/rviz #889: Raspberry pi ROS rviz segmentation fault

ppersia37, 2015-05-17:
i recently installed ROS on Raspberry pi 2 in order to implement SLAM algorithm on it every thing went verry well until i run the rviz (rosrun rviz rviz ) i get segmentation fault msg . any body have a solution for this situation .thakns in advance

wjwwood, 2015-05-18:
If you expect to get help, you might want to consider what information others need to help you, for example:

- What version of OS you are using on the rpi2
- What version of rviz
- What version of ROS
- What was the console output
- What was output to the ogre log file

Anything else you think might be useful. Just saying that there was a segfault is not enough information.

Likely you'll need to compile rviz in debug mode and run it with `gdb` in order to get a traceback of the segfault.

wjwwood, 2015-05-18:
I'm going to close this until you update the issue with more information. If you do so I'll reopen the issue.

ppersia37, 2015-05-18:
ok im using debian wheezy  OS   on rpi2   with  ROS  indigo  version and the Console output is just  
"Segmentation Fault"

wjwwood, 2015-05-18:
Is there an Ogre log file in the directory you ran rviz in? Without a traceback from `gdb` it's going to be really hard to resolve this issue. Are you opening rviz with an rviz config file set?

ppersia37, 2015-05-19:
when  i  run  gdb /opt/ros/indigo/share/rviz/rviz  i get 
/opt/ros/indigo/share/rviz/rviz: No such file or directory.

but  when i run   rospack find rviz  i get 

/opt/ros/indigo/share/rviz

im realy  confused

trainman419, 2015-05-19:
The rviz binary is not in /opt/ros/indigo/share/rviz; it should be in /opt/ros/indigo/lib/rviz

wjwwood, 2015-05-19:
What @trainman419 said. Unfortunately running `rviz` with `gdb` without first rebuilding it in with `-DCMAKE_BUILD_TYPE=Debug` will not produce a useful traceback most likely. However, if you do manage to get a `gdb` traceback with the released version of `rviz`, then please post it, you never know what might help.

ppersia37, 2015-05-19:
when i run  gdb /opt/ros/indigo/lib/rviz/rviz  i get   Reading symbols from /opt/ros/indigo/lib/rviz/rviz...(no debugging symbols found)...done.   

please ca you tell me how to  get a gdb traceback with the released version of rviz

thank you for your help

wjwwood, 2015-05-19:
You need to run `rviz` after `gdb` finishes "reading symbols" by typing `r` and pressing enter. Then when `rviz` crashes you can type `bt` and press enter to get a backtrace. I'd recommend having a look at a `gdb` guide like this one: http://beej.us/guide/bggdb/

ppersia37, 2015-05-30:
thanks for the Description   and im sorry of being late  to  update my issue

this  is  what i get  when i follow your instructions with gdb

(gdb) r
Starting program: /opt/ros/indigo/lib/rviz/rviz
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".

Program received signal SIGILL, Illegal instruction.
0x6f87c608 in ?? () from /usr/lib/arm-linux-gnueabihf/libcrypto.so.1.0.0
(gdb) bt
#0  0x6f87c608 in ?? () from /usr/lib/arm-linux-gnueabihf/libcrypto.so.1.0.0
Cannot access memory at address 0x0
#1  0x6f879004 in OPENSSL_cpuid_setup ()
   from /usr/lib/arm-linux-gnueabihf/libcrypto.so.1.0.0
#2  0x76fe8254 in ?? () from /lib/ld-linux-armhf.so.3
#3  0x7efff59c in ?? ()
Cannot access memory at address 0x0
#4  0x7efff59c in ?? ()
Cannot access memory at address 0x0
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb)

wjwwood, 2015-06-01:
As I feared, that's not a particularly useful traceback. You would need to build `rviz` with debug on to get a better idea of the problem.

wjwwood, 2015-08-05:
Please get us a core dump or debug traceback from this. Until then it's not even clear what the source of the error is and I'm going to close the issue. We can reopen it if more information comes up.

closed 2015-08-05 by wjwwood

dchang0, 2015-12-23:
This appears to be a "bug" in that RVIZ does not handle the SIGILLs that OpenSSL produces **normally** gracefully. I have opened a better-documented bug here:

https://github.com/ros-visualization/rviz/issues/955

Ash-S, 2020-10-21:
Hi @wjwwood, I am facing the similar issue while tried to run 'rosrun rviz rviz' on my SBC. 
Here are the details when i run 'rosrun rviz rviz'

[image]

Here are the details when i run gdb.

[image]

Looking forward to your kind suggestions to resolve this. Thanks

wjwwood, 2020-10-21:
@Ash-S please open a new issue and please do not post pictures of text, it cannot be copied or searched.

Ash-S, 2020-10-21:
> @Ash-S please open a new issue and please do not post pictures of text, it cannot be copied or searched.

@wjwwood , I understood. Yes i created a new issue. Please check that. Thanks.