ros-visualization/rviz #891: RVIZ segfault on Ubuntu ARM

DonatasKozlovskis, 2015-05-29:
Hello!

I recently faced I guess the same segfault in Ubuntu arm with ROS Indigo on jetson tk1.
It was working a about three weeks ago, but after some system updates I am getting the segfault.

<pre>
[ INFO] [1432890900.005677779]: rviz version 1.11.7
[ INFO] [1432890900.005958854]: compiled against OGRE version 1.8.1 (Byatis)
[ INFO] [1432890900.933923738]: Stereo is NOT SUPPORTED
[ INFO] [1432890900.934283311]: OpenGl version: 4.4 (GLSL 4.4).
Segmentation fault
</pre>

After compiling with build type = debug,  the only thing I was able to trace:
 <pre>
[ INFO] [1432891088.177578371]: Stereo is NOT SUPPORTED
[ INFO] [1432891088.177833955]: OpenGl version: 4.4 (GLSL 4.4).
[New Thread 0xa8eb23b0 (LWP 7202)]
[New Thread 0xa86b23b0 (LWP 7203)]

Program received signal SIGSEGV, Segmentation fault.
is_anchored (code=code@entry=0x624c70 "}", bracket_map=bracket_map@entry=0, 
    backref_map=backref_map@entry=0) at pcre_compile.c:7339
7339	pcre_compile.c: No such file or directory.
</pre>

and backtrace output:
<pre>
#0  is_anchored (code=code@entry=0x624c70 "}", bracket_map=bracket_map@entry=0, 
    backref_map=backref_map@entry=0) at pcre_compile.c:7339
#1  0xb37c7d14 in is_anchored (code=code@entry=0x624c70 "}", 
    bracket_map=bracket_map@entry=0, backref_map=backref_map@entry=0)
    at pcre_compile.c:7350
#2  0xb37c7d14 in is_anchored (code=code@entry=0x624c70 "}", 
    bracket_map=bracket_map@entry=0, backref_map=backref_map@entry=0)
    at pcre_compile.c:7350
#3  0xb37c7d14 in is_anchored (code=code@entry=0x624c70 "}", 
    bracket_map=bracket_map@entry=0, backref_map=backref_map@entry=0)
    at pcre_compile.c:7350
#4  0xb37c7d14 in is_anchored (code=code@entry=0x624c70 "}", 
    bracket_map=bracket_map@entry=0, backref_map=backref_map@entry=0)
    at pcre_compile.c:7350
#5  0xb37c7d14 in is_anchored (code=code@entry=0x624c70 "}", 
    bracket_map=bracket_map@entry=0, backref_map=backref_map@entry=0)
    at pcre_compile.c:7350
#6  0xb37c7d14 in is_anchored (code=code@entry=0x624c70 "}", 
    bracket_map=bracket_map@entry=0, backref_map=backref_map@entry=0)
    at pcre_compile.c:7350
#7  0xb37c7d14 in is_anchored (code=code@entry=0x624c70 "}", 
    bracket_map=bracket_map@entry=0, backref_map=backref_map@entry=0)
    at pcre_compile.c:7350
#8  0xb37c7d14 in is_anchored (code=code@entry=0x624c70 "}", 
    bracket_map=bracket_map@entry=0, backref_map=backref_map@entry=0)
    at pcre_compile.c:7350
</pre>

Please let me know if I could somehow provide more details.

DonatasKozlovskis, 2015-05-29:
as well, tail of rviz -l
<pre>
[ INFO] [1432891395.762779648]: Parsing scripts for resource group Autodetect
[ INFO] [1432891395.763192471]: Finished parsing scripts for resource group Autodetect
[ INFO] [1432891395.763610294]: Creating resources for group Autodetect
[ INFO] [1432891395.764189779]: All done
[ INFO] [1432891395.764438690]: Parsing scripts for resource group General
[ INFO] [1432891395.764695183]: Finished parsing scripts for resource group General
[ INFO] [1432891395.764992759]: Creating resources for group General
[ INFO] [1432891395.765255336]: All done
[ INFO] [1432891395.765673409]: Parsing scripts for resource group Internal
[ INFO] [1432891395.766191562]: Finished parsing scripts for resource group Internal
[ INFO] [1432891395.766591636]: Creating resources for group Internal
[ INFO] [1432891395.766930710]: All done
[ INFO] [1432891395.767251786]: Parsing scripts for resource group rviz
[ INFO] [1432891395.771487929]: Parsing script glsl120.program
[ INFO] [1432891395.780156879]: Parsing script nogp.program
[ INFO] [1432891395.785629575]: Parsing script glsl150.program
[ INFO] [1432891395.789912051]: Parsing script point_cloud_point.material
[ INFO] [1432891395.791537010]: Parsing script indexed_8bit_image.material
[ INFO] [1432891395.792259575]: Parsing script point_cloud_tile.material
[ INFO] [1432891395.793895451]: Parsing script default_pick_and_depth.material
[ INFO] [1432891395.797631024]: Parsing script point_cloud_box.material
[ INFO] [1432891395.799957716]: Parsing script point_cloud_flat_square.material
[ INFO] [1432891395.802203160]: Parsing script point_cloud_sphere.material
[ INFO] [1432891395.804940174]: Parsing script point_cloud_square.material
[ INFO] [1432891395.806814794]: Parsing script arial.fontdef
[ INFO] [1432891395.807716438]: Finished parsing scripts for resource group rviz
[ INFO] [1432891395.808157010]: Creating resources for group rviz
[ INFO] [1432891395.808454420]: All done
[ INFO] [1432891395.859507556]: GLRenderSystem::_createRenderWindow "OgreWindow(1)", 100x30 windowed  miscParams: externalGLControl= macAPI=carbon parentWindowHandle=62914593 
[ INFO] [1432891395.887155780]: GLXWindow::create used FBConfigID = 141
Segmentation fault
</pre>

DonatasKozlovskis, 2015-05-30:
more useful backtrace:
<pre>
Breakpoint 1, first_significant_code (code=0x5cdbf0 "}", skipassert=skipassert@entry=0)
    at pcre_compile.c:1651
1651	pcre_compile.c: No such file or directory.
(gdb) bt                       
#0  first_significant_code (code=0x5cdbf0 "}", skipassert=skipassert@entry=0) at pcre_compile.c:1651
#1  0xb35eccf6 in is_anchored (code=code@entry=0x5cdbf0 "}", bracket_map=bracket_map@entry=0, 
    backref_map=backref_map@entry=0) at pcre_compile.c:7341
#2  0xb35f30de in pcre_compile2 (pattern=<optimized out>, options=<optimized out>, 
    errorcodeptr=errorcodeptr@entry=0x0, errorptr=errorptr@entry=0xbeffe16c, erroroffset=0xbeffe170, 
    tables=0xb3618a70 <_pcre_default_tables> "", tables@entry=0x0) at pcre_compile.c:8076
#3  0xb35f3240 in pcre_compile (pattern=<optimized out>, options=<optimized out>, errorptr=0xbeffe16c, 
    erroroffset=<optimized out>, tables=0x0) at pcre_compile.c:7616
#4  0xabfafa72 in pcrecpp::RE::Compile(pcrecpp::RE::Anchor) ()
   from /usr/lib/arm-linux-gnueabihf/libpcrecpp.so.0
#5  0xabfafb7a in pcrecpp::RE::Init(std::string const&, pcrecpp::RE_Options const*) ()
   from /usr/lib/arm-linux-gnueabihf/libpcrecpp.so.0
#6  0xabfafbc4 in pcrecpp::RE::Init(char const*, pcrecpp::RE_Options const*) ()
   from /usr/lib/arm-linux-gnueabihf/libpcrecpp.so.0
#7  0xa8ce84d4 in ?? () from /opt/ros/indigo/lib/liburdf.so
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
</pre>

It turns out that this segfault was caused by URDF package update 1.11.7.
Downgrading back to 1.11.6 makes RVIZ to run successfully

wjwwood, 2015-06-01:
@DonatasKozlovskis thanks for digging into the issue, can you try the changes in this pull request:

https://github.com/ros/robot_model/pull/108

That was merged after `1.11.7`, maybe it resolves the issue. If not, can you open an issue against ros/robot_model?

DonatasKozlovskis, 2015-06-02:
Thank you for the suggestion. 
However, it did not resolve the issue. Thus I am closing this one I will open an issue against ros/robot_model.

closed 2015-06-02 by DonatasKozlovskis