ros-visualization/rviz #894: Imu visualization is missing a scale parameter

jondo, 2015-06-08:
I am afraid I don't really know whether this is the right place for filing this bug:

In ROS Indigo's RViz, subscribing to a sensor_msgs/Imu topic (of e.g. the xsens_driver) yields a display of (in my case) `/imu/data` with a much too large arrow, as described [here](http://docs.ros.org/indigo/api/rviz_plugin_tutorials/html/display_plugin_tutorial.html).

This display should have a scale parameter to reduce the arrow size.

wjwwood, 2015-06-25:
The imu display, and the associated tutorial package, are in this repository:

https://github.com/ros-visualization/visualization_tutorials

I don't have time to add this feature, but if you're interested in extending the tutorial to include a scale option, that'd be great. You could open a pull request against the above repository.

You could also migrate this issue to the issue tracker on that repository, but either way I'm going to close this one for now. Feel free to continue to comment here though.

Thanks!

closed 2015-06-25 by wjwwood

jondo, 2015-07-18:
Ok, I have migrated this to https://github.com/ros-visualization/visualization_tutorials/issues/27.

slabua, 2022-08-31:
Ping?