ros-visualization/rviz #895: librviz Python Bindings - Segmentation Fault when using method "setStatusBar()"

futureneer, 2015-06-25:
Hello,
I am getting a segmentation fault when trying to hide the status bar in the rviz VisualizationFrame widget.  Here is a minimal example which produces the segfault:

```python
#! /usr/bin/env python
import sys
import rviz
from python_qt_binding.QtGui import *
from python_qt_binding.QtCore import *

class TestWidget(QWidget):
    def __init__(self):
        QWidget.__init__(self)

        # config_path = '~/.rviz/default.rviz'
        config_path = ''

        self.frame= rviz.VisualizationFrame()
        self.frame.setSplashPath('')
        self.frame.setMenuBar(None)
        self.frame.setStatusBar(None)
        self.frame.initialize()

        reader = rviz.YamlConfigReader()
        config = rviz.Config()
        reader.readFile(config,config_path)
        self.frame.load(config)

        self.manager = self.frame.getManager()

        layout = QVBoxLayout()
        layout.addWidget( self.frame )

        top_button = QPushButton( "Top View" )
        layout.addWidget( top_button )

        self.setLayout(layout)

if __name__ == '__main__':
    app = QApplication( sys.argv )
    test_widget = TestWidget()
    test_widget.resize( 500, 500 )
    test_widget.show()
    app.exec_()
```

By commenting out the line:
```python
    # self.frame.setStatusBar(None)
```

the code runs just fine.

Let me know what you think, and thanks!!

Kel

wjwwood, 2015-06-25:
Thanks for the detailed report. Nothing comes to mind off the top of my head. Have you tried sending an empty string instead of None?

I'd have to reproduce this and look through the tracebacks with gdb. I'm not sure when I'll have time to do that.

futureneer, 2015-06-25:
Hi, thanks for the reply. I did as you said, and it says it doesn't take strings. Incidentally, the setHideButtonVisibility method also causes a segfault as well. Not sure if they are related.

I guess I could create an empty QStatusBar and put that in, since that's the type.  I'll try it.

futureneer, 2015-06-25:
No dice, the empty status bar doesn't do it.  Its not a critical thing, I'm just trying to create a clean interface as possible, and though I could free up some real estate by getting rid of the status bar.  I'll deal if its a pain.

wjwwood, 2015-06-25:
Looks like it might be related to the function which updates the fps:

```
#0  0x00007ffff2efe9e0 in QString::operator==(QString const&) const () from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#1  0x00007ffff141b285 in QLabel::setText(QString const&) () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#2  0x00007fffe38d8ff9 in rviz::VisualizationFrame::updateFps() () from /opt/ros/jade/lib/librviz.so
#3  0x00007ffff2fd187a in QMetaObject::activate(QObject*, QMetaObject const*, int, void**) ()
   from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#4  0x00007fffe38e4b99 in rviz::VisualizationManager::onUpdate() () from /opt/ros/jade/lib/librviz.so
#5  0x00007ffff2fd187a in QMetaObject::activate(QObject*, QMetaObject const*, int, void**) ()
   from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#6  0x00007ffff2fd5a31 in QObject::event(QEvent*) () from /usr/lib/x86_64-linux-gnu/libQtCore.so.4
#7  0x00007ffff1026e2c in QApplicationPrivate::notify_helper(QObject*, QEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#8  0x00007ffff102d4a0 in QApplication::notify(QObject*, QEvent*) () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#9  0x00007ffff2051186 in ?? () from /usr/lib/python2.7/dist-packages/PyQt4/QtGui.so
#10 0x00007ffff2fbd4dd in QCoreApplication::notifyInternal(QObject*, QEvent*) ()
```

futureneer, 2015-06-25:
Yea, I was just looking at the gdb backtrace too... hmm

```

```

futureneer, 2015-06-25:
Could it be that if you get rid of the status bar, the symbol that the FPS display is accessing doesn't exist anymore?

wjwwood, 2015-06-25:
Maybe, I'm interleaving this with other work. Currently compiling rviz in debug.

futureneer, 2015-06-25:
So, I tried testing to see if fps_label_ exists in that function, and printing out if it does. When I executed the code, it reported that it existed, and still segfaults in the same manner.

Changing line 184 to:
```c++
if(fps_label_){
      std::cerr << "FPS FOUND" << std::endl;
      fps_label_->setText( QString::number(int(fps)) + QString(" fps") );
    }
```

wjwwood, 2015-06-25:
Hmm, it looks like a problem in the `setText` function. It could be that it attempts to compare the current QString with the one you passed and the current one is invalid. Not sure.

futureneer, 2015-06-25:
Ok, so commenting out line 184 does the trick, and no longer segfaults.  I'll try and see what is being placed as an argument in setText when this happens.

wjwwood, 2015-06-25:
Ok, this patch seems to fix the issue: https://github.com/ros-visualization/rviz/pull/896

futureneer, 2015-06-25:
Ok, so I am running on the hydro-devel branch.  I cherry-picked this commit, and the issue remains.  Did I do something wrong? My visualization_frame file exactly matches the patch, but I still get a segfault.  It seems that even though I pass the setStatusBar a QStatusBar, the using_standard_status_bar_ is still true.

futureneer, 2015-06-25:
Also, it doesn't look like setStatusBar is getting called before the updateFPS, even though I call setStatusBar before I initialize the VisualizationFrame.

futureneer, 2015-06-25:
I think this is because fps_label_ is already set as a perminent widget of statusBar on construction.

futureneer, 2015-06-25:
I am trying another approach.  Do I need to do anything special to rebuild the python bindings besides catkin make?

futureneer, 2015-06-25:
Ok, for my purposes, I think I will just hide the statusBar directly from Python.  I think that has the least impact on everything.  Thanks for the help!

wjwwood, 2015-06-25:
That's odd, I was using indigo-devel (high recommend you upgrade if you're starting new work) and I was able to get your Python example to work with that patch as well as adding `setStatusBar(0);` at the end of the `VisualizationFrame`'s constructor.

wjwwood, 2015-08-05:
The `setStatusBar(0);` at the end of the constructor was hiding the python error. I have a new fix in #896 which actually resolves this issue.

wjwwood, 2015-08-05:
I'll close this with #896, please comment if anyone runs into it again.

closed 2015-08-05 by wjwwood