ros-visualization/rviz #908: Rviz crashes trying to load mesh in plugin after reset

jitrc, 2015-08-15:
http://docs.ros.org/indigo/api/rviz_plugin_tutorials/html/tool_plugin_tutorial.html
plant flag twice, reset and try planting flag twice more, it crashes showing it cannot find the resouece
package://rviz_plugin_tutorials/media/flag.dae

------------------------
[1439668555.838213851]: OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource package://rviz_plugin_tutorials/media/flag.dae in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/buildd/ogre-1.8-1.8.1+dfsg/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
Qt has caught an exception thrown from an event handler. Throwing
exceptions from an event handler is not supported in Qt. You must
reimplement QApplication::notify() and catch all exceptions there.

terminate called after throwing an instance of 'Ogre::FileNotFoundException'
  what():  OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource package://rviz_plugin_tutorials/media/flag.dae in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/buildd/ogre-1.8-1.8.1+dfsg/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
Aborted (core dumped)

wjwwood, 2015-08-17:
Hmm, seems like the plugin is not handling reset correctly. Though I'm not sure how that would affect the resource management. Thanks for the report, but I don't have time to look into it at the moment. If you figure out anything else please let us know. Perhaps someone will have time to look into it.

I'll mark it as untargeted for now.

zfxw0206, 2015-10-01:
I have the same issue. Does anybody know how to fix it?

jitrc, 2015-10-01:
Unfortunately I also did not get the time to fix it. But as a workaround
disabled using a dae file as a resource to show the flagged position.

cellard0or, 2017-07-31:
I am currently experiencing the exact same problem (using Xenial) using not a dae file but a `Ogre::SceneManager::PT_SPHERE`. Did anyone have new insights about this issue? Meanwhile, can someone tell me, what exactly happens to a rviz::Display on a reset? Something seems to be not cleaned up correctly here..

LazyTitanik, 2023-07-19:
I have been troubling with this problem. Turns out, you have to "pre-load" the mesh file using rviz::loadMeshFromResource(filename)

See https://github.com/ros-visualization/visualization_tutorials/blob/0ba771c1ad56ff78afab17d243ac6e628db6b706/rviz_plugin_tutorials/src/plant_flag_tool.cpp#L90

Hope this helps.