ros-visualization/rviz #912: Rviz conflicts with gazebo6 because of ogre 1.8/1.9 dependency on Ubuntu

leftys, 2015-08-20:
ros indigo ubuntu package for rviz (together with dev packages for gazebo 2, 4 and 5) depends on libogre-1.8-dev. libgazebo6-dev depends on libogre-1.9-dev. The only way to install gazebo6-dev or gazebo6-ros packages is therefore to uninstall rviz. Could you change the rviz package so that it requires ```libogre-1.8-dev | libogre-1.9-dev```? 
Please, redirect me if this question should be asked elsewhere :).

I use repositories: 
deb http://packages.osrfoundation.org/gazebo/ubuntu trusty main 
deb http://packages.ros.org/ros/ubuntu trusty main

trainman419, 2015-08-20:
The supported versions of gazebo and ogre on Ubuntu Trusty are Gazebo 5 and Ogre 1.8. If you want to use other versions, you can build rviz from source.

See issue #910

leftys, 2015-08-20:
Ok, I assume there is a reason why to depend on ogre 1.8 and not 1.9. If that is correct, feel free to close the issue.

wjwwood, 2015-08-20:
Yeah, it's to support installing Jade and Indigo side-by-side: https://github.com/ros-infrastructure/rep/pull/89#issuecomment-69232117

closed 2015-08-20 by wjwwood