ros-visualization/rviz #913: rviz 1.11.8 does not load stl

nxdefiant, 2015-08-26:
rviz 1.11.8 does not display my stl model, in the console I get:

> [ INFO] [1440539561.574488292]: rviz version 1.11.8
> [ INFO] [1440539561.574592016]: compiled against OGRE version 1.9.0 (Ghadamon)
> [ INFO] [1440539561.722585436]: Stereo is NOT SUPPORTED
> [ INFO] [1440539561.722696805]: OpenGl version: 3 (GLSL 1.3).
> [ERROR] [1440539562.762987357]: The STL file 'package://wild_thumper/meshes/wild_thumper_assembly_corrected.stl' is malformed. According to the binary STL header it should have '23240' triangles, but it has too much data for that to be the case.
> [ERROR] [1440539562.763706699]: Failed to load file [package://wild_thumper/meshes/wild_thumper_assembly_corrected.stl]

..and a few following warnings about the same file. The same stl worked with rviz 1.11.7.
Since the model is not mine, I can not attach it here. It is downloaded from [grabcad.com](https://grabcad.com/library/wild-thumper-correct-dimensions-1) and converted to stl with freecad.

wjwwood, 2015-08-26:
@caguero Any idea?

wjwwood, 2015-09-01:
@nxdefiant can you please post the STL version? I believe it will be ok for you to link it here as long as you attribute it to grabcad.com.

Failing that, can you share the STL with me privately? It's going to be very hard to reproduce your problem without the actual STL which fails or without more detailed instruction on how to create the STL.

Do you know if you converted it to an ascii STL or a binary STL using freecad?

nxdefiant, 2015-09-06:
Here it is: [wild_thumper_assembly_corrected.stl](http://vontaene.de/ros/wild_thumper_assembly_corrected.stl)

Should be binary stl, but I don't think I had the option to choose.

wjwwood, 2015-09-19:
@nxdefiant thanks a lot for the test file. After comparing it to our test files, I found that one of our new checks had a bug in it where the number of triangles was interpreted as a 16-bit unsigned integer instead of a 32-bit unsigned integer.

https://github.com/ros-visualization/rviz/pull/920 should fix this and adds a test case to cover it.

I'll try to get this fix out in a release at the beginning of next week.

nxdefiant, 2015-09-19:
Confirmed, thanks

closed 2015-09-19 by nxdefiant

Liangconghui, 2015-09-20:
hi, wjwwood.
 I encountered the same problem.
Could you let me know how to solve this problem.
My stl files can be read into riviz one week ago, after I did the update & upgrade, it is cann't be imported.
I tried to generate new stl files from Solidworks, but it is still failed.
Could you please tell more details about this problem or I need to wait .....?
Thanks!

CH

wjwwood, 2015-09-21:
@Liangconghui You need to either:

- Wait for the next version of `rviz` to make it to our public debian repositories and then update.
 - You can track its progress here: http://www.ros.org/debbuild/jade.html?q=rviz
- Switch to using the shadow-fixed repository which will contain the newest version of rviz (which I haven't released yet, but should this week).
- Finally, you can build `rviz` from source and include the patch from #920.

The last options is probably the best solution today.

nag92, 2015-09-23:
I am having the same issue. I am running ros- indigo. I ran a update and Rviz stop displaying my STL. When will this issue be resolved? is there something I can do in the mean time?

wjwwood, 2015-09-23:
@nag92 I explained the options in the previous comment: https://github.com/ros-visualization/rviz/issues/913#issuecomment-142046820

I've released `rviz` with this patch, but it will probably be a week or so before the deb is public.

enryan, 2015-09-30:
Is this problem resolved for  ros-hydro-rviz  1.10.20-0precise-20150806-0137-+0000 as well?

wjwwood, 2015-09-30:
@enryan No, sorry, Hydro is EOL and it wasn't an option to build new debs after this fix was merged. See:

http://www.ros.org/news/2015/09/ros-hydromedusa-final-release.html
https://github.com/ros/rosdistro/pull/9469

zfxw0206, 2015-10-01:
When I want to display my robot in Rviz, I got this error:

[ERROR] [1443724473.542191667]: The STL file 'package://robot_description/meshes/base_link.stl' is malformed. According to the binary STL header it should have '18016' triangles, but it has too much data for that to be the case.
[ERROR] [1443724473.542256768]: Failed to load file [package://robot_description/meshes/base_link.stl]
[ WARN] [1443724473.542533659]: OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource package://robot_description/meshes/base_link.stl in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/buildd/ogre-1.8-1.8.1+dfsg/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
[ERROR] [1443724473.542718763]: Could not load model 'package://robot_description/meshes/base_link.stl' for link 'base_link': OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource package://robot_description/meshes/base_link.stl in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/buildd/ogre-1.8-1.8.1+dfsg/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
[ WARN] [1443724473.542816771]: OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource package://robot_description/meshes/base_link.stl in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/buildd/ogre-1.8-1.8.1+dfsg/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
[ERROR] [1443724473.542883091]: Could not load model 'package://robot_description/meshes/base_link.stl' for link 'base_link': OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource package://robot_description/meshes/base_link.stl in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/buildd/ogre-1.8-1.8.1+dfsg/OgreMain/src/OgreResourceGroupManager.cpp (line 756)

Does anybody have any idea? Thank you!

wjwwood, 2015-10-02:
@zfxw0206 As I said in the previous comments in this thread, we've fixed the problem and release a new version with the fix, you just have to wait for the new version to get to the public repositories or build the latest version of rviz from source.

arthurnunesq, 2015-10-05:
@wjwwood, I'm facing the same problem as @enryan, I'm still using  ros-hydro-rviz 1.10.20-0precise-20150806-0137-+0000. Is it possible to downgrade only rviz back to 1.10.19? I tried doing it [using apt-get ](http://askubuntu.com/a/138327/430572), but it seems that only the latest version is available at packages.ros.org. If not, do you know what is the best option to fix the issue in ros-hydro setups?

wjwwood, 2015-10-06:
@arthurnunesq I guess downgrading is the best solution. I'll look into making a manual patch and rolling that out (without the build farm). But I can't promise that I can fix the version in Hydro.

arthurnunesq, 2015-10-06:
@wjwwood, Ok, thank you. Could you give me some clue on how to proper downgrade RVIZ?
- Do I need to build the whole ROS distribution and manually reinstall it?
- Is it possible to download and build just RVIZ? And then what? Should I uninstall only the current version of rviz using dpkg and then manually install the the one I've built? Does the build generates a *.deb file?
- What about RVIZ dependencies and related packages (ros-hydro-rqt-rviz, ros-hydro-rviz-plugin-tutorials, etc)? Do I need to downgrade them as well?

I'd appreciate any help or hints.

wjwwood, 2015-10-06:
It should be sufficient to build the latest version of rviz from the `hydro-devel` branch. Then you can run any program or launch file after sourcing the custom build of rviz and it should work. Something like:

```
$ git clone https://github.com/ros-visualization/rviz.git -b hydro-devel
$ mkdir -p rviz/build
$ cd rviz/build
$ source /opt/ros/indigo/setup.bash
$ cmake .. -DCMAKE_INSTALL_PREFIX=./install
$ make -j8 install
$ source ./install/setup.bash
$ # run or launch anything after sourcing this setup file
```

arthurnunesq, 2015-10-06:
@wjwwood I'll give it a try, thank you!

arthurnunesq, 2015-10-07:
@wjwwood I was able to build rviz (I had to install [ros-hydro-cmake-modules](http://answers.ros.org/question/141151/building-ros_control-on-hydro/?answer=203866#post-id-203866)). I sourced /rviz/build/install/setup.bash and then sourced my workspace "source /catkin_ws/devel/setup.bash". However, after launching rviz I still get 
``
[ERROR] [1444245812.567871657, 1444154608.371106045] [/rviz]: The STL file 'package://tt_description/meshes/g440.stl' is malformed. According to the binary STL header it should have '19292' triangles, but it has too much data for that to be the case.
[ERROR] [1444245812.567931269, 1444154608.371106045] [/rviz]: Failed to load file [package://tt_description/meshes/g440.stl]
``
Were the steps you suggested supposed to work when sourcing workspaces as well?

wjwwood, 2015-10-07:
@arthurnunesq did you get the `hydro-devel` branch?

I'll try to reproduce your problem in a VM.

arthurnunesq, 2015-10-07:
@wjwwood Yes, I did. Below the sequence of commands I've used to download and build rviz. Basically, I've performed the steps you suggested, I just source hydro instead of indigo
```
$ sudo apt-get install ros-hydro-cmake-modules
$ git clone https://github.com/ros-visualization/rviz.git -b hydro-devel
$ mkdir -p rviz/build
$ cd rviz/build
$ source /opt/ros/hydro/setup.bash
$ cmake .. -DCMAKE_INSTALL_PREFIX=./install
$ make -j8 install
```
Then, I tried to run a launch file from my development workspace. The launch file runs rviz, some rqt windows and playbacks a rosbag which loads the robot description that caused problems with the buggy rviz version.
```
$ source ~/rviz/build/install/setup.bash
$ source ~/catkin_ws/devel/setup.bash
$ roslaunch some_package_in_my_workspace some_launch_file.launch
```
After rviz is launched and the rosbag starts, the "STL file" error happens.

wjwwood, 2015-10-07:
That's the problem then, you cannot source two `setup.bash` files in a row and expect them to merge.  You need to source the `rviz` setup file and then build your `catkin_ws` or the reverse:

```
$ source ~/rviz/build/install/setup.bash
$ # rebuild ~/catkin_ws
$ source ~/catkin_ws/devel/setup.bash
$ roslaunch some_package_in_my_workspace some_launch_file.launch
```

Or

```
$ source ~/catkin_ws/devel/setup.bash
$ # rebuild ~/rviz
$ source ~/rviz/build/install/setup.bash
$ roslaunch some_package_in_my_workspace some_launch_file.launch
```

wjwwood, 2015-10-08:
One other option, which I try to avoid suggesting because it is sort of error prone, is that you can call the second `setup.bash` with the `--extend` option, which achieve what you're trying to accomplish.

arthurnunesq, 2015-10-09:
@wjwwood Your first suggestion didn't work, it kept using the buggy RVIZ version:
```
$ source ~/rviz/build/install/setup.bash
$ # rebuild ~/catkin_ws
$ source ~/catkin_ws/devel/setup.bash
$ roslaunch some_package_in_my_workspace some_launch_file.launch
```
The second suggestion didn't work, because ROS stopped recognizing the launch file in my workspace:
```
$ source ~/catkin_ws/devel/setup.bash
$ # rebuild ~/rviz
$ source ~/rviz/build/install/setup.bash
$ roslaunch some_package_in_my_workspace some_launch_file.launch
```
Calling `source` with the `--extend` option worked, which shows that I've downloaded the correct rviz version:
```
$ source ~/rviz/build/install/setup.bash
$ source ~/catkin_ws/devel/setup.bash --extended
$ roslaunch some_package_in_my_workspace some_launch_file.launch
```
However, as you said, [that is not recommended](http://answers.ros.org/question/64702/overlaying-multiple-catkin-devel-spaces-at-the-same-time/). The strange thing is that, from what I understood from  the answer on this link, your first suggestion should have worked. First, the "setup.bash" of the new rviz build is sourced and then my workspace is built, generating a new "setup.bash" that include the environment changes of the rviz's "setup.bash" . 
So, using the new workspace's "setup.bash" should now kind of source the rviz's one as well, isn't it? 
Do you know what might be happening?
Is there a way for me to build a deb package from this new rviz and install it over the buggy one, in order to avoid this "setup.bash" trouble?

wjwwood, 2015-10-09:
You may need to delete the build, devel, and install folders to get rid of stale state? I don't think that should be necessary, but that might work.

zweistein, 2015-10-13:
I have the same issue on Indigo, does anyone know when will be an official release of version without the  this bug? (and or without an other bug [issue 834](https://github.com/ros-visualization/rviz/issues/834))

wjwwood, 2015-10-13:
@zweistein I'm working on a release right now, it will be a week or two until the version with the unreleased fixes make it to our public repository.

As far as I know, https://github.com/ros-visualization/rviz/issues/834 has not be fixed.

arthurnunesq, 2015-10-13:
@wjwwood Deleting the devel and build directories solved the issue, thank you. 

Just to summarize, here are the procedures to install the patched rviz version on a ROS Hydro setup:

Download and rebuild rviz:
```
$ sudo apt-get install ros-hydro-cmake-modules
$ git clone https://github.com/ros-visualization/rviz.git -b hydro-devel
$ mkdir -p rviz/build
$ cd rviz/build
$ source /opt/ros/hydro/setup.bash
$ cmake .. -DCMAKE_INSTALL_PREFIX=./install
$ make -j8 install
```

Then source the new rviz setup and rebuild the ROS workspace. Make sure to delete the workspace's "build" and "devel" folders before rebuilding it.

```
$ source ~/rviz/build/install/setup.bash
$ # rebuild ~/catkin_ws
$ source ~/catkin_ws/devel/setup.bash
$ roslaunch some_package_in_my_workspace some_launch_file.launch
```

masasin, 2015-11-03:
I did exactly that. I still get the same error.

seanyun, 2015-12-02:
I have the same error with any Rviz newer than 1.11.7 (indigo)

wjwwood, 2015-12-02:
@seanyun this is fixed in `1.11.10`, please make sure you're using the latest version and then double check to see if you have the problem. If you do open a new issue with details about the error.

seanyun, 2015-12-02:
I still see the same error in 1.11.10.

[ERROR] [1449096828.185171155]: Could not load model 'package://####.stl'
for link '####_link': OGRE EXCEPTION(6:FileNotFoundException): Cannot
locate resource package://####.stl in resource group Autodetect or any
other group. in ResourceGroupManager::openResource at
/build/buildd/ogre-1.8-1.8.1+dfsg/OgreMain/src/OgreResourceGroupManager.cpp
(line 756)
[ WARN] [1449096828.185365021]: OGRE EXCEPTION(6:FileNotFoundException):
Cannot locate resource package://####_link.stl in resource group Autodetect
or any other group. in ResourceGroupManager::openResource at
/build/buildd/ogre-1.8-1.8.1+dfsg/OgreMain/src/OgreResourceGroupManager.cpp
(line 756)

My stl files are proprietary so I can't really post them, but this works
fine in 1.11.7.

thanks,
Seungkook

wjwwood, 2015-12-02:
@seanyun The error reported here looks like this:

```
The STL file 'package://wild_thumper/meshes/wild_thumper_assembly_corrected.stl' is malformed. According to the binary STL header it should have '23240' triangles, but it has too much data for that to be the case.
```

The error you just posted looks to be completely different. Please open a new issue.

seanyun, 2015-12-02:
Actually, I have that error too.
I'll make a post.

Seungkook

tslator, 2015-12-21:
Is there any information on when 1.11.10 will be available via ROS installation?

wjwwood, 2015-12-22:
@tslator `1.11.10` is available: http://www.ros.org/debbuild/indigo.html?q=rviz

awesomebytes, 2016-01-23:
I have the same problem than @seanyun with some STL files that work on Rviz on Hydro, now I just moved to Indigo and Rviz fails. On rviz 1.11.10

The error:
````
[ERROR] [1453560154.362913817]: The STL file 'package://XXXX/meshes/finger_flex_collision.stl' is malformed. According to the binary STL header it should have '108' triangles, but it has too much data for that to be the case.
[ERROR] [1453560154.362971254]: Failed to load file [package://XXXX/meshes/finger_flex_collision.stl]
[ WARN] [1453560154.363115588]: OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource package://XXXX/meshes/finger_flex_collision.stl in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/buildd/ogre-1.8-1.8.1+dfsg/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
````

I currently cannot give the file as an example, but just reporting it happens.

As a workaround I opened the stl files with meshlab and saved over them in stl again and Rviz was fine with that.

I've seen https://github.com/ros-visualization/rviz/commit/7c6cbcb8d133181c68e54e7338ae3cf204d08966 fixes it, I hope there is a new release coming with the fix soon :)

Thank you for your awesome work guys!

PR #920 (2015-09-19): fix reading of number of triangles in STL files
Fixes: #913