ros-visualization/rviz #916: Clickable image

jolting, 2015-09-10:
I think it would to be nice to be able to click on an image and do something useful with it similar to the way you can select a point with the point tool.

I was thinking something like this:
jolting/rviz@fc6922cc2316aecc7d04ffc9073bdd287fe9596c

wjwwood, 2015-10-12:
That's an interesting idea. If you have time I'd be happy to look over a pull request to the effect. It might be something worth exposing to interactive markers as well.

miguelriemoliveira, 2022-04-07:
Hello everyone,

I think I have a working solution for extending the image display with mouse clicking functionality. Would that be interesting? Should I do a pull request?

rhaschke, 2022-04-08:
Sure.

miguelriemoliveira, 2022-04-08:
OK, I will try and get back with the results.

miguelriemoliveira, 2022-04-11:
Hi @rhaschke, I have just created a [PR](https://github.com/ros-visualization/rviz/pull/1737) with the new functionality. 
Hope it can be integrated into rviz.