ros-visualization/rviz #917: crash when opening .rviz configurations

gitunit, 2015-09-11:
i am able to run rviz most of the time flawlessly but whenever i open some .rviz configs that i've not created i get a crash and the following output:

```
Qt has caught an exception thrown from an event handler. Throwing
exceptions from an event handler is not supported in Qt. You must
reimplement QApplication::notify() and catch all exceptions there.

terminate called after throwing an instance of 'Ogre::ItemIdentityException'
  what():  OGRE EXCEPTION(5:ItemIdentityException): Can not find codec for 'png' image format.
There are no formats supported (no codecs registered). in Codec::getCodec at /build/buildd/ogre-1.8-1.8.1+dfsg/OgreMain/src/OgreCodec.cpp (line 69)
```

i tried to reinstall rviz, but that didn't help.

gitunit, 2015-09-12:
fyi, this doesn't happen for the self compiled version. still there are some issues with textures, will update current version and try recompiling it.

wjwwood, 2015-09-14:
What system are you running on? It seems that your Ogre was not compiled with the same code options. Are you using any image or camera displays in the problem `.rviz` files? Need more information.

gitunit, 2015-09-14:
ubuntu 14.04. and ros indigo. i used displays from rtabmap_ros (build from source). as i found out today, the problem doesn't occur with the binaries from the indigo package. i already contacted the developer from rtabmap_ros, he will look into this issue soon. then we will know for sure what's the issue with the displays and if he get's the same problem. as far as i know more, i will post here.

wjwwood, 2015-09-14:
Ok, let me know if you have more info, or even better, a simple way to reproduce the issue. Thanks.

gitunit, 2015-09-23:
to reproduce the problem, you cant try: 

install https://github.com/introlab/rtabmap_ros for indigo (i did so from source).

then run: roslaunch rtabmap_ros stereo_oudoor.launch

the owner of this package had also a crash this morning but could resolve it by rebuilding catkin_ws but that didn't help in my case.

agilmor, 2017-01-20:
I had a similar issue and I've fixed it by rebuilding Ogre with Freeimage support.
I use Ogre and ROS from sources.

blakejwc, 2017-02-07:
@agilmor I'm trying to build Ogre for this https://github.com/jhu-lcsr/rviz_teleop_camera_plugin project, but when ever I open the image or camera displays, I get the Ogre error mentioned in this thread.

How did you build Ogre with Freeimage support?  I'm hoping that your steps might help with this stereo rviz project.

agilmor, 2017-02-08:
@blakejwc, I just did this in our Debian-Jessie:
```
$ sudo apt-get install libfreeimage-dev # this the key command! ;-)

$ hg clone https://bitbucket.org/sinbad/ogre/
$ cd ogre/
$ hg update v1-9
$ mkdir build && cd build
$ cmake .. -DCMAKE_INSTALL_PREFIX=/usr/local/opt/ogre-1.9
$ make && make install
```

Note that the `CMAKE_INSTALL_PREFIX` is not necessary at all.
And please note that when you run the `cmake` command you should see something like this:

```
-----------------------------------------------------------------------------
-- The following external packages were located on your system.
-- This installation will have the extra features provided by these packages.
+ zlib
+ freeimage
+ freetype
+ X11
+ Xt
+ Xaw
+ OpenGL
+ OpenGL 3+
+ OpenGL ES 1.x
+ OpenGL ES 2.x
+ OpenGL ES 3.x
+ POCO
+ Doxygen
+ TinyXML
+ CppUnit
-----------------------------------------------------------------------------
```

The `freeimage` should be listed here.

blakejwc, 2017-02-14:
@agilmor, Thank you for the steps.  I downloaded libfreeimage-dev (I even build it from source), and I do have `freeimage` in the cmake output.

Sadly, I'm still getting the error.

Here are the steps I did to build ogre and rviz with stereo (taken from [here](http://wiki.ros.org/rviz/Tutorials/Rviz%20in%20Stereo))
```
$ cd /path/to/stereo_rviz
$ hg clone https://bitbucket.org/sinbad/ogre ogre_version_for_rviz
$ cd ogre_version_for_rviz
$ hg update -C -r 6515
# I patched in the OGRE_CONFIG_ENABLE_QUAD_BUFFER_STEREO flag on this commit

$ cd /path/to/stereo_rviz/ogre_build;
$ cmake -DOGRE_FULL_RPATH=ON -DOGRE_CONFIG_ENABLE_QUAD_BUFFER_STEREO=ON -DCMAKE_INSTALL_PREFIX:PATH="/path/to/stereo_rviz/ogre_bin" -DOGRE_INSTALL_SAMPLES_SOURCE:BOOL="1" -DOGRE_INSTALL_SAMPLES:BOOL="1" -DOGRE_BUILD_SAMPLES:BOOL="1" -DFREETYPE_INCLUDE_DIR:PATH="/usr/include/freetype2" -DFREETYPE_FT2BUILD_INCLUDE_DIR:PATH="/usr/include/freetype2" ../ogre_version_for_rviz
$ make && make install

$ cd ../
$ PKG_CONFIG_PATH=/path/to/stereo_rviz/ogre_bin/lib/pkgconfig:${PKG_CONFIG_PATH} catkin_make --cmake-args -DCMAKE_SHARED_LINKER_FLAGS:STRING=" -Wl,-rpath,/path/to/stereo_rviz/ogre_bin/lib" -DCMAKE_CXX_FLAGS:STRING="-DOGRE_STEREO_ENABLE=1" -DUseQt5:BOOL="0"
```

I'm exactly sure where to go from hear, or how to debug this further.  Any help would be greatly appreciated.

agilmor, 2017-02-15:
@blakejwc, The freeimage fix is related to the error `Can not find codec for 'png' image format.`.
Are you really having this error? Or you Ogre exception is different?
If you are, are you really sure that `+ freeimage` is listed in the list of included packages, not in the list of packages that you could still add?

If you are, then I'm not sure how to help you here... sorry.

blakejwc, 2017-02-16:
The error I'm getting:
```
Error: [ WARN] [1486157423.903190776, 423.502999999]: OGRE EXCEPTION(5:ItemIdentityException): Can not find codec for 'png' image format.
There are no formats supported (no codecs registered). in Codec::getCodec at /path/to/stereo_rviz/ogre_version_for_rviz/OgreMain/src/OgreCodec.cpp (line 69)
Qt has caught an exception thrown from an event handler. Throwing
exceptions from an event handler is not supported in Qt. You must
reimplement QApplication::notify() and catch all exceptions there.

terminate called after throwing an instance of 'Ogre::ItemIdentityException'
  what(): OGRE EXCEPTION(5:ItemIdentityException): Can not find codec for 'png' image format.
There are no formats supported (no codecs registered). in Codec::getCodec at /path/to/stereo_rviz/ogre_version_for_rviz/OgreMain/src/OgreCodec.cpp (line 69)
Aborted (core dumped)
```

It's also weird because it says _no codecs registered_.

liuxin00738, 2018-03-29:
Hi blakejwc, do you still have the problem? I am having similar error. I have freeimage and ogre says it find it. Any thought will help!

blakejwc, 2018-04-17:
Sadly, I'm still stuck on this problem.  It's been on the back-burner for a while, but we ended up digging into it again a couple weeks ago.  I'll try and get you a summary of what we found, which was mostly just ambiguous build and dependency issues.  Seems like some people just tried all the steps on fresh 14.04 installs and had some success, but not always. :(

liuxin00738, 2018-04-17:
I got the rviz stereo working by:
using driver 375.39 with ubuntu14.04.5 with metacity flashback desktop.
My card is quad k2000, with an asus monitor (same as the one in the ros rivz tutorial).
The desktop is the step that make the stereo display works.

Also make sure you remove the old ogre lib. If you install moveit using apt-get, it will install an ogre library for you, which cause problem. What I do is I install moveit from source and this error disappear.

tanay-bits, 2018-04-17:
@liuxin00738 , are you able to run the rviz image_view plugin in stereo? Did you get rid of the png codec problem? Which versions of Ogre and rviz are you using?

liuxin00738, 2018-04-17:
I did not use that image_view plutin. I think the stereo only work for visualization markers. meshes,  etc.
Yes I get rid of the png codec problem, I am using Ogre and rviz (indigo version) from this tutorial.
http://wiki.ros.org/rviz/Tutorials/Rviz%20in%20Stereo