ros-visualization/rviz #918: interactive_marker_tutorials fail with "Error getting time of latest transform between /base_link and base_link"

rhaschke, 2015-09-14:
Starting any example from the [interactive_marker_tutorials](http://wiki.ros.org/interactive_marker_tutorials) fails with an error:

    Error getting time of latest transform between /base_link and base_link

Seemingly, [link name comparison doesn't ignore a leading slash anymore](https://github.com/ros-visualization/rviz/blob/indigo-devel/src/rviz/default_plugin/interactive_markers/interactive_marker.cpp#L310L311): All the examples in the tutorial register a marker to the frame "/base_link" while rviz removes the leading slash in the global fixed frame.
Removing the leading slash in the examples solves the issue.

However, eventually rviz should ignore the leading slash everywhere, not only in the fixed frame.

tested on ROS Indigo.

wjwwood, 2015-09-14:
I'd say this it more likely to be related to this line's behavior changing: https://github.com/ros-visualization/rviz/blob/indigo-devel/src/rviz/default_plugin/interactive_markers/interactive_marker.cpp#L321

@tfoote are you aware of any change in tf which might be the issue?

tfoote, 2015-09-14:
Not recently. Quite a while ago we stopped doing the tf_prefixing resolution which would have made `base_link` evaluate to `/base_link`. This would be as of 1.10 when we switch to tf2 under the hood in hydro.

wjwwood, 2015-09-14:
@tfoote What's the right thing to do here then? Should the tutorial be updated (with a note as to why) or should rviz strip the leading `/`? Why does setting the fixed frame in `rviz` to `map` work as well as `/map`?

tfoote, 2015-09-15:
I would update the tutorial to not prepend slashes. The problem is that some code paths normalize them and some don't. The internal tf2 representation does not use them. There is a backwards compatibility layer in the tf wrapper which strips them for you. I don't know the different code paths between fixed frame vs marker frames.

rhaschke, 2015-09-15:
Will be fixed in the tutorial with [#30](https://github.com/ros-visualization/visualization_tutorials/pull/30).
Same issue was fixed in python tutorial two months ago...

closed 2015-09-15 by rhaschke