ros-visualization/rviz #919: Uncouple rendering and rviz

arntanguy, 2015-09-18:
Hi,
Recently, I had need for a module that performs a real-time rendering of a mask of our robot
from its own sensor viewpoint, in order to be able to ignore the robot in embedded vision algorithm (SLAM).
As I couldn't find any readily available packages for this purpose, and had a limited time frame, I opted to do it on my own with the tools I already have. That's a lot of duplicate work, parsing URDF, creating a usable kinematic tree, and a basic renderer.

But that got me thinking, that RVIZ, having so many rendering abilities already ready for 
use in a robotics context could perfectly fit the bill, if only the rendering part and interface could 
be uncoupled into two separate libraries.

I had a quick look at the robot module, and I don't suppose it would be much work to uncouple at least this one. Not sure about the rest.

I'm just throwing the idea out there, but if there is support for it, I'm willing to chip in, and do it myself, in my limited free time.

Best regards,
Arnaud

wjwwood, 2015-09-21:
Just so I understand your objective, you want to take like camera data, joint information, and a model of the robot, render the robot over the image, blocking out the image where ever the robot is in the frame (presumably with a green screen or something like that)?

You're right, that this could be accomplished with rviz's camera display in combination with the robot display. I wasn't the original architect of rviz, so I'm not entirely sure what would be required to extract this capability, but each of the displays (including the robot display and the camera display) are fairly well encapsulated in the default plugin and in their own files:

- https://github.com/ros-visualization/rviz/blob/1.11.8/src/rviz/default_plugin/camera_display.h
- https://github.com/ros-visualization/rviz/blob/1.11.8/src/rviz/default_plugin/camera_display.cpp
- https://github.com/ros-visualization/rviz/blob/1.11.8/src/rviz/default_plugin/robot_model_display.h
- https://github.com/ros-visualization/rviz/blob/1.11.8/src/rviz/default_plugin/robot_model_display.cpp

I suppose that these components could be extracted, but you'd also need to extract the rendering elements, because you'd presumably want to run this headless. If not then you could conceivably write a display plugin which combines the robot display and the camera display, but which only overlays the robot display on the camera image (something that's not currently configurable) and which renders the robot model as a flat color and then republishes the newly rendered image. This means you'd have to be running rviz, but if that works for you then it might be easier than trying to refactor rviz to just render the image without running the ui.

arntanguy, 2015-09-25:
You understood the objective correctly.
As I now have my own working system, I'll stick to that, it would be too much trouble refactoring rviz, where everything is unfortunately tightly coupled together.

Thanks for your answer,
Best regards,

Arnaud

closed 2015-10-10 by arntanguy