ros-visualization/rviz #923: Map display not updated when tf changes

NikolausDemmel, 2015-10-01:
It seems that the map display does not update if the transformation from the fixed frame to the frame_id of the displayed map changes.

It does lookup this transformation correctly when the map message is received, but never updates the display thereafter.

Note that this does not play a role for map messages whose transformation to the fixed frame is constant. This might be the case often, but not always.

trainman419, 2015-10-02:
It looks like this may be fixed by https://github.com/ros-visualization/rviz/pull/915 , but I don't think that fix has made it to the public repos yet.

wjwwood, 2015-10-02:
I agree with @trainman419, this sounds like #915. If so the conclusion there may apply here as well, which is that you should always set your fixed frame to the highest root of your tree and just change the target frame to control what the camera is attached to.

NikolausDemmel, 2015-10-02:
> It looks like this may be fixed by #915 , but I don't think that fix has made it to the public repos yet.

Yes, this seems to be the same issue indeed.

@wjwwood: would this make it to the public repos at the next rviz release, or do we need to wait for rviz release *and* subsequent indigo "sync"? Do you have a rough timeframe for both? 

> If so the conclusion there may apply here as well, which is that you should always set your fixed frame to the highest root of your tree and just change the target frame to control what the camera is attached to.

In my case I have multiple gridmaps, not all of which are in the fix-frame. Even though the fixed-frame in rviz is set to "world" and "world" is the root of my TF tree, I still don't get updates for gridmaps that don't have frame_id "world" when the transform to "world" changes.

wjwwood, 2015-10-12:
> Do you have a rough timeframe for both?

Yeah, I'll do the rviz release in a few days (hopefully) and a sync should come a fews days to a week after that, as we are queueing up some releases of the core packages (`ros_comm` et al).

> In my case I have multiple gridmaps, not all of which are in the fix-frame. Even though the fixed-frame in rviz is set to "world" and "world" is the root of my TF tree, I still don't get updates for gridmaps that don't have frame_id "world" when the transform to "world" changes.

I'm not sure what would cause that behavior, but it may very well be related to this issue. It would be informative if you could try out this pr locally with your use case and report on that. My fear is that there will be quite some more work to be done if this is the root of your issue.

NikolausDemmel, 2015-10-13:
> I'm not sure what would cause that behavior, but it may very well be related to this issue. It would be informative if you could try out this pr locally with your use case and report on that. My fear is that there will be quite some more work to be done if this is the root of your issue.

I just built the devel version and the specific issue I was having seems to be fixed there.

wjwwood, 2015-10-13:
@NikolausDemmel can we close this then? I'm about to do a release.

closed 2015-10-15 by NikolausDemmel