ros-visualization/rviz #924: RVIZ Fails to Load Plugins

rhololkeolke, 2015-10-02:
I have just gone through a source installation using homebrew on Max OS X 10.10.5. Everything builds properly, however, when I start RVIZ it fails to find any plugins. I get the following error messages when starting:

~~~
[31m[ERROR] [1443822410.597903000]: The XML document "/Users/rhol/ros_catkin_ws/install_isolated/share/rviz_plugin_tutorials/plugin_description.xml" given to add must have either "library" or           "class_libraries" as the root tag[0m
[31m[ERROR] [1443822410.598457000]: The XML document "/Users/rhol/ros_catkin_ws/install_isolated/share/rviz/plugin_description.xml" given to add must have either "library" or           "class_libraries" as the root tag[0m
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
[31m[ERROR] [1443822412.203886000]: The XML document "/Users/rhol/ros_catkin_ws/install_isolated/share/rviz_plugin_tutorials/plugin_description.xml" given to add must have either "library" or           "class_libraries" as the root tag[0m
[31m[ERROR] [1443822412.204393000]: The XML document "/Users/rhol/ros_catkin_ws/install_isolated/share/rviz/plugin_description.xml" given to add must have either "library" or           "class_libraries" as the root tag[0m
[31m[ERROR] [1443822412.218883000]: The XML document "/Users/rhol/ros_catkin_ws/install_isolated/share/rviz_plugin_tutorials/plugin_description.xml" given to add must have either "library" or           "class_libraries" as the root tag[0m
[31m[ERROR] [1443822412.219405000]: The XML document "/Users/rhol/ros_catkin_ws/install_isolated/share/rviz/plugin_description.xml" given to add must have either "library" or           "class_libraries" as the root tag[0m
libpng warning: iCCP: known incorrect sRGB profile
[31m[ERROR] [1443822412.292074000]: The XML document "/Users/rhol/ros_catkin_ws/install_isolated/share/rviz_plugin_tutorials/plugin_description.xml" given to add must have either "library" or           "class_libraries" as the root tag[0m
[31m[ERROR] [1443822412.292619000]: The XML document "/Users/rhol/ros_catkin_ws/install_isolated/share/rviz/plugin_description.xml" given to add must have either "library" or           "class_libraries" as the root tag[0m
[31m[ERROR] [1443822412.292925000]: PluginlibFactory: The plugin for class 'rviz/Orbit' failed to load.  Error: According to the loaded plugin descriptions the class rviz/Orbit with base class type rviz::ViewController does not exist. Declared types are [0m
2015-10-02 17:50:10.669 rviz[89513:2608173] modalSession has been exited prematurely - check for a reentrant call to endModalSession:
[31m[ERROR] [1443822610.675354000]: PluginlibFactory: The plugin for class 'rviz/MoveCamera' failed to load.  Error: According to the loaded plugin descriptions the class rviz/MoveCamera with base class type rviz::Tool does not exist. Declared types are [0m
[rospack] Error: no package given
[librospack]: error while executing command
[rospack] Error: no package given
[librospack]: error while executing command
[31m[ERROR] [1443822610.742142000]: PluginlibFactory: The plugin for class 'rviz/Interact' failed to load.  Error: According to the loaded plugin descriptions the class rviz/Interact with base class type rviz::Tool does not exist. Declared types are [0m
[rospack] Error: no package given
[librospack]: error while executing command
[rospack] Error: no package given
[librospack]: error while executing command
[31m[ERROR] [1443822610.769387000]: PluginlibFactory: The plugin for class 'rviz/Select' failed to load.  Error: According to the loaded plugin descriptions the class rviz/Select with base class type rviz::Tool does not exist. Declared types are [0m
[rospack] Error: no package given
[librospack]: error while executing command
[rospack] Error: no package given
[librospack]: error while executing command
[31m[ERROR] [1443822610.795808000]: PluginlibFactory: The plugin for class 'rviz/SetInitialPose' failed to load.  Error: According to the loaded plugin descriptions the class rviz/SetInitialPose with base class type rviz::Tool does not exist. Declared types are [0m
[rospack] Error: no package given
[librospack]: error while executing command
[rospack] Error: no package given
[librospack]: error while executing command
[31m[ERROR] [1443822610.823075000]: PluginlibFactory: The plugin for class 'rviz/SetGoal' failed to load.  Error: According to the loaded plugin descriptions the class rviz/SetGoal with base class type rviz::Tool does not exist. Declared types are [0m
[rospack] Error: no package given
[librospack]: error while executing command
[rospack] Error: no package given
[librospack]: error while executing command
[31m[ERROR] [1443822610.852078000]: PluginlibFactory: The plugin for class 'rviz/Grid' failed to load.  Error: According to the loaded plugin descriptions the class rviz/Grid with base class type rviz::Display does not exist. Declared types are [0m
[31m[ERROR] [1443822610.866407000]: PluginlibFactory: The plugin for class 'rviz/Interact' failed to load.  Error: According to the loaded plugin descriptions the class rviz/Interact with base class type rviz::Tool does not exist. Declared types are [0m
[31m[ERROR] [1443822610.866600000]: PluginlibFactory: The plugin for class 'rviz/MoveCamera' failed to load.  Error: According to the loaded plugin descriptions the class rviz/MoveCamera with base class type rviz::Tool does not exist. Declared types are [0m
[31m[ERROR] [1443822610.866749000]: PluginlibFactory: The plugin for class 'rviz/Select' failed to load.  Error: According to the loaded plugin descriptions the class rviz/Select with base class type rviz::Tool does not exist. Declared types are [0m
[31m[ERROR] [1443822610.866882000]: PluginlibFactory: The plugin for class 'rviz/FocusCamera' failed to load.  Error: According to the loaded plugin descriptions the class rviz/FocusCamera with base class type rviz::Tool does not exist. Declared types are [0m
[rospack] Error: no package given
[librospack]: error while executing command
[rospack] Error: no package given
[librospack]: error while executing command
[31m[ERROR] [1443822610.893527000]: PluginlibFactory: The plugin for class 'rviz/Measure' failed to load.  Error: According to the loaded plugin descriptions the class rviz/Measure with base class type rviz::Tool does not exist. Declared types are [0m
[rospack] Error: no package given
[librospack]: error while executing command
[rospack] Error: no package given
[librospack]: error while executing command
[31m[ERROR] [1443822610.920758000]: PluginlibFactory: The plugin for class 'rviz/SetInitialPose' failed to load.  Error: According to the loaded plugin descriptions the class rviz/SetInitialPose with base class type rviz::Tool does not exist. Declared types are [0m
[31m[ERROR] [1443822610.920917000]: PluginlibFactory: The plugin for class 'rviz/SetGoal' failed to load.  Error: According to the loaded plugin descriptions the class rviz/SetGoal with base class type rviz::Tool does not exist. Declared types are [0m
[31m[ERROR] [1443822610.921054000]: PluginlibFactory: The plugin for class 'rviz/PublishPoint' failed to load.  Error: According to the loaded plugin descriptions the class rviz/PublishPoint with base class type rviz::Tool does not exist. Declared types are [0m
[rospack] Error: no package given
[librospack]: error while executing command
[rospack] Error: no package given
[librospack]: error while executing command
[31m[ERROR] [1443822610.948407000]: PluginlibFactory: The plugin for class 'rviz/Orbit' failed to load.  Error: According to the loaded plugin descriptions the class rviz/Orbit with base class type rviz::ViewController does not exist. Declared types are [0m
2015-10-02 17:50:11.519 rviz[89513:2608173] modalSession has been exited prematurely - check for a reentrant call to endModalSession:
libpng warning: iCCP: known incorrect sRGB profile
[0m[ INFO] [1443822410.574607000]: rviz version 1.11.9[0m
[0m[ INFO] [1443822410.574650000]: compiled against OGRE version 1.7.4 (Cthugha)[0m
[0m[ INFO] [1443822412.043594000]: Stereo is NOT SUPPORTED[0m
[0m[ INFO] [1443822412.043681000]: OpenGl version: 2.1 (GLSL 1.2).[0m
~~~

In addition to the attached terminal output a number of messages boxes appear. As an example here is what one of the message boxes says:

~~~
The class required for this view controller, 'rviz/Orbit', could not be loaded.
Error:
[... cut ...]
I've double checked and the path given in the log and it matches the path of the plugin_description.xml file. I've also double checked the XML and the root tag is indeed a library tag. I've run the plugin_description.xml file through an XML verifier to make sure the XML isn't malformed. The verifier finds nothing wrong with the XML.

I've also double checked the `install_isolated/lib` folder and `libdefault_plugin.dylib` is indeed there.

If you believe that this is an error with pluginlib let me know and I'll open an issue on that project.

wjwwood, 2015-10-12:
Yeah, not sure about that. Can you post the `plugin_description.xml` from `rviz` and the `rviz_plugin_tutorials` packages?

Also try the obvious things, like delete the build, devel and install folders and build against from scratch, and in a clean terminal.

rhololkeolke, 2015-10-20:
I've tried a clean build, but it still doens't work. I've attached the two `plugin_description.xml` files. I've checked the library paths and nothing seems out of the ordinary. The libs appear to be in the correct location.

Here is the rviz plugin_description.xml. I got this from the `install_isolated/share/rviz` folder.

```xml
<library path="libdefault_plugin">
  <class name="rviz/Axes" type="rviz::AxesDisplay" base_class_type="rviz::Display">
    <description>
      Displays an axis at the Target Frame's origin.  &lt;a href="http://www.ros.org/wiki/rviz/DisplayTypes/Axes"&gt;More Information&lt;/a&gt;.
    </description>
  </class>
  <class name="rviz/DepthCloud" type="rviz::DepthCloudDisplay" base_class_type="rviz::Display">
    <description>
      Displays point clouds based on depth maps.  &lt;a href="http://www.ros.org/wiki/rviz/DisplayTypes/DepthCloud"&gt;More Information&lt;/a&gt;.
    </description>
    <message_type>sensor_msgs/Image</message_type>
    <message_type>sensor_msgs/CompressedImage</message_type>
    <message_type>theora_image_transport/Packet</message_type>
  </class>
  <class name="rviz/Effort" type="rviz::EffortDisplay" base_class_type="rviz::Display">
    <description>
      Displays from sensor_msgs/JointState/effort messages.
    </description>
    <message_type>sensor_msgs/JointState</message_type>
  </class>
  <class name="rviz/FluidPressure" type="rviz::FluidPressureDisplay" base_class_type="rviz::Display">
    <description>
      Displays from sensor_msgs/FluidPressure messages.
    </description>
    <message_type>sensor_msgs/FluidPressure</message_type>
  </class>
  <class name="rviz/Grid" type="rviz::GridDisplay" base_class_type="rviz::Display">
    <description>
      Displays a grid along the ground plane, centered at the origin of the target frame of reference.  &lt;a href="http://www.ros.org/wiki/rviz/DisplayTypes/Grid"&gt;More Information&lt;/a&gt;.
    </description>
  </class>
  <class name="rviz/GridCells" type="rviz::GridCellsDisplay" base_class_type="rviz::Display">
    <description>
      Displays data from a nav_msgs::GridCells message as billboards.  &lt;a href="http://www.ros.org/wiki/rviz/DisplayTypes/GridCells"&gt;More Information&lt;/a&gt;.
    </description>
    <message_type>nav_msgs/GridCells</message_type>
  </class>
  <class name="rviz/Illuminance" type="rviz::IlluminanceDisplay" base_class_type="rviz::Display">
    <description>
      Displays illuminance values from a sensor_msgs/Illuminance topic.  &lt;a href="http://www.ros.org/wiki/rviz/DisplayTypes/Illuminance"&gt;More Information&lt;/a&gt;.
    </description>
    <message_type>sensor_msgs/Illuminance</message_type>
  </class>
  <class name="rviz/Image" type="rviz::ImageDisplay" base_class_type="rviz::Display">
    <description>
      Displays an image from a sensor_msgs/Image topic, similar to image_view.  &lt;a href="http://www.ros.org/wiki/rviz/DisplayTypes/Image"&gt;More Information&lt;/a&gt;.
    </description>
    <message_type>sensor_msgs/Image</message_type>
    <message_type>sensor_msgs/CompressedImage</message_type>
    <message_type>theora_image_transport/Packet</message_type>
  </class>
  <class name="rviz/InteractiveMarkers" type="rviz::InteractiveMarkerDisplay" base_class_type="rviz::Display">
    <description>
      Connects to one or more Interactive Marker servers, allowing you to interact with the provided 3D objects. &lt;a href="http://www.ros.org/wiki/rviz/DisplayTypes/InteractiveMarker"&gt;More Information&lt;/a&gt;.
    </description>
    <message_type>visualization_msgs/InteractiveMarkerUpdate</message_type>
  </class>
  <class name="rviz/LaserScan" type="rviz::LaserScanDisplay" base_class_type="rviz::Display">
    <description>
      Displays the data from a sensor_msgs::LaserScan message as points in the world, drawn as points, billboards, or cubes.  &lt;a href="http://www.ros.org/wiki/rviz/DisplayTypes/LaserScan"&gt;More Information&lt;/a&gt;.
    </description>
    <message_type>sensor_msgs/LaserScan</message_type>
  </class>
  <class name="rviz/Map" type="rviz::MapDisplay" base_class_type="rviz::Display">
    <description>
[... 213 lines cut ...]
<library path="lib/librviz_plugin_tutorials">
  <class name="rviz_plugin_tutorials/Teleop"
         type="rviz_plugin_tutorials::TeleopPanel"
         base_class_type="rviz::Panel">
    <description>
      A panel widget allowing simple diff-drive style robot base control.
    </description>
  </class>
  <class name="rviz_plugin_tutorials/Imu"
         type="rviz_plugin_tutorials::ImuDisplay"
         base_class_type="rviz::Display">
    <description>
      Displays direction and scale of accelerations from sensor_msgs/Imu messages.
    </description>
    <message_type>sensor_msgs/Imu</message_type>
  </class>
  <class name="rviz_plugin_tutorials/PlantFlag"
         type="rviz_plugin_tutorials::PlantFlagTool"
         base_class_type="rviz::Tool">
    <description>
      Tool for planting flags on the ground plane in rviz.
    </description>
  </class>
</library>
```

wjwwood, 2015-10-20:
I'm unable to reproduce this with my Mac book. There's nothing that would indicate to me what the issue might be. I'd suggest making sure you have the latest versions of the releases. Without being able to reproduce it or more information (I'm not sure what that information would be) then I probably cannot help further.

rhololkeolke, 2015-10-27:
I changed the logging level to Debug and added some extra debug statements. It appears as if there is something wrong with my TinyXML version. I'm not entirely sure what could be causing the issue as I installed TinyXML with homebrew and it says that everything is the latest version. I've already tried uninstalling and reinstalling with homebrew and it doesn't seem to be working.

The error happens in the 

```c++
template <class T> void ClassLoader<T>::processSingleXMLPluginFile(const std::string& xml_file, std::map<std::string, ClassDesc>& classes_available)
```
method inside the `pluginlib/include/pluginlib/class_loader_imp.h` file. I added some code to print the contents of the file specified in `xml_file`. It prints out the full XML file with no problems. So I know the correct paths are being provided and the file is readable as ASCII. However, I cannot get any valid data out of the `TiXmlDocument` class. I tried using the TiXmlPrinter class, but that gives a segfault. Printing things like the `config->ValueStr()` manually gives me gibberish. 

Just to make sure that this was not an RVIZ specific problem I created a test package that just loads the same XML file and and attempts to parse and print the output with TinyXML. Same results. I double checked the paths being found by the CMake Module and everything looks correct. At this point I'm just assuming there is some weird bug with TinyXML for my particular system. I no longer think this is an RVIZ problem.

Thanks for the help.

closed 2015-10-27 by rhololkeolke

wjwwood, 2015-10-27:
@rhololkeolke Ok, well that is strange. Please post a comment here if you find the solution. Even though it is not an rviz issue, other rviz users might run into it.

thezest76, 2019-09-15:
Hello,

Could you please tell me how to visualize "tensor filed"  ?
How to  show the current status of motion planning ?
I have Rviz on, but I cannot locate any relative topic to "motion field"
Thanks for the nice SW

Best regards

bergercookie, 2019-11-14:
FWIW, I stumbled upon the same issue.

I solved it by manually removing the `$HOME/.rviz` directory

kashifzr, 2024-04-03:
ERROR] [1712150616.305336725, 258.810000000]: PluginlibFactory: The plugin for class 'rviz_plugins/AerialMapDisplay' failed to load.  Error: According to the loaded plugin descriptions the class rviz_plugins/AerialMapDisplay with base class type rviz::Display does not exist. Declared types are  rviz/AccelStamped rviz/Axes rviz/Camera rviz/DepthCloud rviz/Effort rviz/FluidPressure rviz/Grid rviz/GridCells rviz/Illuminance rviz/Image rviz/InteractiveMarkers rviz/LaserScan rviz/Map rviz/Marker rviz/MarkerArray rviz/Odometry rviz/Path rviz/PointCloud rviz/PointCloud2 rviz/PointStamped rviz/Polygon rviz/Pose rviz/PoseArray rviz/PoseWithCovariance rviz/Range rviz/RelativeHumidity rviz/RobotModel rviz/TF rviz/Temperature rviz/TwistStamped rviz/WrenchStamped rviz_plugin_tutorials/Imu

@bergercookie How can be remove this error, I know you did remove ./rviz directory manually from $HOME

rhaschke, 2024-04-03:
@kashifzr: you are missing a single plugin, while this issue is about missing all plugins.
In your case, you probably simply missed installing the corresponding plugin package.