ros-visualization/rviz #929: rviz qt window panes ("Displays", "Tools", "Views", ...) don't update

NikolausDemmel, 2015-10-13:
It recently quite frequently happens to me that the QT GUI elements in rviz ("Displays", "Tools", "Views", ...) suddenly stop updating. I.e. I can still click (blindly) and e.g. activate / deactivate displays, or hide the panes, but the pane is never redrawn. If I then for example hide the pane and show it again, it just contains a jumbled part of the 3D render window.

My current workaround is to save the config and restart rviz, but that is quite inconvenient.

This is on Ubuntu 14.04.3, ROS indigo, but I remember having this issue every now and again for a very long time (years). Until now I hasn't really bothered me too much. I couldn't tell, but it could be that this was always on thinkpad laptops w/ Nvidia Optimus card. My current machine where I observe this is a T440p which I'm running on Nvidia-only for graphics (i.e. no bumblebee switching or nvidia-prime stuff).

I can't yet pinpoint what user action triggers this and cannot deterministically reproduce, but during normal usage it happens frequently. I have a feeling it happens more frequently when a "robot display" is involved, in my case displaying an STL model, but I can't say for sure.

Has anyone else had such problems? Any idea what causes it and how to fix it?

wjwwood, 2015-10-13:
I haven't see this, but I also don't spend all day using rviz. It would be interesting to see if others have experienced this issue.

trainman419, 2015-10-13:
I've used rviz quite a bit on Ubuntu 14.04 with the Nvidia graphics drivers, and I haven't seen this issue.

NikolausDemmel, 2015-10-17:
Thanks for the answers.

This might actually be some variant of [this nvidia / compiz / ... bug](https://bugs.launchpad.net/ubuntu/+source/compiz/+bug/1072206) instead. I recently noticed also instances where the complete rviz window is completely transparent on startup, i.e. I see just a window frame and the Desktop (the application loads fine; for example the mouse courser correctly changes when you blindly hover over the gui elements in the transparent window).

Moreover, I have noticed similar rendering issues with different opengl / Qt applications (for now just cases where they are white or transparent on startup). I am not seeing this with regular windows, as many describe in the above bug report, but then again I am not running Unity (but rather gnome-session-flashback).

Also, I have recently only seen this on my machine (Thinkpad with Optimus Nvidia card) and a collegues (same model).

closed 2015-10-17 by NikolausDemmel