ros-visualization/rviz #935: Possible issue with interactive marker control with local (moving) coordinate frame

jjnak, 2015-10-21:
Hi,

When using interactive markers with MOVE_3D and MOVE_ROTATE_3D option, I found that when the marker is clicked, the marker jumps. This happens when the marker is attached to a local coordinate frame (specified header.frame_id in InteractiveMarker) which is different from Fixed Frame specified in rviz. The amount of jump is related to the difference between the fixed frame and the local coordinates.

My guess is that this is possibly handled in moveViewPlane() in interactive_maker_control.cpp in default_plugin in rviz. With the MOVE_PLANE option, this jump does not occur. Also, when the ring and arrow are clicked with ROTATE_AXIS and MOVE_AXIS option, the jump does not happen. 

I would most appreciate it if you could investigate this issue and provide a possible solution to it. I'm using ROS Hydro on Ubuntu 12.04. 

The same issue has been posted in the ROS forum 
http://answers.ros.org/question/138616/rviz-tf-and-interactivemarker-pose-with-respect-to-which-frame/

Best regards,

-Jun

jjnak, 2015-10-21:
Hi,

Also, this could be related to beginMouseMovement() in handleMouseEvent(). So, it would be great if you could look into the coordinate transform between the current frame and the fixed frame.

Thank you,

-Jun

jjnak, 2015-10-22:
Hi,

I further looked into this issue, and seemingly setting 
parent_to_grab_position_  = Ogre::Vector3(0,0,0);

solved this problem and looks like it is working fine. So, either by setting parent_to_grab_position_ to zero, or changing the following two functions as follows might be a possible solution (removing  parent_to_grab_position_ from these).

 // find plane on which mouse is moving
    Ogre::Plane plane( event.viewport->getCamera()->getRealDirection(),
                       parent_position_at_mouse_down_);

    parent_->setPose( mouse_position_on_plane,
                      parent_->getOrientation(),
                      name_ );

Your confirmation would be much appreciated!

Best regards,
-Jun

jjnak, 2015-10-23:
Hi,

Further to my previous post, modification of moveViewPlane() in the following way achieved the behavior that I had expected:

    Ogre::Vector3 camera_direction_ = event.viewport->getCamera()->getRealDirection();
    Ogre::Plane plane( reference_node_->getOrientation().Inverse()*camera_direction_,
                       parent_position_at_mouse_down_);

and

    parent_->setPose( mouse_position_on_plane ,
                      parent_->getOrientation(),
                      name_ );

Best regards,

-Jun

wjwwood, 2016-03-22:
Crossreference: https://github.com/ros-visualization/rviz/pull/937#issuecomment-200076436

rhaschke, 2016-03-23:
@jjnak Could you please check, that this issue is resolved by #961. You describe exactly the same behaviour I observed too.