ros-visualization/rviz #939: rviz 1.11.10 has compile erros

youngbo-shim, 2015-10-29:
I want to have the new version rviz (1.11.10) but I could not do compile.. 
I encountered this problem..
Could you let me know how to solve this problem?

[image]

wjwwood, 2015-10-29:
Please put the error, in its entirety, in the issue as plain text. I cannot search or copy text in a screenshot and neither can a search engine. I'll reopen once you've provided the error as text.

youngbo-shim, 2015-10-29:
[image]

[ 58%] Built target rviz
Linking CXX executable ../../devel/lib/rviz/rviz
//usr/lib/libfreeimage.so.3: undefined reference to `TIFFTileRowSize@LIBTIFF_4.0'
//usr/lib/libfreeimage.so.3: undefined reference to `TIFFReadDirectory@LIBTIFF_4.0'
//usr/lib/libfreeimage.so.3: undefined reference to `TIFFCurrentDirectory@LIBTIFF_4.0'
//usr/lib/libfreeimage.so.3: undefined reference to `TIFFLastDirectory@LIBTIFF_4.0'
//usr/lib/libfreeimage.so.3: undefined reference to `TIFFSetMode@LIBTIFF_4.0'
//usr/lib/libfreeimage.so.3: undefined reference to `TIFFIsTiled@LIBTIFF_4.0'
//usr/lib/libfreeimage.so.3: undefined reference to `TIFFIsByteSwapped@LIBTIFF_4.0'
/opt/ros/indigo/lib/libresource_retriever.so: undefined reference to `curl_easy_setopt@CURL_OPENSSL_3'
//usr/lib/libfreeimage.so.3: undefined reference to `TIFFDefaultStripSize@LIBTIFF_4.0'
//usr/lib/libfreeimage.so.3: undefined reference to `TIFFComputeStrip@LIBTIFF_4.0'
/opt/ros/indigo/lib/libresource_retriever.so: undefined reference to `curl_easy_cleanup@CURL_OPENSSL_3'
//usr/lib/libfreeimage.so.3: undefined reference to `TIFFReadTile@LIBTIFF_4.0'
//usr/lib/libfreeimage.so.3: undefined reference to `TIFFWriteScanline@LIBTIFF_4.0'
/opt/ros/indigo/lib/libresource_retriever.so: undefined reference to `curl_easy_init@CURL_OPENSSL_3'
//usr/lib/libfreeimage.so.3: undefined reference to `TIFFGetField@LIBTIFF_4.0'
//usr/lib/libfreeimage.so.3: undefined reference to `TIFFScanlineSize@LIBTIFF_4.0'
//usr/lib/libfreeimage.so.3: undefined reference to `TIFFFieldWithTag@LIBTIFF_4.0'
//usr/lib/libfreeimage.so.3: undefined reference to `TIFFTileSize@LIBTIFF_4.0'
//usr/lib/libfreeimage.so.3: undefined reference to `TIFFWriteDirectory@LIBTIFF_4.0'
/opt/ros/indigo/lib/libresource_retriever.so: undefined reference to `curl_global_cleanup@CURL_OPENSSL_3'
//usr/lib/libfreeimage.so.3: undefined reference to `TIFFGetTagListEntry@LIBTIFF_4.0'
//usr/lib/libfreeimage.so.3: undefined reference to `TIFFClose@LIBTIFF_4.0'
//usr/lib/libfreeimage.so.3: undefined reference to `TIFFClientOpen@LIBTIFF_4.0'
//usr/lib/libfreeimage.so.3: undefined reference to `TIFFSetTagExtender@LIBTIFF_4.0'
//usr/lib/libfreeimage.so.3: undefined reference to `TIFFGetTagListCount@LIBTIFF_4.0'
//usr/lib/libfreeimage.so.3: undefined reference to `TIFFGetFieldDefaulted@LIBTIFF_4.0'
//usr/lib/libfreeimage.so.3: undefined reference to `TIFFSwabArrayOfLong@LIBTIFF_4.0'
//usr/lib/libfreeimage.so.3: undefined reference to `TIFFSetSubDirectory@LIBTIFF_4.0'
//usr/lib/libfreeimage.so.3: undefined reference to `TIFFStripSize@LIBTIFF_4.0'
/opt/ros/indigo/lib/libresource_retriever.so: undefined reference to `curl_easy_perform@CURL_OPENSSL_3'
//usr/lib/libfreeimage.so.3: undefined reference to `_TIFFDataSize@LIBTIFF_4.0'
//usr/lib/libfreeimage.so.3: undefined reference to `TIFFReadRGBAImage@LIBTIFF_4.0'
//usr/lib/libfreeimage.so.3: undefined reference to `TIFFReadEncodedStrip@LIBTIFF_4.0'
//usr/lib/libfreeimage.so.3: undefined reference to `TIFFSetField@LIBTIFF_4.0'
/opt/ros/indigo/lib/libresource_retriever.so: undefined reference to `curl_global_init@CURL_OPENSSL_3'
//usr/lib/libfreeimage.so.3: undefined reference to `TIFFMergeFieldInfo@LIBTIFF_4.0'
//usr/lib/libfreeimage.so.3: undefined reference to `TIFFDataWidth@LIBTIFF_4.0'
//usr/lib/libfreeimage.so.3: undefined reference to `TIFFSetDirectory@LIBTIFF_4.0'
//usr/lib/libfreeimage.so.3: undefined reference to `TIFFReadEXIFDirectory@LIBTIFF_4.0'
collect2: error: ld returned 1 exit status
make[2]: *** [devel/lib/rviz/rviz] Error 1
make[1]: *** [src/rviz/CMakeFiles/executable.dir/all] Error 2
make: *** [all] Error 2

closed 2015-11-02 by youngbo-shim

wjwwood, 2015-11-02:
It looks like it is not linking against the right version of the tiff library? What operating system are you building on? I'm unable to reproduce on my Ubuntu 14.04 machine.

johnson055, 2017-11-13:
if you're using conda, try this, it worked for me
conda uninstall libtiff