ros-visualization/rviz #945: rviz 1.11.8+ cannot load STL

seanyun, 2015-12-02:
The stl files which work fine with rviz 1.11.7 are not working with 1.11.8+.
Here's the error from 1.11.10 (same as any version 1.11.8+):

[ERROR] [1449097635.190619726]: The STL file 'package://####_description/meshes/####_link.stl' is malformed. According to the binary STL header it should have '208' triangles, but it has too much data for that to be the case.
[ERROR] [1449097635.190681335]: Failed to load file [package://####_description/meshes/####_link.stl]
[ WARN] [1449097635.190821463]: OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource package://####_description/meshes/####_link.stl in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/buildd/ogre-1.8-1.8.1+dfsg/OgreMain/src/OgreResourceGroupManager.cpp (line 756)
[ERROR] [1449097635.191032928]: Could not load model 'package://####_description/meshes/####_link.stl' for link '####_link': OGRE EXCEPTION(6:FileNotFoundException): Cannot locate resource package://####_description/meshes/base_link.stl in resource group Autodetect or any other group. in ResourceGroupManager::openResource at /build/buildd/ogre-1.8-1.8.1+dfsg/OgreMain/src/OgreResourceGroupManager.cpp (line 756)

I commented out the real names because the stl files are proprietary. So Unfortunately I can't post the files.

wjwwood, 2015-12-02:
@seanyun would it be possible for you to share one of these STL files with me confidentially? That might help me understand why this is happening.

seanyun, 2015-12-03:
I'm sorry that I can't send the file now. I'll talk about it with my company and let you know.
Meanwhile, I played with the rviz source file, and it seems the difference of the sizes is just '1'.

[ERROR] [1449100802.924760814]: The STL file 'package://####.stl' is malformed. num triangles: 208  binary_stl_header_len: 84 num_bytes: 10485 expected_size: 10484
[ERROR] [1449100802.951429978]: The STL file 'package://####.stl' is malformed. num triangles: 1896  binary_stl_header_len: 84 num_bytes: 94885 expected_size: 94884
[ERROR] [1449100802.980868243]: The STL file 'package://####.stl' is malformed. num triangles: 1896  binary_stl_header_len: 84 num_bytes: 94885 expected_size: 94884

seanyun, 2015-12-03:
When i commented out the 'return false' in the source code, Rviz runs well and displays the meshes correctly.

  if (num_bytes != expected_size)
  {
    ROS_WARN_STREAM("The STL file '" << origin << "' is malformed. According "
                     "to the binary STL header it should have '" <<
                     num_triangles << "' triangles, but it has too " <<
                     (num_bytes > expected_size ? "much" : "little") <<
                     " data for that to be the case.");
    //return false;
  }

caguero, 2015-12-09:
@seanyun According to the [binary STL spec](https://en.wikipedia.org/wiki/STL_(file_format)), a STL with 208 triangles should have a size of 2996 bytes:
```
             80  Header
              4  Num triangles
208 * 50 = 2912  Triangles
---------------------
          10484 bytes
```
Your mesh file seems to have 10485 bytes, so strictly speaking seems malformed. Could be the case that there's an extra byte at the end of the file?

caguero, 2015-12-10:
@seanyun See pull request https://github.com/ros-visualization/rviz/pull/951. We have relaxed the conditions and if a file has a size bigger than expected it will print a warning but rviz will still try to load it.

wjwwood, 2016-03-22:
I believe this to be fixed as of #951. Please comment here if you disagree.

closed 2016-03-22 by wjwwood