ros-visualization/rviz #953: Persist which marker namespaces were visible

stuglaser, 2015-12-12:
In the Marker display type, you can click checkboxes to show or hide each namespace.  If the time resets then this selection is lost and all marker namespaces become visible again.

It would be great if the selection of which are visible persisted when the time was reset.  Even better if it persisted when rviz was restarted.

wjwwood, 2016-03-22:
I believe this was fixed in https://github.com/ros-visualization/rviz/pull/962. I'm going to close, please comment if you think it is still an issue.

stuglaser, 2016-03-22:
It's nearly fixed.  When you save the config and restart rviz the active namespaces persist.  However, if time jumps backwards or if you click rviz's `Reset` button, then the active namespaces are reset.

cc @bponsler

wjwwood, 2016-03-22:
I see, I'll reopen this for now then. Thanks.

bponsler, 2016-04-07:
I just created PR #988 to keep the marker display plugin from erasing the map of enabled namespaces when the reset button is pressed. I did the same for the tf plugin. Both plugins were also updated to update the map of enabled frames as the frames are enabled/disabled using the UI check boxes so that the reset button does not return to the original state of the configuration file but instead keeps whatever namespaces/frames are enabled when the reset happens.

I think these changes will solve this issue.

wjwwood, 2016-04-07:
Closed by #988

closed 2016-04-07 by wjwwood