ros-visualization/rviz #954: Undefined reference YAML::Emitter::Write

SquirrelCZE, 2015-12-21:
Hi folks, I'have been trying to compile ros-jade-desktop and this package... but run into a problem during compilation

[ 56%] Linking CXX executable ../../devel/lib/rviz/rviz
../../devel/lib/librviz.so: undefined reference to `YAML::Emitter::Write(std::string const&)'

any ideas what I can done to get rid of this? I try to compile it on archlinux system with actual yaml-cpp library and actual ros-jade-rviz... 

thanks

P.S: if you would need any more info, I am more than willing to give it :)

wjwwood, 2015-12-21:
What version of `yaml-cpp` are you using?

SquirrelCZE, 2015-12-21:
yaml-cpp 0.5.2-5

wjwwood, 2015-12-22:
The fact that it is a linker error is interesting. It implies that the header that it compiled against no longer matches the shared library it is linking against. Did you update the version of `yaml-cpp` between the first build of `rviz` and a subsequent build? Without being able to reproduce the issue I'm just guessing.

SquirrelCZE, 2015-12-24:
I just reinstalled it...

Iv've got problems with another packaged and now even "catkin_make" doesn't work and got errors ... all linking errors it seems

I wil post question in anwsers.ros.org :/

SquirrelCZE, 2015-12-25:
Well, it seems like removing all ros-jade packages and relative packages from system and reinstall it all worked to repair the problem. I can close this here then I suppose

closed 2015-12-25 by SquirrelCZE