ros-visualization/rviz #955: rviz Segmentation Fault on Ubuntu ARM due to OpenSSL

dchang0, 2015-12-23:
A few of us ODROID ARM users are having trouble running rviz (Jade or Indigo) on the official ODROID Ubuntu 14.04 OS image.

It looks like this at first glance:
root@odroid:~# rosrun rviz rviz
Segmentation fault

root@odroid:~# rosrun rviz rviz -l
Segmentation fault

Notice that it never gets far enough to show any rviz console output.

GDB output:

root@odroid:/var/log# gdb /opt/ros/jade/bin/rviz
GNU gdb (Ubuntu 7.7.1-0ubuntu5~14.04.2) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law. Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /opt/ros/jade/bin/rviz...(no debugging symbols found)...done.
(gdb) r
Starting program: /opt/ros/jade/bin/rviz 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
Cannot access memory at address 0x0

Program received signal SIGILL, Illegal instruction.
0xb4d60208 in ?? () from /lib/arm-linux-gnueabihf/libcrypto.so.1.0.0
(gdb) 

It seems that the problem is actually in OpenSSL. So I did some Googling and found that this is a normal "feature" of OpenSSL testing the hardware for the presence of features. This bug report is from the Raspberry Pi:

https://bugs.launchpad.net/raspbian/+bug/1154042

The important quote to note is from peter green:

"Afaict openssl probes the capabilities of the user's CPU by trying to do things and trapping the illegal instruction errors. So a couple of sigills during startup is normal. When using a debugger in order to find the real failure in your application you must continue past the startup sigills."

So I go and test this on the ODROID with GDB:

root@odroid:/var/log# gdb openssl
GNU gdb (Ubuntu 7.7.1-0ubuntu5~14.04.2) 7.7.1
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from openssl...(no debugging symbols found)...done.
(gdb) r
Starting program: /usr/bin/openssl 
Cannot access memory at address 0x0

Program received signal SIGILL, Illegal instruction.
0xb6ea9208 in ?? () from /lib/arm-linux-gnueabihf/libcrypto.so.1.0.0
(gdb) 

Sure enough, the problem is that OpenSSL is producing a SIGILL.

Is it possible to modify RVIZ so that it can tolerate OpenSSL's use of SIGILLs to report that it has not found certain hardware features?

Thanks,
D.

wjwwood, 2015-12-24:
I'm not sure this is something rviz should try to handle, have you tried setting `OPENSSL_armcap=0` as described in this SO answer: http://stackoverflow.com/a/16398622/671658

It seems to me that openssl is (incorrectly?) assuming it can catch SIGILL but cannot for some reason. If you have a particular idea of how rviz could handle this (with a SIGILL handler or something), please propose a pull request.

Thanks for the detailed bug report.

dchang0, 2015-12-24:
Hmm. Good point--is this a "bug" or a "feature" in OpenSSL? Or, perhaps a different way of asking the same question: if I submitted this as a bug to the OpenSSL team, would they close it as a "Won't Fix?"

BTW, thanks for the tip to try setting 'OPENSSL_armcap=0'  It may provide the answer to the question above.

I tried this:

`root@odroid:/usr/bin# export OPENSSL_armcap=0`
'root@odroid:/usr/bin# gdb openssl'
'GNU gdb (Ubuntu 7.7.1-0ubuntu5~14.04.2) 7.7.1'
'Copyright (C) 2014 Free Software Foundation, Inc.'
'License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>'
'This is free software: you are free to change and redistribute it.'
'There is NO WARRANTY, to the extent permitted by law.  Type "show copying"'
'and "show warranty" for details.'
'This GDB was configured as "arm-linux-gnueabihf".'
'Type "show configuration" for configuration details.'
'For bug reporting instructions, please see:'
'<http://www.gnu.org/software/gdb/bugs/>.'
'Find the GDB manual and other documentation resources online at:'
'<http://www.gnu.org/software/gdb/documentation/>.'
'For help, type "help".'
'Type "apropos word" to search for commands related to "word"...'
'Reading symbols from openssl...(no debugging symbols found)...done.'
'(gdb) r'
'Starting program: /usr/bin/openssl '
'OpenSSL> '

Then I tried running RVIZ again:

'root@odroid:/usr/bin# gdb /opt/ros/jade/bin/rviz'
'GNU gdb (Ubuntu 7.7.1-0ubuntu5~14.04.2) 7.7.1'
'Copyright (C) 2014 Free Software Foundation, Inc.'
'License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>'
'This is free software: you are free to change and redistribute it.'
'There is NO WARRANTY, to the extent permitted by law.  Type "show copying"'
'and "show warranty" for details.'
'This GDB was configured as "arm-linux-gnueabihf".'
'Type "show configuration" for configuration details.'
'For bug reporting instructions, please see:'
'<http://www.gnu.org/software/gdb/bugs/>.'
'Find the GDB manual and other documentation resources online at:'
'<http://www.gnu.org/software/gdb/documentation/>.'
'For help, type "help".'
'Type "apropos word" to search for commands related to "word"...'
'Reading symbols from /opt/ros/jade/bin/rviz...(no debugging symbols found)...done.'
'(gdb) r'
'Starting program: /opt/ros/jade/bin/rviz '
'[Thread debugging using libthread_db enabled]'
'Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".'

'Program received signal SIGSEGV, Segmentation fault.'
'0xb4ededc4 in ?? () from /lib/arm-linux-gnueabihf/libpcre.so.3'
'(gdb) '

So it does seem that `OPENSSL_armcap=0` does need to be set but that some other problem arises afterwards in RVIZ that needs to be surpassed. I'll chase that down later and post back if I find anything...

Does the responsibility of setting `OPENSSL_armcap=0` get placed on ROS? It seems that this is more of a "feature" since OpenSSL gave the world a way to turn it off. Would it make sense for ROS to do the platform detection and set that if ARM is detected?

wjwwood, 2015-12-24:
I'm not sure if it should be set automatically by rviz's setup file on armhf (or what ever the affected architectures are), or if it should be documented. Unfortunately I don't think rviz has a chance to catch this and tell the user what the problem is and how to fix it.

The new segfault you get sounds familiar: https://github.com/ros/robot_model/issues/110

I'm not sure if that ever got resolved or worked around.

dchang0, 2015-12-24:
I'm happy with mere documentation of the OpenSSL env. variable. At least that way there won't be people out there eager to use RVIZ on ARM that got turned off/turned away thinking it's buggy. (I found at least four posters out there who encountered this exact problem and spoke up about it--how many more encountered it but said nothing we can only guess at.)

Thanks for the tip to #110. I'll look into it further.

trainman419, 2015-12-24:
This is a red herring. Rviz and openssl on ARM correctly catches and handles this exception. It only shows up during debugging because the debugger is set up to also hanlde this exception.

This IS NOT a segmentation fault (SIGSEGV).

Simply type `continue` when debugging and you see a SIGILL from openssl.

wjwwood, 2015-12-24:
I think @trainman419 is right, this probably only happens when you're in the debugger. That should have been obvious to me since you get a segfault, but then a SIGILL when run in the debugger. The original problem is the segfault in `libpcre`.

dchang0, 2015-12-24:
Okay, let's give it a try. I'll unset the env. variable and run continue...

dchang0, 2015-12-24:
Yup, @trainman419 is right:

`root@odroid:/usr/bin# unset OPENSSL_armcap`
`root@odroid:/usr/bin# echo $OPENSSL_armcap`

`root@odroid:/usr/bin# gdb /opt/ros/jade/bin/rviz`
`GNU gdb (Ubuntu 7.7.1-0ubuntu5~14.04.2) 7.7.1`
`Copyright (C) 2014 Free Software Foundation, Inc.`
`License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>`
`This is free software: you are free to change and redistribute it.`
`There is NO WARRANTY, to the extent permitted by law.  Type "show copying"`
`and "show warranty" for details.`
`This GDB was configured as "arm-linux-gnueabihf".`
`Type "show configuration" for configuration details.`
`For bug reporting instructions, please see:`
`<http://www.gnu.org/software/gdb/bugs/>.`
`Find the GDB manual and other documentation resources online at:`
`<http://www.gnu.org/software/gdb/documentation/>.`
`For help, type "help".`
`Type "apropos word" to search for commands related to "word"...`
`Reading symbols from /opt/ros/jade/bin/rviz...(no debugging symbols found)...done.`
`(gdb) r`
`Starting program: /opt/ros/jade/bin/rviz `
`[Thread debugging using libthread_db enabled]`
`Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".`
`Cannot access memory at address 0x0`

`Program received signal SIGILL, Illegal instruction.`
`0xb4d60208 in ?? () from /lib/arm-linux-gnueabihf/libcrypto.so.1.0.0`
`(gdb) continue`
`Continuing.`
``
`Program received signal SIGSEGV, Segmentation fault.`
`0xb4ededc4 in ?? () from /lib/arm-linux-gnueabihf/libpcre.so.3`
`(gdb) `

closed 2015-12-24 by dchang0

dchang0, 2015-12-24:
@wjwwood:

I followed the solution in the #110 bug that you referenced and confirm that it did indeed fix my problem.

What I did was install libpcre3 8.35-7 from the Ubuntu armhf package repository:

http://ports.ubuntu.com/pool/main/p/pcre3/libpcre3_8.35-7.1ubuntu1_armhf.deb

It is probably not preferable to upgrading collada-dom to 2.4.4 as was indicated in #110, but I did not find a ready-made .deb package for collada-dom 2.4.4 for armhf--only i386 and amd64.

With that, this "bug" is really and truly fixed. Hope that helps someone else...

wmorlan3, 2016-03-18:
I have upgraded to libpcre3 8.35 on Debian, and am still having this problem.  can anybody offer assistance?

yenchenlin, 2016-12-10:
@dchang0 thanks a lot for your solution! It works!

However, I'm interested in how you figure it out installing `install libpcre3 8.35-7` can fix the bug?