ros-visualization/rviz #957: Floating point exception in billboard path display when path is empty

stwirth, 2016-01-04:
Steps to reproduce:
  * Add a path display, change Topic to /path and Line Style to Billboards
  * ` rostopic pub -r 10 /path nav_msgs/Path '{header: {frame_id: map, stamp: now}, poses: []}'`

rviz crashes with `Floating point exception`.

Version info:
```
[ INFO] rviz version 1.11.10
[ INFO] compiled against OGRE version 1.8.1 (Byatis)
[ INFO] Stereo is NOT SUPPORTED
[ INFO] OpenGl version: 4.5 (GLSL 4.5).
```

spmaniato, 2016-03-01:
Just adding that I experienced this too. I'm in an Ubuntu Trusty VM running ROS Indigo, in case that's relevant.

```
[ INFO] [1456862922.857658950]: rviz version 1.11.10
[ INFO] [1456862922.857746020]: compiled against OGRE version 1.8.1 (Byatis)
[ INFO] [1456862923.564955971]: Stereo is NOT SUPPORTED
[ INFO] [1456862923.567417920]: OpenGl version: 2.1 (GLSL 1.2).
Floating point exception (core dumped)
```

A single message is enough to cause it.

```
rostopic pub --once path nav_msgs/Path "header:
  seq: 0
  stamp:
    secs: 0
    nsecs: 0
  frame_id: 'map'
poses: []"
```

wjwwood, 2016-03-22:
This appears to be fixed in the `indigo-devel` branch (probably fixed by this pr): https://github.com/ros-visualization/rviz/pull/942

closed 2016-03-22 by wjwwood