ros-visualization/rviz #964: default_plugin missing in exported libraries

dornhege, 2016-01-20:
This was recently removed in 1f897d9 and #948. The problem is that any code relying on rviz plugins fails to link. This affects, for example, the MoveIt benchmarks gui. Building code on top of the rviz interactive markers might have been an unsupported use case, but I don't see any way to cleanly link this otherwise.

jbohren, 2016-02-14:
As an aside, the RViz default plugin product is literally called `libdefault_plugin.so` I think _this_ needs to be fixed as well. That being said, I think the #948 makes a good point, and if you're explicitly linking against the plugin like `moveit_benchmarks_gui` does, it should just list the library in the `target_link_libraries` directive. Again, though, this library should _really_ be called something other than `default_plugin`.

rhaschke, 2016-03-07:
It's not only about listing libdefault_plugin.so in target_link_libraries. One needs to ensure build order as well - which was perfectly achieved by having it listed in exported targets.

wjwwood, 2016-03-07:
> it should just list the library in the target_link_libraries directive

Yeah, that's not a great idea, especially if I change the name of the library. As @rhaschke suggested that should really be done through a CMake variable, which before could have been exported targets. @rhaschke also, dependencies would be resolved if you're linking against it. I'm not aware of any use of the default plugin that would require you to use it with `add_dependencies`.

wjwwood, 2016-03-22:
I created https://github.com/ros-visualization/rviz/pull/979 which provides a cmake variable `rviz_DEFAULT_PLUGIN_LIBRARIES` which you can use for linking. I will merge that into indigo, but I will only change the library name in Kinetic (in case some people are referencing it by name in their link lines).

wjwwood, 2016-03-22:
I believe this to be covered by #979, and I'll follow up on @jbohren comment about the libraries actual name in a Kinetic specific pr (after I branch for Kinetic which I will do after I do a release for indigo-devel).

closed 2016-03-22 by wjwwood