ros-visualization/rviz #967: QUESTION: Migrating to Qt5

OAkyildiz, 2016-02-03:
Hello,

As the title says, do you plan to migrate to Qt5?

wjwwood, 2016-02-03:
We've been discussing a move to Qt5 for all of ROS for the next LTS ROS Kinetic (due out this May). There will likely be more movement on this in about three weeks when we're supposed to make a decision about whether or not to go for it. If we did, we'd update the Kinetic REP3 pr: https://github.com/ros-infrastructure/rep/pull/106

There is a pr for Qt5 (conditional) support for rviz already, but I've not had time to try it out: https://github.com/ros-visualization/rviz/pull/909

wjwwood, 2016-02-03:
I'm gonna close this for now since we already have two places where we can discuss this. You can continue to comment here if you wish though.

closed 2016-02-03 by wjwwood

OAkyildiz, 2016-02-13:
Thanks for the info!