ros-visualization/rviz #975: rosrun rviz rviz problem: Segmentation fault (core dumped)

won13y, 2016-03-09:
I was able to run rviz correctly in my notebook(Dell XPS13), ubuntu 14.04 LTS

But, Some day it produced the following error:

"Segmentation fault (core dumped)"

wkyoun@wkyoun-XPS-13-9343:~$ rosrun rviz rviz

    [ INFO] [1457530684.442584629]: rviz version 1.11.10
    [ INFO] [1457530684.442650491]: compiled against OGRE version 1.8.1 (Byatis)
    [ INFO] [1457530684.610051218]: Stereo is NOT SUPPORTED
    [ INFO] [1457530684.610144623]: OpenGl version: 3 (GLSL 1.3).
    Segmentation fault (core dumped)

since Ogre.log Run rviz with the "-l" option rosrun rviz rviz -l to produce an "Ogre.log" file in the current directory. 

I Include the Ogre.log file to describe the problem that I encounter

wkyoun@wkyoun-XPS-13-9343:~$  rosrun rviz rviz -l

    [ INFO] [1457530520.604631051]: rviz version 1.11.10
    [ INFO] [1457530520.604687677]: compiled against OGRE version 1.8.1 (Byatis)
    [ INFO] [1457530520.752815835]: Creating resource group General
    [ INFO] [1457530520.752875497]: Creating resource group Internal
    [ INFO] [1457530520.752908250]: Creating resource group Autodetect
    [ INFO] [1457530520.752960535]: SceneManagerFactory for type 'DefaultSceneManager' registered.
    [ INFO] [1457530520.753098423]: Registering ResourceManager for type Material
    [ INFO] [1457530520.753126427]: Registering ResourceManager for type Mesh
    [ INFO] [1457530520.753147397]: Registering ResourceManager for type Skeleton
    [ INFO] [1457530520.753178324]: MovableObjectFactory for type 'ParticleSystem' registered.
    [ INFO] [1457530520.753205429]: OverlayElementFactory for type Panel registered.
    [ INFO] [1457530520.753223080]: OverlayElementFactory for type BorderPanel registered.
    [ INFO] [1457530520.753239641]: OverlayElementFactory for type TextArea registered.
    [ INFO] [1457530520.753261621]: Registering ResourceManager for type Font
    [ INFO] [1457530520.753285238]: ArchiveFactory for archive type FileSystem registered.
    [ INFO] [1457530520.753304164]: ArchiveFactory for archive type Zip registered.
    [ INFO] [1457530520.753322702]: ArchiveFactory for archive type EmbeddedZip registered.
    [ INFO] [1457530520.753338725]: DDS codec registering
    [ INFO] [1457530520.753362667]: FreeImage version: 3.15.4
    [ INFO] [1457530520.753381648]: This program uses FreeImage, a free, open source image library supporting all common bitmap formats. See http://freeimage.sourceforge.net for details
    [ INFO] [1457530520.753499220]: Supported formats: bmp,ico,jpg,jif,jpeg,jpe,jng,koa,iff,lbm,mng,pbm,pbm,pcd,pcx,pgm,pgm,png,ppm,ppm,ras,tga,targa,tif,tiff,wap,wbmp,wbm,psd,cut,xbm,xpm,gif,hdr,g3,sgi,exr,j2k,j2c,jp2,pfm,pct,pict,pic,3fr,arw,bay,bmq,cap,cine,cr2,crw,cs1,dc2,dcr,drf,dsc,dng,erf,fff,ia,iiq,k25,kc2,kdc,mdc,mef,mos,mrw,nef,nrw,orf,pef,ptx,pxn,qtk,raf,raw,rdc,rw2,rwl,rwz,sr2,srf,srw,sti
    [ INFO] [1457530520.753525207]: Registering ResourceManager for type HighLevelGpuProgram
    [ INFO] [1457530520.753549297]: Registering ResourceManager for type Compositor
    [ INFO] [1457530520.753706891]: MovableObjectFactory for type 'Entity' registered.
    [ INFO] [1457530520.753731210]: MovableObjectFactory for type 'Light' registered.
    [ INFO] [1457530520.753749953]: MovableObjectFactory for type 'BillboardSet' registered.
    [ INFO] [1457530520.753768367]: MovableObjectFactory for type 'ManualObject' registered.
    [ INFO] [1457530520.753788843]: MovableObjectFactory for type 'BillboardChain' registered.
    [ INFO] [1457530520.753806951]: MovableObjectFactory for type 'RibbonTrail' registered.
    [ INFO] [1457530520.753862480]: *-*-* OGRE Initialising
    [ INFO] [1457530520.753888108]: *-*-* Version 1.8.1 (Byatis)
    [ INFO] [1457530520.753911738]: Loading library /usr/lib/x86_64-linux-gnu/OGRE-1.8.0/RenderSystem_GL
    [ INFO] [1457530520.755312633]: Installing plugin: GL RenderSystem
    [ INFO] [1457530520.755350068]: OpenGL Rendering Subsystem created.
    [ INFO] [1457530520.756876545]: Plugin successfully installed
    [ INFO] [1457530520.756939920]: Loading library /usr/lib/x86_64-linux-gnu/OGRE-1.8.0/Plugin_OctreeSceneManager
    [ INFO] [1457530520.757804983]: Installing plugin: Octree Scene Manager
    [ INFO] [1457530520.757843339]: Plugin successfully installed
    [ INFO] [1457530520.757878177]: Loading library /usr/lib/x86_64-linux-gnu/OGRE-1.8.0/Plugin_ParticleFX
    [ INFO] [1457530520.758815918]: Installing plugin: ParticleFX
    [ INFO] [1457530520.758849715]: Particle Emitter Type 'Point' registered
    [ INFO] [1457530520.758880309]: Particle Emitter Type 'Box' registered
    [ INFO] [1457530520.758907986]: Particle Emitter Type 'Ellipsoid' registered
    [ INFO] [1457530520.758934267]: Particle Emitter Type 'Cylinder' registered
    [ INFO] [1457530520.758961444]: Particle Emitter Type 'Ring' registered
    [ INFO] [1457530520.758987346]: Particle Emitter Type 'HollowEllipsoid' registered
    [ INFO] [1457530520.759011016]: Particle Affector Type 'LinearForce' registered
    [ INFO] [1457530520.759033068]: Particle Affector Type 'ColourFader' registered
    [ INFO] [1457530520.759054922]: Particle Affector Type 'ColourFader2' registered
    [ INFO] [1457530520.759077105]: Particle Affector Type 'ColourImage' registered
[... 152 lines cut ...]
    [ INFO] [1457530520.828950895]: Creating resources for group Internal
    [ INFO] [1457530520.828975850]: All done
    [ INFO] [1457530520.829000406]: Parsing scripts for resource group rviz
    [ INFO] [1457530520.830947464]: Parsing script glsl120.program
    [ INFO] [1457530520.832242083]: Parsing script nogp.program
    [ INFO] [1457530520.833079390]: Parsing script default_pick_and_depth.material
    [ INFO] [1457530520.833419506]: Parsing script point_cloud_point.material
    [ INFO] [1457530520.833695570]: Parsing script indexed_8bit_image.material
    [ INFO] [1457530520.833846219]: Parsing script point_cloud_tile.material
    [ INFO] [1457530520.834215843]: Parsing script point_cloud_flat_square.material
    [ INFO] [1457530520.834731886]: Parsing script point_cloud_sphere.material
    [ INFO] [1457530520.835382631]: Parsing script point_cloud_square.material
    [ INFO] [1457530520.835620290]: Parsing script point_cloud_box.material
    [ INFO] [1457530520.835830084]: Parsing script arial.fontdef
    [ INFO] [1457530520.835924584]: Finished parsing scripts for resource group rviz
    [ INFO] [1457530520.835970464]: Creating resources for group rviz
    [ INFO] [1457530520.836020084]: All done
    [ INFO] [1457530520.838282262]: GLRenderSystem::_createRenderWindow "OgreWindow(1)", 100x30 windowed  miscParams: externalGLControl= macAPI=carbon parentWindowHandle=62914594 
    [ INFO] [1457530520.840247518]: GLXWindow::create used FBConfigID = 154
    [ INFO] [1457530521.156601634]: Mesh: Loading rviz_sphere.mesh.
    [ INFO] [1457530521.333805780]: Texture: SelectionRect0Texture: Loading 1 faces(PF_R8G8B8A8,1x1x1) with 0 generated mipmaps from Image. Internal format is PF_A8R8G8B8,1x1x1.
    [ INFO] [1457530521.594350434]: Mesh: Loading rviz_cylinder.mesh.
    [ INFO] [1457530521.594610209]: Mesh: Loading rviz_cone.mesh.
    [ INFO] [1457530521.895053267]: Texture: ROSImageTexture0: Loading 1 faces(PF_A8R8G8B8,420x300x1) with 0 generated mipmaps from Image. Internal format is PF_A8R8G8B8,420x300x1.
    Segmentation fault (core dumped)

wjwwood, 2016-03-22:
@won13y sorry for the delay in responding. Unfortunately, without a `core` file I cannot see the backtrace of the segfault. See: http://wiki.ros.org/rviz/Troubleshooting#How_to_make_a_usable_bug_report

tslator, 2016-03-29:
I'm experiencing the same issue.  Rviz was working yesterday and now it's not.  I can't include the core file, it too big.  How can I transfer the core file?

[rvizgdb.txt](https://github.com/ros-visualization/rviz/files/194306/rvizgdb.txt)
[system_info.txt](https://github.com/ros-visualization/rviz/files/194307/system_info.txt)
[ogre.txt](https://github.com/ros-visualization/rviz/files/194317/ogre.txt)

wjwwood, 2016-03-29:
@tslator did anything change on your system? The public (default) version of rviz hasn't changed in quite some time. Are you installing from apt-get or building it from source?

@tslator can you put the core file in a gist.github.com maybe?

tslator, 2016-03-29:
Nothing since last night when I shutdown.  I'm installing from apt-get.  I uploaded the core to https://www.dropbox.com/sh/e1rfui8i19z9sm7/AADyQ65124djikb8v8T3YIn6a?dl=0

wjwwood, 2016-03-29:
@tslator It says the folder is empty.

tslator, 2016-03-29:
Sorry about that.  Not sure what happened.  Its there now.

tslator, 2016-04-01:
Did the core file shed any light on this issue?

wjwwood, 2016-04-01:
@tslator I can see it now, but I haven't had time to set it up and look at it. I'll try to do that today. Thanks for your patience.

wjwwood, 2016-04-02:
@tslator I looked into your core file and it is crashing deep in the GLXGL code of Ogre (the 3d library we use):

https://bitbucket.org/sinbad/ogre/src/525a7f3bcd4e51b1085c42a8b55fe07eb7700458/RenderSystems/GL/src/GLX/OgreGLXGLSupport.cpp?at=default&fileviewer=file-view-default#OgreGLXGLSupport.cpp-303

What I can say about this is that your issue is different from the original post, based on how far their ogre.log got compared to yours (yours stops after creating the GL subsystem, theirs continues). I can also say that it is unlikely to be an issue with `rviz`.

I think the most likely problem is that something on your system changed (video driver or X.org configurations) and now you no longer have GLX support. You can try running `glxgears` on the terminal to see if you have support for it. If that doesn't work you'll need to figure out why before trying rviz again.

The best case scenario is that Ogre could fail more gracefully when this happens, but that's out of our hands in inside of rviz.

tslator, 2016-04-02:
glxgears works just fine.  I did a complete re-install of Ubuntu and ROS (from packages) and got the same issue.  Then, I did a complete re-install of Ubuntu and built ROS from source -- that worked.  For now, I'll work from source.  Thanks for effort and assistance.

wjwwood, 2016-04-03:
So installing from binaries didn't work, but building ROS (and `rviz`) from source did? Are you using a branch of `rviz` (like `indigo-devel`) or the released version (a tag like `1.11.10`)?

wjwwood, 2016-04-03:
@won13y I need more information from you like a `core` file to make any progress with your issue. I'll just close this until you get back to me. I can reopen it then.

closed 2016-04-03 by wjwwood

ilidar, 2016-09-27:
Looks like I've got the same issue, but on macOS Sierra 💃 . Here is my [crash](https://github.com/ros-visualization/rviz/files/496588/rviz_crash.txt) and [output](https://github.com/ros-visualization/rviz/files/496589/rviz_out.txt) files. Any ideas ?

wjwwood, 2016-09-28:
@ilidar Sorry, I probably won't have time to dig into the issue right now. The code you're referencing is a very old bit of OS X specific code. As I read it, it is doing the right thing. The post you referenced is also very old (circa 2013).

gowmo, 2017-03-17:
I experience the same issue .. what is the solution.

immigssi, 2017-03-19:
I experienced the same issue, but because i was using virtual machine on a second screen and because of that there were some permission issue of graphics etc....what i did was simple: just restarted my virtual machine on my original display, run rviz it worked, then i transferred the display to second screen and use it normally.....no issue

westeast, 2017-05-23:
I have the same issue 👍 ,on mac .  Does not  my mac graphic card support OpenGL?

```
rosrun rviz rviz
[ INFO] [1495561024.692380000]: rviz version 1.12.4
[ INFO] [1495561024.692436000]: compiled against Qt version 5.8.0
[ INFO] [1495561024.692442000]: compiled against OGRE version 1.7.4 (Cthugha)
Segmentation fault: 11
[#38#zouxidong@01:37:05 ~/git/enmodel]$ rosrun rviz rviz  -l
[ INFO] [1495561034.888849000]: rviz version 1.12.4
[ INFO] [1495561034.888904000]: compiled against Qt version 5.8.0
[ INFO] [1495561034.888911000]: compiled against OGRE version 1.7.4 (Cthugha)
[ INFO] [1495561036.083851000]: Creating resource group General
[ INFO] [1495561036.083907000]: Creating resource group Internal
[ INFO] [1495561036.083923000]: Creating resource group Autodetect
[ INFO] [1495561036.083994000]: SceneManagerFactory for type 'DefaultSceneManager' registered.
[ INFO] [1495561036.084218000]: Registering ResourceManager for type Material
[ INFO] [1495561036.084250000]: Registering ResourceManager for type Mesh
[ INFO] [1495561036.084267000]: Registering ResourceManager for type Skeleton
[ INFO] [1495561036.084303000]: MovableObjectFactory for type 'ParticleSystem' registered.
[ INFO] [1495561036.084350000]: OverlayElementFactory for type Panel registered.
[ INFO] [1495561036.084366000]: OverlayElementFactory for type BorderPanel registered.
[ INFO] [1495561036.084379000]: OverlayElementFactory for type TextArea registered.
[ INFO] [1495561036.084398000]: Registering ResourceManager for type Font
[ INFO] [1495561036.084418000]: ArchiveFactory for archive type FileSystem registered.
[ INFO] [1495561036.084439000]: ArchiveFactory for archive type Zip registered.
[ INFO] [1495561036.084450000]: DDS codec registering
[ INFO] [1495561036.084478000]: FreeImage version: 3.17.0
[ INFO] [1495561036.084492000]: This program uses FreeImage, a free, open source image library supporting all common bitmap formats. See http://freeimage.sourceforge.net for details
[ INFO] [1495561036.084661000]: Supported formats: bmp,ico,jpg,jif,jpeg,jpe,jng,koa,iff,lbm,mng,pbm,pbm,pcd,pcx,pgm,pgm,png,ppm,ppm,ras,tga,targa,tif,tiff,wap,wbmp,wbm,psd,cut,xbm,xpm,gif,hdr,g3,sgi,rgb,rgba,bw,exr,j2k,j2c,jp2,pfm,pct,pict,pic,3fr,arw,bay,bmq,cap,cine,cr2,crw,cs1,dc2,dcr,drf,dsc,dng,erf,fff,ia,iiq,k25,kc2,kdc,mdc,mef,mos,mrw,nef,nrw,orf,pef,ptx,pxn,qtk,raf,raw,rdc,rw2,rwl,rwz,sr2,srf,srw,sti,x3f,webp,jxr,wdp,hdp
[ INFO] [1495561036.084691000]: Registering ResourceManager for type HighLevelGpuProgram
[ INFO] [1495561036.084781000]: Registering ResourceManager for type Compositor
[ INFO] [1495561036.084949000]: MovableObjectFactory for type 'Entity' registered.
[ INFO] [1495561036.084968000]: MovableObjectFactory for type 'Light' registered.
[ INFO] [1495561036.084978000]: MovableObjectFactory for type 'BillboardSet' registered.
[ INFO] [1495561036.084988000]: MovableObjectFactory for type 'ManualObject' registered.
[ INFO] [1495561036.085012000]: MovableObjectFactory for type 'BillboardChain' registered.
[ INFO] [1495561036.085035000]: MovableObjectFactory for type 'RibbonTrail' registered.
[ INFO] [1495561036.085190000]: *-*-* OGRE Initialising
[ INFO] [1495561036.085215000]: *-*-* Version 1.7.4 (Cthugha)
[ INFO] [1495561036.085239000]: Loading library /usr/local/Cellar/ogre/1.7.4_2/lib/libRenderSystem_GL
[ INFO] [1495561036.089648000]: Installing plugin: GL RenderSystem
[ INFO] [1495561036.089741000]: OpenGL Rendering Subsystem created.
[ INFO] [1495561036.089925000]: Plugin successfully installed
[ INFO] [1495561036.089951000]: Loading library /usr/local/Cellar/ogre/1.7.4_2/lib/libPlugin_OctreeSceneManager
[ INFO] [1495561036.094892000]: Installing plugin: Octree & Terrain Scene Manager
[ INFO] [1495561036.094949000]: Plugin successfully installed
[ INFO] [1495561036.094974000]: Loading library /usr/local/Cellar/ogre/1.7.4_2/lib/libPlugin_ParticleFX
[ INFO] [1495561036.099312000]: Installing plugin: ParticleFX
[ INFO] [1495561036.099347000]: Particle Emitter Type 'Point' registered
[ INFO] [1495561036.099365000]: Particle Emitter Type 'Box' registered
[ INFO] [1495561036.099392000]: Particle Emitter Type 'Ellipsoid' registered
[ INFO] [1495561036.099442000]: Particle Emitter Type 'Cylinder' registered
[ INFO] [1495561036.099456000]: Particle Emitter Type 'Ring' registered
[ INFO] [1495561036.099469000]: Particle Emitter Type 'HollowEllipsoid' registered
[ INFO] [1495561036.099484000]: Particle Affector Type 'LinearForce' registered
[ INFO] [1495561036.099497000]: Particle Affector Type 'ColourFader' registered
[ INFO] [1495561036.099510000]: Particle Affector Type 'ColourFader2' registered
[ INFO] [1495561036.099523000]: Particle Affector Type 'ColourImage' registered
[ INFO] [1495561036.099549000]: Particle Affector Type 'ColourInterpolator' registered
[ INFO] [1495561036.099568000]: Particle Affector Type 'Scaler' registered
[ INFO] [1495561036.099581000]: Particle Affector Type 'Rotator' registered
[ INFO] [1495561036.099592000]: Particle Affector Type 'DirectionRandomiser' registered
[ INFO] [1495561036.099604000]: Particle Affector Type 'DeflectorPlane' registered
[ INFO] [1495561036.099616000]: Plugin successfully installed
[ INFO] [1495561036.099649000]: CPU Identifier & Features
[ INFO] [1495561036.099667000]: -------------------------
[ INFO] [1495561036.099677000]:  *   CPU ID: GenuineIntel: Intel(R) Core(TM) i5-5257U CPU @ 2.70GHz
[ INFO] [1495561036.099692000]:  *      SSE: yes
[ INFO] [1495561036.099702000]:  *     SSE2: yes
[... cut ...]
[ INFO] [1495561036.099785000]:  *      PRO: yes
[ INFO] [1495561036.099794000]:  *       HT: no
[ INFO] [1495561036.099803000]: -------------------------
[ INFO] [1495561036.099820000]: ********************************************
***  Starting Mac OS X OpenGL Subsystem  ***
********************************************
[ INFO] [1495561036.099907000]: GLRenderSystem::_createRenderWindow "OgreWindow(0)", 1x1 windowed  miscParams: FSAA=4 externalGLControl= externalWindowHandle=0 macAPI=carbon
Segmentation fault: 11

```

agustincosta, 2018-09-10:
In my case this was solved updating the Mesa 3D graphics library. I am running ROS kinetic in Ubuntu 16.04 on a VM. If that is your case you should also disable hadware 3D acceleration.

This is a link on how to update the Mesa 3D graphics on Ubuntu 16.04: http://ubuntuhandbook.org/index.php/2018/01/how-to-install-mesa-17-3-3-in-ubuntu-16-04-17-10/

I hope this helps because I lost days troubleshooting this as a beginner in ROS and Ubuntu myself.

huckl3b3rry87, 2018-09-16:
fixed my issue as well thanks so much @agustincosta !!

https://github.com/ros-visualization/rviz/issues/1291

karthikv123, 2018-11-15:
this solves my issue, helped me a lot. thank you @agustincosta.

xufengyin, 2019-01-03:
I was suffered with the same issue for several days after I downloaded and compiled the rviz-kinetic-level from source. 
Finally, with gdb I got this information: 

Thread 1 "rviz" received signal SIGSEGV, Segmentation fault. 
0x00007ffff79bd615 in bool boost::math::detail::isfinite_impl<double>(double, boost::math::detail::native_tag const&) () from ....../librviz.so

And with the "printf" debug method, I located this issue was caused by ros::Time(0). 
I tried to replaced all ros::Time(0) I could found with ros::Time(0,0). Surprise! The Segmentation fault disappeared! The rviz is open! 

I don't know why. Maybe someone can help me to give out the exact reason.