ros-visualization/rviz #986: undefined reference to `rviz::OrbitViewController::*`

OAkyildiz, 2016-04-03:
Hello,
I have installed rviz from source and when installing rrtabmap_ros, I get
 
```
/home/oakyildiz/rtabmap_ws/devel/lib/librtabmap_ros.so: undefined reference to 
`rviz::OrbitViewController::reset()'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/oakyildiz/rtabmap_ws/devel/lib/rtabmap_ros/map_assembler] Error 1
make[1]: *** [CMakeFiles/map_assembler.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```
undefined reference error repeats itself for every member function of `OrbitViewController`.
(The following also applies for librviz.so for ros-indigo-rviz as well as source installed one)
So I checked librviz.so:
```
objdump -T ./lib/librviz.so | grep "OrbitViewController"
```

returns empty, but:
```
objdump -T ./lib/librviz.so | grep "ViewController"

```
Results in:
```
00000000001b15e0  w   DO .rodata	0000000000000054  Base        _ZTSN12class_loader20class_loader_private18AbstractMetaObjectIN4rviz14ViewControllerEEE
0000000000412a30 u    DO .data	0000000000000010  Base        _ZZN9pluginlib11ClassLoaderIN4rviz14ViewControllerEE21unloadLibraryForClassERKSsE33__rosconsole_define_location__loc
00000000001710e0  w   DF .text	0000000000000383  Base        _ZN9pluginlib11ClassLoaderIN4rviz14ViewControllerEE25determineAvailableClassesERKSt6vectorISsSaISsEE
000000000016be50 g    DF .text	00000000000000d5  Base        _ZN4rviz14ViewController14handleKeyEventEP9QKeyEventPNS_11RenderPanelE
000000000016e170 g    DF .text	0000000000000007  Base        _ZN4rviz23ViewControllerContainer15addChildToFrontEPNS_8PropertyE
000000000016e130 g    DF .text	0000000000000021  Base        _ZNK4rviz23ViewControllerContainer12getViewFlagsEi
0000000000196810 g    DF .text	0000000000000043  Base        _ZN4rviz35FramePositionTrackingViewController11qt_metacastEPKc
0000000000412a90 u    DO .data	0000000000000010  Base        _ZZN9pluginlib11ClassLoaderIN4rviz14ViewControllerEE32extractPackageNameFromPackageXMLERKSsE33__rosconsole_define_location__loc
00000000000e5cb0  w   DF .text	000000000000000c  Base        _ZN4rviz14ViewController10setClassIdERK7QString
00000000000f18a0 g    DF .text	0000000000000117  Base        _ZN4rviz35FramePositionTrackingViewController15getNewTransformEv
000000000019d790  w   DF .text	0000000000000013  Base        _ZN4rviz23ViewControllerContainerD2Ev
000000000016e8b0  w   DF .text	00000000000001cb  Base        _ZN5QHashI7QStringN4rviz16PluginlibFactoryINS1_14ViewControllerEE18BuiltInClassRecordEE13duplicateNodeEPN9QHashData4NodeEPv
0000000000412b30 u    DO .data	0000000000000010  Base        _ZZN9pluginlib11ClassLoaderIN4rviz14ViewControllerEEC1ESsSsSsSt6vectorISsSaISsEEE33__rosconsole_define_location__loc_0
00000000000e5c50  w   DF .text	0000000000000002  Base        _ZN4rviz14ViewController14transitionFromEPS0_
0000000000174490  w   DF .text	000000000000004f  Base        _ZN5QHashI7QStringN4rviz16PluginlibFactoryINS1_14ViewControllerEE18BuiltInClassRecordEE13detach_helperEv
000000000016e6d0  w   DF .text	0000000000000012  Base        _ZN9pluginlib11ClassLoaderIN4rviz14ViewControllerEE22getRegisteredLibrariesEv
00000000001767c0  w   DF .text	00000000000002d7  Base        _ZN9pluginlib11ClassLoaderIN4rviz14ViewControllerEED2Ev
0000000000412a70 u    DO .data	0000000000000010  Base        _ZZN9pluginlib11ClassLoaderIN4rviz14ViewControllerEED1EvE33__rosconsole_define_location__loc
00000000000e5ce0  w   DF .text	000000000000005b  Base        _ZNK4rviz14ViewController10getClassIdEv
000000000016c010 g    DF .text	0000000000000156  Base        _ZN4rviz14ViewController22updateStereoPropertiesEv
0000000000171970  w   DF .text	000000000000034c  Base        _ZN9pluginlib11ClassLoaderIN4rviz14ViewControllerEE29getErrorStringForUnknownClassERKSs
00000000000e5c40  w   DF .text	0000000000000002  Base        _ZN4rviz14ViewController5mimicEPS0_
000000000016f310  w   DF .text	00000000000000ba  Base        _ZN4rviz16PluginlibFactoryINS_14ViewControllerEE18BuiltInClassRecordD1Ev
000000000016d0f0 g    DF .text	000000000000003e  Base        _ZNK4rviz14ViewController12getViewFlagsEi
000000000040a090  w   DO .data.rel.ro	0000000000000018  Base        _ZTIN4rviz14ViewControllerE
000000000016c800 g    DF .text	0000000000000012  Base        _ZN4rviz14ViewControllerD0Ev
00000000001968b0  w   DF .text	0000000000000002  Base        _ZN4rviz14ViewController6lookAtERKN4Ogre7Vector3E
00000000001a15f0  w   DO .rodata	000000000000001e  Base        _ZTSN4rviz20FailedViewControllerE
000000000016f7c0  w   DF .text	0000000000000666  Base        _ZN9pluginlib11ClassLoaderIN4rviz14ViewControllerEE31getPackageFromPluginXMLFilePathERKSs
000000000016ea80  w   DF .text	00000000000000ec  Base        _ZN5QHashI7QStringN4rviz16PluginlibFactoryINS1_14ViewControllerEE18BuiltInClassRecordEE11deleteNode2EPN9QHashData4NodeE
00000000000f1550 g    DF .text	0000000000000012  Base        _ZN4rviz35FramePositionTrackingViewControllerD0Ev
00000000000f1480 g    DF .text	0000000000000067  Base        _ZN4rviz35FramePositionTrackingViewController17updateTargetFrameEv
000000000040a2d0  w   DO .data.rel.ro	0000000000000018  Base        _ZTIN4rviz23ViewControllerContainerE
0000000000412b20 u    DO .data	0000000000000010  Base        _ZZN9pluginlib11ClassLoaderIN4rviz14ViewControllerEE25determineAvailableClassesERKSt6vectorISsSaISsEEE33__rosconsole_define_location__loc
00000000001b16a0  w   DO .rodata	0000000000000030  Base        _ZTSN4rviz16PluginlibFactoryINS_14ViewControllerEEE
000000000016d430  w   DF .text	00000000000000c1  Base        _ZN4QMapIN4rviz14ViewController10CursorTypeE7QCursorE13detach_helperEv
0000000000403bc0 g    DO .data.rel.ro	0000000000000020  Base        _ZN4rviz35FramePositionTrackingViewController16staticMetaObjectE
000000000016e6f0  w   DF .text	000000000000010e  Base        _ZN9pluginlib11ClassLoaderIN4rviz14ViewControllerEE17getPluginXmlPathsEv
0000000000176060  w   DF .text	000000000000075d  Base        _ZN9pluginlib11ClassLoaderIN4rviz14ViewControllerEEC1ESsSsSsSt6vectorISsSaISsEE
000000000016ca40 g    DF .text	00000000000005d0  Base        _ZN4rviz14ViewController10initializeEPNS_14DisplayContextE
00000000001744e0  w   DF .text	00000000000001ec  Base        _ZN4rviz16PluginlibFactoryINS_14ViewControllerEE19getDeclaredClassIdsEv
00000000001718b0  w   DF .text	000000000000005e  Base        _ZN9pluginlib11ClassLoaderIN4rviz14ViewControllerEE15getClassPackageERKSs
0000000000174e40  w   DF .text	000000000000033f  Base        _ZN12class_loader20class_loader_private14createInstanceIN4rviz14ViewControllerEEEPT_RKSsPNS_11ClassLoaderE
000000000016daf0 g    DF .text	00000000000000e2  Base        _ZN4rviz11ViewManager4copyEPNS_14ViewControllerE
00000000000f1640 g    DF .text	00000000000000d4  Base        _ZN4rviz35FramePositionTrackingViewController5mimicEPNS_14ViewControllerE
[... 133 lines cut ...]
000000000016f3d0  w   DF .text	00000000000003ee  Base        _ZN9pluginlib11ClassLoaderIN4rviz14ViewControllerEE32extractPackageNameFromPackageXMLERKSs
000000000016eee0  w   DF .text	0000000000000165  Base        _ZNK4rviz16PluginlibFactoryINS_14ViewControllerEE19getClassDescriptionERK7QString
000000000016c170 g    DF .text	00000000000005b1  Base        _ZN4rviz14ViewControllerC2Ev
0000000000175ce0  w   DF .text	0000000000000372  Base        _ZN4rviz16PluginlibFactoryINS_14ViewControllerEE7makeRawERK7QStringPS3_
00000000000f1720 g    DF .text	0000000000000179  Base        _ZN4rviz35FramePositionTrackingViewControllerC2Ev
000000000016e320 g    DF .text	000000000000004f  Base        _ZN4rviz11ViewManager14setCurrentFromEPNS_14ViewControllerE
0000000000412a80 u    DO .data	0000000000000010  Base        _ZZN9pluginlib11ClassLoaderIN4rviz14ViewControllerEE32extractPackageNameFromPackageXMLERKSsE33__rosconsole_define_location__loc_0
000000000019d4e0  w   DF .text	0000000000000002  Base        _ZN4rviz14ViewController10onActivateEv
00000000001bb220  w   DO .rodata	0000000000000021  Base        _ZTSN4rviz23ViewControllerContainerE
000000000016d1f0 g    DF .text	0000000000000178  Base        _ZNK4rviz14ViewController4saveENS_6ConfigE
0000000000412a20 u    DO .data	0000000000000010  Base        _ZZN9pluginlib11ClassLoaderIN4rviz14ViewControllerEE19getClassLibraryPathERKSsE33__rosconsole_define_location__loc
00000000000e5c30  w   DF .text	0000000000000002  Base        _ZN4rviz14ViewController16handleMouseEventERNS_18ViewportMouseEventE
[... cut ...]
0000000000173170  w   DF .text	00000000000008c2  Base        _ZN9pluginlib11ClassLoaderIN4rviz14ViewControllerEE23getAllLibraryPathsToTryERKSsS5_
00000000001727f0  w   DF .text	0000000000000672  Base        _ZN9pluginlib11ClassLoaderIN4rviz14ViewControllerEE22refreshDeclaredClassesEv
0000000000412b10 u    DO .data	0000000000000010  Base        _ZZN9pluginlib11ClassLoaderIN4rviz14ViewControllerEE25determineAvailableClassesERKSt6vectorISsSaISsEEE33__rosconsole_define_location__loc_0
```

Is this resulting from CMakeLists.txt in rviz?

wjwwood, 2016-04-03:
Make sure you are linking against the default plugin, see:

https://github.com/ros-visualization/rviz/issues/964 and https://github.com/ros-visualization/rviz/pull/979

OAkyildiz, 2016-04-03:
Thanks, a lot. And additional thanks for quick reply.

closed 2016-04-03 by OAkyildiz