ros-visualization/rviz #995: Transparency from Collada/DAE meshes is ignored

VictorLamoine, 2016-04-11:
I attach an end-effector to a robot and set the transparency channel to 0.5 in the visual DAE mesh.
The transparency property is ignored by RViz and the mesh is dislayed 100% opaque:

[image]

Suzanne monkey should be transparent.

Here is what the DAE mesh file looks like:
```xml
    <effect id="Material_001-effect">
      <profile_COMMON>
        <technique sid="common">
          <phong>
            <emission>
              <color sid="emission">0 0 0 1</color>
            </emission>
            <ambient>
              <color sid="ambient">0 0 0 1</color>
            </ambient>
            <diffuse>
              <color sid="diffuse">0.1567682 0.1241165 0.8000001 1</color>
            </diffuse>
            <specular>
              <color sid="specular">0.5 0.5 0.5 1</color>
            </specular>
            <shininess>
              <float sid="shininess">50</float>
            </shininess>
            <transparency>
              <float sid="transparency">0.5</float>
            </transparency>
            <index_of_refraction>
              <float sid="index_of_refraction">1</float>
            </index_of_refraction>
          </phong>
        </technique>
      </profile_COMMON>
    </effect>
```

Here is a repo containing a full example:
https://github.com/VictorLamoine/rviz_mesh_transparency

wjwwood, 2016-10-18:
Thanks for the report. Assuming assimp exposes this information for us, it should be relatively simple to make Ogre render the mesh transparently.

I don't have time to look into this enhancement, but I should have time to review a pull request if you decide to implement it.

VictorLamoine, 2016-10-19:
Loading of the materials happens here:
https://github.com/ros-visualization/rviz/blob/kinetic-devel/src/rviz/mesh_loader.cpp#L437

In the dae file exported by Blender the transparency is expressed in:
```xml
<transparent opaque="A_ONE">
  <color>1 1 1 0.5</color>
</transparent>
```
We need to do two things:
- Retrieve the opacity value from the material with ASSIMP
- Assign the opacity value to the Ogre material

My test code so far:
https://github.com/ros-visualization/rviz/compare/kinetic-devel...VictorLamoine:mesh_transparency?expand=1

Problems:
- I can't read the fourth value with ASSIMP because the `aiColor3D` structure only stores `r`,`g`,`b`
- Setting the `ambient.a` and `diffuse.a` value of the Ogre material to `0.1` does not make the material transparent. (is there a shading option that prevents that?)

Useful URLs:
- https://sourceforge.net/p/assimp/discussion/817654/thread/1b4a13eb/
- http://assimp.sourceforge.net/lib_html/materials.html

VictorLamoine, 2016-10-19:
@de-vri-es if you have some knowledge of ASSIMP I'd be happy to get some help

VictorLamoine, 2018-08-08:
I have made some progress.

I have simplified a lot the example repository, very few dependencies now (and updated for Melodic):
https://gitlab.com/InstitutMaupertuis/rviz_mesh_transparency

I am now able to fetch the transparency value from the DAE file but changing the value does not affect the graphics (don't know why).

https://github.com/ros-visualization/rviz/compare/melodic-devel...VictorLamoine:mesh_transparency?w=1