ros-visualization/rviz #997: librviz: Setting View Controllers

OAkyildiz, 2016-04-12:
Hello,

I am using librviz in a package for 3d scenes and  I want to set the view controllers for the `VisualizationManagers` I have.

The displays added to a scene are modified by `subProp` instead of getters and setters (preference, I assumed). I couldn't find where to look up for the property - string matches. _(I checked the names in .rviz config files, but  that didn't work for all of them)._

In that light, is there a unique way to set the `ViewControllers` and controlling them, or would initializing a `ViewController` (Ortho and ThirdPersonFllower in my case) object and  assigning it would suffice? _(I didn't see a setViewController, I think that's what stopped me in the first place.)_

Thanks,

OAkyildiz, 2016-04-18:
Hello,

Could you please tell me how to find sub_prop names, at the least?

Thanks,

wjwwood, 2016-04-18:
@OAkyildiz I would, but I don't know myself (I didn't write that part of the code). I looked at this for a minute or two, but couldn't find it trivially. I'll try to spend some more time on it this week, but you might have to crawl through to code to find out.

If you're lucky, one of the older rviz maintainers (authors) might have time to answer your question.

OAkyildiz, 2016-04-19:
Thanks @wjwwood (I was about to rephrase my question, I was trying to say "If you could answer that part, it can help me for now"),
That's what I thought since  it doesn't seem to be documented.  I tried to initialize a controller and then used rviz::RenderPanel::setViewController but the process died because some attributes were undefined. I was going to look into *VewController to see what attributes are not defined on initializing the object.

lucasw, 2017-03-04:
I've been trying to select a view controller within an rviz rqt plugin by looking at what rviz/view_manager is doing, maybe some of this is useful.  The view renders what it ought to from a fixed view, but the mouse doesn't translate or rotate it- handleMouseEvent isn't getting hooked up?

```
initPlugin(...)...
...
  manager_ = new rviz::VisualizationManager(render_panel_);
  render_panel_->initialize(manager_->getSceneManager(), manager_);
  manager_->initialize();
  manager_->startUpdate();

  rviz::ViewController* view = new rviz::OrbitViewController();
  if (view)
  {
    view->initialize(manager_);
    render_panel_->setViewController(view);
  }
  else
  {
    ROS_ERROR_STREAM("could not make a view");
  }

  manager_->setFixedFrame("my_frame");
...
```

This (mis)behaves identically but allows setting view by name, without having to have headers to the specific view type, the plugin just has to be loaded at run time:

```
  rviz::PluginlibFactory<rviz::ViewController>* factory;
  factory = new rviz::PluginlibFactory<rviz::ViewController>("rviz",
      "rviz::ViewController");
  QString class_id = "rviz/Orbit";
  QString error;
  rviz::ViewController* view = factory->make(class_id, &error);
```

lucasw, 2017-03-07:
It turns out render panel setViewController should not be called by the user, because the view manager is in charge of the view controller, and calling it has parts of the system using the newly set view controller and other parts still using an old one that view manager is using.

So instead call `getViewManager()->setCurrentViewControllerType` on the visualization manager:

```
  manager_ = new rviz::VisualizationManager( render_panel_ );
  render_panel_->initialize( manager_->getSceneManager(), manager_ );
  manager_->initialize();
  manager_->startUpdate();
  ...

  QString class_id = "rviz/Orbit";
  manager_->getViewManager()->setCurrentViewControllerType(class_id);
```

OAkyildiz, 2017-03-24:
Thanks Lucas! I will try this as soon as I get back to work on this.

OAkyildiz, 2018-06-03:
Display sub-properties are the strings of property names presented in the property tables of each `DisplayType` entry in [ROS Wiki:rviz/DisplayTypes]( http://wiki.ros.org/rviz/DisplayTypes). 
I think Lucas showed the way to properly change control view.
I was revisiting this for another project, thanks again!