ros-visualization/rviz #998: librviz:

OAkyildiz, 2016-04-12:
Hello, 

I am using `Qt` for making an application,  I have an `rviz::Scene` and some GUI elements that I want to overlay on top of this scene. The closest I got was by making this scene the parent of the Widgets, but since `renderPanel` is using an external renderer, Qt's native renderer can't get the pixel data to handle transparency.

I would like to know can I get transparency to work.  Do I need to access  the OGRE renderer the `visualizationManager`/`RenderPanel` is using  to get my widgets rendered by that? Or having my widgets rendered by Qt is enough for it to work?  (OR as a third option, can Qt render the panel instead of  OGRE?)

Thanks,

simonschmeisser, 2016-04-26:
Do you want to use QWidget or Qml? Qt4 or Qt5? I suggest you look for some general non rviz spezific example of combining your choice of gui elements with OpenGL. It should then be pretty simple to subclass/copy-paste rviz::RenderPanel to fit in there.

OAkyildiz, 2016-04-27:
I am using QWidget on Qt4.

wjwwood, 2016-10-18:
I'm not aware of anyway to do this. The code that allows Ogre to write to the Qt window is pretty fragile and quite OS specific. Short of changing how Ogre and Qt interface within rviz I don't know how you'd give control over to Qt to do the transparency.

OAkyildiz, 2018-07-17:
Update from the future: there are `jsk_visualization' and 'jsk_rviz_plugins' in that achieve the same thing by creating beefed-up  `Marker`s with the same display manager in the main DisplayFrame. I have a 'DisplayFrame' on top of an ImageView now but it's rather fragile as you said, so I will, at some point in the near future repurpose `rqt_image_view` instead.

closed 2018-07-17 by OAkyildiz