ros-visualization/rviz #999: rviz-1.11.14 with qt4 fails to build on gentoo

racko, 2016-04-15:
Building rviz-1.11.14 with -qt5 fails on my system because, as far as I understand it, dev-ros/python_qt_binding-0.3.0 still uses qt5. Building it with qt5 works. 

> sip: QMainWindow has not been defined
> Traceback (most recent call last):
>   File "//usr/share/python_qt_binding/cmake/sip_configure.py", line 76, in <module>
>     subprocess.check_call(cmd)
>   File "/usr/lib64/python2.7/subprocess.py", line 540, in check_call
>     raise CalledProcessError(retcode, cmd)
> subprocess.CalledProcessError: Command '['/usr/bin/sip', '-c', '/var/tmp/portage/dev-ros/rviz-1.11.14/work/rviz-1.11.14-python2_7/devel/bin/sip/rviz_sip', '-b', '/var/tmp/portage/dev-ros/rviz-1.11.14/work/rviz-1.11.14-python2_7/devel/bin/sip/rviz_sip/pyqtscripting.sbf', '-I', '/usr/share/sip/PyQt5', '-w', '-x', 'VendorID', '-t', 'WS_X11', '-t', 'Qt_5_5_1', '-x', 'PyQt_SSL', '-x', 'Py_v3', 'rviz.sip']' returned non-zero exit status 1
> src/python_bindings/sip/CMakeFiles/librviz_sip.dir/build.make:94: recipe for target 'devel/bin/sip/rviz_sip/Makefile' failed
> make[2]: *** [devel/bin/sip/rviz_sip/Makefile] Error 1
> make[2]: Leaving directory '/var/tmp/portage/dev-ros/rviz-1.11.14/work/rviz-1.11.14-python2_7'
> CMakeFiles/Makefile2:1488: recipe for target 'src/python_bindings/sip/CMakeFiles/librviz_sip.dir/all' failed
> make[1]: *** [src/python_bindings/sip/CMakeFiles/librviz_sip.dir/all] Error 2

I can't figure out if this an rviz issue or a gentoo issue because I don't understand how it is supposed to work at all: cmake/sip_configure.py in python_qt_binding-0.3.0 seems to explicitly mention PyQt5 in a lot of places and I don't see an option to tell it to use qt4.

Nevertheless I created a gentoo bug report, too:
https://bugs.gentoo.org/show_bug.cgi?id=580072

[build.log.gz](https://github.com/ros-visualization/rviz/files/220911/build.log.gz)

racko, 2016-04-15:
Ah, I got it. [](https://github.com/ros-visualization/rviz/commit/351f2d4fe49da98de320aadb4106683918e79f58) fixed this issue.

racko, 2016-04-15:
I overlooked the fact that the commit linked above is in the kinetic-devel branch. So the issue still persists in indigo.

wjwwood, 2016-04-15:
> Now that I think about it: Should rviz-1.11.14 even depend on python_qt_binding-0.3.0? Am I correct in assuming that it depends on python_qt_binding<0.3.0 and is incompatible with python_qt_binding>=0.3.0?

Exactly right. 0.3.0 is for Kinetic and uses Qt5 and PyQt5. If you want to use the indigo branch of rviz, you'll need version 0.2.19 of python_qt_binding.

racko, 2016-04-18:
Thank you very much, that cleared things up for me.

closed 2016-04-18 by racko