ros-visualization/rviz #1000: Rviz not functional

cuixiongyi, 2016-04-16:
Hi
I run into this problem last week. When I open rviz, it opens, but the windows look different and totally 
non functional. And it will crash if I tap any key on keyboard.

Looks like this:

[image]

After click the plus icon:

[image]

This problem is very repeatable. I reinstalled Ubuntu 14.04.4 multiple times and it happens every time.  with jade and indigo.

I just installed ros on Ubuntu 15.04, and jade and it happens.

I install ros using this
```
sudo apt-get install ros-indigo-desktop-full # or jade
```

One thing I noticed is that, just right after I installed ros, rviz open fine and look good. But when I open it and use it to visualize a pointcloud, my computer will go extreme slow with one CPU core 99% usage.
And after that  I open it again, that problem will happen.

It used to work fine for a long time on Ubuntu 14.04.4. Before the first time this problem happens, I installed octave, which require some opengl related library. But after that I wipe out the hard drive and reinstall the Ubuntu 14.04, it still happens.

Thank you

Debug info:
```
gdb /opt/ros/jade/lib/rviz/rviz 
```
https://gist.github.com/cuixiongyi/694abf7e89439daeb499e168b937e8d6

```
rosrun rviz rviz -l -v > log
```
https://gist.github.com/cuixiongyi/83ba3a21f4ad43155cc55d7a20a44925#file-gistfile1-txt

```
rosrun rviz rviz --log-level-debug > logdebug.txt
```
https://gist.github.com/cuixiongyi/09c88d23061d54137d0bbf874c061cbe

OAkyildiz, 2016-04-18:
I see a segfault in your gdb output:
```
Program received signal SIGSEGV, Segmentation fault.
0x00007ffff7b221a4 in rviz::ToolManager::handleChar(QKeyEvent*, rviz::RenderPanel*) ()
   from /opt/ros/jade/lib/librviz.so
(gdb) bt
```
Can you do a
```
ldd /opt/ros/jade/lib/librviz.so | grep -i "Qt" #also withouth grep
```
and also, (disregard this if you uninstalled `indigo` and installed `jade` to see if it will work) but  make sure you are not trying to run  indigo jade while you have indigo, or vice-versa.  

From what I see, you  updated some of the dependencies the ros was linked against and it broke. (Qt is notorious about that)

cuixiongyi, 2016-04-19:
Thank you very much for replying. Here is it.
Every time I wipe the hard drive and install a new ubuntu 14.04.4 and then install ros on it. I installed ros indigo 3 times and jade for 1 time. 
But at least 2 times I didn't update the system neither at the install process nor after it installed.

```
 ldd /opt/ros/jade/lib/librviz.so | grep -i "Qt"
```
https://gist.github.com/cuixiongyi/4900f0b980c850ba1cad29ab8499b8df

```
ldd /opt/ros/jade/lib/librviz.so
```
https://gist.github.com/cuixiongyi/8c45ef73106e8919b0b6d39b80c970ac

OAkyildiz, 2016-04-19:
As I've dealt with catkin and CMake more, I am not too familiar with gdb but I see a lot  of `in ?? () ` which  can mean these packages were compiled  without debug symbols](http://stackoverflow.com/questions/8390881/gdb-doesnt-show-function-names) (likely) or actually have an issue with those mapped regions.

I see that this is the case for `libglib-2.0.so` and `libQtGui.so`, so I might suggest invoking 
`ldd` for these to see if there is a bad dep. I am assuming they are installed by apt-get, so if you can install them from source with debug symbols  by
```
 cmake .. -DCMAKE_BUILD_TYPE=Debug <-Dadditional_args>
```
might give you more info. But before that you can try to debug with 
`valgrind --leak-check=full -v ./path/to/executable`

If one of these packages happen to be the problem,  make sure your `$LD_LIBRARY_PATH`, `$PATH`, `$ROS_PACKAGE_PATH' etc includes these (in that case error would be different afaik). Reinstalling  these packages might  be handy as well..

That being said, sticking with the LTS distributions (i.e. `trusty`, `xenial` - will be released on Apr 21) and the package versions that supports these will give you less headache. That being said, even a new LTS will have problems:`ros-jade` doesn't support `xenial` , but `kinetic` will, but how many packages will immediately support `kinetic`? Unless you need a specific feature, it is not worth your effort to have a bigger major or minor  version number. I recently tried to link [rtabmap](wiki.ros.org/rtabmap_ros) against Qt5, `rviz` who didn't have a qt5 branch at that time, had no problem with it but `rtabmap`, which actually came with Qt5 flags in its CMake cost me a lot of time with no result in the end.

cuixiongyi, 2016-04-19:
Hey you at WPI, me too!

cmastalli, 2016-05-02:
Hi guys,

I having the same problem in ROS-Indigo with Ubuntu 14.04. The problem start after updating the ros binaries. I have installed rviz 1.11.14.

Any recommendation?

gtinchev, 2016-06-01:
Hi guys,

Absolutely same in indigo, as @cmastalli mentioned. Any recommendations?
@cuixiongyi did you find a solution?

Cheers,
Georgi

cmastalli, 2016-06-02:
@gtinchev I solved by installing rviz from source. Another person in my lab had the same problem!

gtinchev, 2016-06-02:
@cmastalli I tried reinstalling rviz, but I track down with gdb and strace and it turned out that libPythonQt was the one causing it. I was sourcing it from a repo that we use in the lab. So try and check your LD_LIBRARY_PATH envvar, comment it out and use ros only - @cuixiongyi. This is what resolved my problem.

peci1, 2016-09-26:
For me, the problem was a newer version of yaml-cpp got in LD_LIBRARY_PATH. Can rviz somehow assure it really loads the 0.3 version and not the newer (and incompatible) 0.5 version?

wjwwood, 2016-10-18:
> For me, the problem was a newer version of yaml-cpp got in LD_LIBRARY_PATH. Can rviz somehow assure it really loads the 0.3 version and not the newer (and incompatible) 0.5 version?

I'm not aware of anyway to detect that after compile time.

I'm going to close this for now since it appears to be a ABI breakage problem that can be resolved by getting the right version of dependencies installed. I can reopen if necessary.

wjwwood, 2016-10-18:
Also, if installing from source works then try `sudo apt-get dist-upgrade` (note this will update all the software on your machine).

closed 2016-10-18 by wjwwood

pkru, 2018-01-09:
I had the same issue and here is how I solved it, hope it helps.
system: ubuntu 16.04, ros kinetic.
Reason is rviz uses its own yaml-cpp when it is installed, the yaml.h file is in usr/include and the so files are in /usr/lib/x86_64-linux-gnu/. If you install your own yaml, rviz is going to stop working since rviz should use the yaml installed from lib-yaml-cpp0.5v5 but is getting the newly installed yaml instead.

Solution: uninstall yaml on your system first, you need to remove your own yaml so files by doing `sudo rm -r /use/local/lib/libyam-cpp.so*` so that ros would not use this version you installed.

then by doing `sudo apt-get remove libyaml-cpp0.5v5`, this would remove the header files installed by you, since rviz depends on this, this would automatically remove rviz too.

After this is done, install rviz by doing `sudo apt-get install ros-kinetic-rviz
`, since rviz depends on libyaml-cpp, it would automatically install the yaml it depends on and override the other yaml files that are conflicting with this. 

after this, it started working

wjwwood, 2018-01-09:
> Reason is rviz uses its own yaml-cpp when it is installed

I think it's more accurate to say, rviz is linked against the system yamlcpp (provided by the debian package), it doesn't include it's own copy of yamlcpp.

But having a custom yamlcpp library ahead of the system one would be an issue.