ros-visualization/rviz #1002: unicode in Marker TEXT_VIEW_FACING text?

lucasw, 2016-04-18:
I tried inserting a degree symbol ° U+00B0 into the Marker text field after regular text, I can see it fine on rostopic echo but in rviz there is nothing after the standard text.  Is this a limitation of MovableText or Ogre setCaption()?

trainman419, 2016-04-18:
I suspect the font used in MovableText doesn't support unicode, and I think the MovableText widget assumes that the text is 8-bit characters.

lucasw, 2016-04-19:
This post mentions that additional characters can be loaded in, but there is a trade-off with texture memory (that maybe is a lot less important than 2006, except on embedded systems)
http://ogre3d.org/forums/viewtopic.php?f=2&t=33028

    font->setParameter("code_points", "1-512");

http://www.ogre3d.org/docs/manual/manual_44.html

"This directive allows you to specify which unicode code points should be generated as glyphs into the font texture. If you don’t specify this, code points 33-166 will be generated by default which covers the basic Latin 1 glyphs. If you use this flag, you should specify a space-separated list of inclusive code point ranges of the form ’start-end’. Numbers must be decimal."

After glancing at it I don't see anything in MovableText that would limit to 8-bit- I can try this out later and maybe generate a pr.

lucasw, 2016-04-22:
I see arial is the only font in https://github.com/ros-visualization/rviz/tree/kinetic-devel/ogre_media/fonts .

I need to learn what ogre is doing with the font there- it must be parsing the fontdef, and then loading the ttf, but I don't think it is touching the png (though it must be generating a similar texture internally).

So far setting "code_points" didn't have any effect, but maybe I was doing it at the wrong point in the font loading.  (Try setting it to a lower value and see if that results in later characters not rendering)