ros-visualization/rviz #1003: rviz default plugins don't load on kinetic

jgoppert, 2016-04-20:
```
The class required for this view controller, 'rviz/Orbit', could not be loaded.
Error:
Could not find library corresponding to plugin rviz/Orbit. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
```

Running ubuntu 16.04 with ros kinetic.

wjwwood, 2016-04-20:
What version of rviz are you using? Did you install from apt-get or build from source?

jgoppert, 2016-04-20:
I used apt-get.

jgoppert, 2016-04-20:
Trying source now.

jgoppert, 2016-04-20:
Same result, I also uninstalled apt-get rviz before I ran source build rviz  to make sure I was using the correct version.

```bash
~/git/catkin$ rosrun rviz rviz 
[ INFO] [1461190043.941959764]: rviz version 1.12.0
[ INFO] [1461190043.941990010]: compiled against Qt version 5.5.1
[ INFO] [1461190043.941996745]: compiled against OGRE version 1.9.0 (Ghadamon)
[ INFO] [1461190044.336475866]: Stereo is NOT SUPPORTED
[ INFO] [1461190044.336622441]: OpenGl version: 4.5 (GLSL 4.5).
[ERROR] [1461190044.378611482]: PluginlibFactory: The plugin for class 'rviz/Orbit' failed to load.  Error: Could not find library corresponding to plugin rviz/Orbit. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[ERROR] [1461190047.440806959]: PluginlibFactory: The plugin for class 'rviz/MoveCamera' failed to load.  Error: Could not find library corresponding to plugin rviz/MoveCamera. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
```

wjwwood, 2016-04-20:
Looks like I forgot to update the plugin description when I changed the name of the default plugin library, can you try: https://github.com/ros-visualization/rviz/pull/1004

You can also just apply the diff of the `plugin_description.xml` directly to try it out without building from source.

jgoppert, 2016-04-20:
Can confirm this fixes it.

closed 2016-04-20 by jgoppert