ros-visualization/rviz #1025: Cannot set float parameters in RViz if locale decimal separator is not dot.

VictorLamoine, 2016-06-10:
I'm using the French locale and our decimal mark is the comma, not the dot. ([like many other countries](https://upload.wikimedia.org/wikipedia/commons/thumb/a/a8/DecimalSeparator.svg/1024px-DecimalSeparator.svg.png?1465589334705))

In RViz when trying to set the near clip distance in the Views panel; I cannot enter values between 0 and 1. (eg: `0,56` becomes `56`).

This can be easily debugged by adding a `cout` in the code ([here](https://github.com/ros-visualization/rviz/blob/kinetic-devel/src/rviz/view_controller.cpp#L253-L254)):
```
  std::cout << "Near clip distance = " << n << std::endl;
```
Install the french locale and switch to it:
```
export LANG=fr_FR.UTF-8
```

The issue is solved on my machine when using `export LANG=C` before running RViz in the terminal.

wjwwood, 2016-06-10:
@VictorLamoine as always, these things tend to favor the environment in which they were originally developed. I'm not familiar with the locale mechanism used to handle this difference, but I'm happy to review pull requests that improve the portability of how we interpret decimals.

wjwwood, 2016-10-18:
I think this should be resolved in https://github.com/ros-visualization/rviz/pull/1043. I tried rviz by setting my locale to fr_FR and I had to use `,` as the decimal delimiter.

closed 2016-10-18 by wjwwood