ros-visualization/rviz #1026: Segfault on launch with Debian build of Rviz

cdellin, 2016-06-16:
(Note: I'm not sure where to file this bug ... perhaps there's a place that's appropriate for packaging bugs?)

I installed the `ros-kinetic-rviz` package, and I get the following segfault:

```
$ gdb --args /opt/ros/kinetic/lib/rviz/rviz
...
Program received signal SIGSEGV, Segmentation fault.
0x00007fffd9608dbc in ?? () from /usr/lib/x86_64-linux-gnu/libQtTest.so.4
(gdb) where
#0  0x00007fffd9608dbc in ?? () from /usr/lib/x86_64-linux-gnu/libQtTest.so.4
#1  0x00007ffff7dea9ba in call_init (l=<optimized out>, argc=argc@entry=1, 
    argv=argv@entry=0x7fffffffe128, env=env@entry=0x7fffffffe138)
    at dl-init.c:78
#2  0x00007ffff7deaaa3 in call_init (env=0x7fffffffe138, argv=0x7fffffffe128, 
    argc=1, l=<optimized out>) at dl-init.c:36
#3  _dl_init (main_map=0x7ffff7ffe1a8, argc=1, argv=0x7fffffffe128, 
    env=0x7fffffffe138) at dl-init.c:126
#4  0x00007ffff7ddd1ca in _dl_start_user () from /lib64/ld-linux-x86-64.so.2
#5  0x0000000000000001 in ?? ()
#6  0x00007fffffffe41a in ?? ()
#7  0x0000000000000000 in ?? ()
```

It looks like it could be a Qt-related problem.  Here are the libraries the linker finds:
```
$ ldd /opt/ros/kinetic/lib/rviz/rviz | grep Qt
	libQt5Widgets.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5 (0x00007f15d0efd000)
	libQt5Gui.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5 (0x00007f15d07c7000)
	libQt5Core.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Core.so.5 (0x00007f15d008b000)
	libQtGui.so.4 => /usr/lib/x86_64-linux-gnu/libQtGui.so.4 (0x00007f15b4248000)
	libQtTest.so.4 => /usr/lib/x86_64-linux-gnu/libQtTest.so.4 (0x00007f15b4022000)
	libQtCore.so.4 => /usr/lib/x86_64-linux-gnu/libQtCore.so.4 (0x00007f15b3b33000)
```

... so I expect there's a problem with Rviz being linked against both Qt4 and Qt5?

I'm running an freshly-installed and up-to-date Debian 8 Jessie system.  I followed the ROS installation instructions here: http://wiki.ros.org/action/fullsearch/Installation/Debian

wjwwood, 2016-06-16:
> ... so I expect there's a problem with Rviz being linked against both Qt4 and Qt5?

I think that's right. Some dependency of rviz's must be pulling in Qt4. What version of rviz is it?

cdellin, 2016-06-16:
Here's the version information:

```
$ cat /etc/apt/sources.list.d/ros-latest.list 
deb http://packages.ros.org/ros/ubuntu jessie main
$ grep version /opt/ros/kinetic/share/rviz/package.xml
  <version>1.12.1</version>
$ dpkg-query --show ros-kinetic-rviz
ros-kinetic-rviz	1.12.1-0jessie-20160610-152205-0700
$ apt-cache depends ros-kinetic-rviz | grep qt
  Depends: libqt5core5a
  Depends: libqt5gui5
  Depends: libqt5widgets5
  Depends: libqt5opengl5
  Depends: ros-kinetic-python-qt-binding
```

I'm not sure what could be pulling in Qt4 ... any ideas?

cdellin, 2016-06-16:
Smoking gun?

```
$ apt-cache rdepends libqtcore4 | grep ros-kinetic
  ros-kinetic-rtabmap-ros
  ros-kinetic-rtabmap
  ros-kinetic-opencv3
  ros-kinetic-octovis
```

wjwwood, 2016-06-16:
The latest version of rviz (which it appears you have) does not depend on anyone of those packages. Maybe you have an rviz plugin that does though? But I wouldn't have pinned it on that since your rviz executable is linked against qt4 directly.

cdellin, 2016-06-16:
(to clarify my above comment:)

```
cdellin@xxx:~$ apt-cache depends ros-kinetic-rviz
  ...
  Depends: libqt5core5a
  Depends: libqt5gui5
  Depends: libqt5widgets5
  Depends: libqt5opengl5
  Depends: ros-kinetic-opencv3
  ...
cdellin@xxx:~$ apt-cache depends ros-kinetic-opencv3
  ...
  Depends: libqt4-test
  Depends: libqtcore4
  Depends: libqtgui4
  Depends: libvtk5-qt4-dev
  ...
```

wjwwood, 2016-06-16:
Ah, I removed that dependency on OpenCV here:

https://github.com/ros-visualization/rviz/pull/1009

But I haven't released that yet. Sorry. I'll try to get a release out this week.

cdellin, 2016-06-16:
Ah, excellent! Thanks.

SquirrelCZE, 2016-09-19:
Was this solved? 

I can't run actual rviz on actual debian jessie...

```
$ rosrun --debug rviz rviz
[rosrun] Looking in catkin libexec dirs: /opt/ros/kinetic/lib/rviz
[rosrun] Looking in rospack dir: /opt/ros/kinetic/share/rviz
[rosrun] Searching for rviz with permissions /111
[rosrun] Running  /opt/ros/kinetic/lib/rviz/rviz
Segmentation fault

Program received signal SIGSEGV, Segmentation fault.
0x00007fffd9601dbc in ?? () from /usr/lib/x86_64-linux-gnu/libQtTest.so.4
(gdb) where
#0  0x00007fffd9601dbc in ?? () from /usr/lib/x86_64-linux-gnu/libQtTest.so.4
#1  0x00007ffff7dea9ba in call_init (l=<optimized out>, argc=argc@entry=1, argv=argv@entry=0x7fffffffe388, 
    env=env@entry=0x7fffffffe398) at dl-init.c:78
#2  0x00007ffff7deaaa3 in call_init (env=0x7fffffffe398, argv=0x7fffffffe388, argc=1, l=<optimized out>)
    at dl-init.c:36
#3  _dl_init (main_map=0x7ffff7ffe1a8, argc=1, argv=0x7fffffffe388, env=0x7fffffffe398) at dl-init.c:126
#4  0x00007ffff7ddd1ca in _dl_start_user () from /lib64/ld-linux-x86-64.so.2
#5  0x0000000000000001 in ?? ()
#6  0x00007fffffffe5f2 in ?? ()
#7  0x0000000000000000 in ?? ()

$ ldd /opt/ros/kinetic/lib/rviz/rviz | grep Qt
        libQt5Widgets.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5 (0x00007f3741843000)
        libQt5Gui.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5 (0x00007f374110d000)
        libQt5Core.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Core.so.5 (0x00007f37409d1000)
        libQtGui.so.4 => /usr/lib/x86_64-linux-gnu/libQtGui.so.4 (0x00007f3724b87000)
        libQtTest.so.4 => /usr/lib/x86_64-linux-gnu/libQtTest.so.4 (0x00007f3724961000)
        libQtCore.so.4 => /usr/lib/x86_64-linux-gnu/libQtCore.so.4 (0x00007f3724472000)

$ dpkg-query --show ros-kinetic-rviz
ros-kinetic-rviz        1.12.1-0jessie-20160724-122351-0700

$ apt-cache depends ros-kinetic-opencv3 | grep qt4
  Depends: libqt4-test
  Depends: libvtk5-qt4-dev

```

wjwwood, 2016-09-19:
It hasn't been released yet. It will probably be after ROSCon before I'll have time to catch up on rviz and make a new release.

mryellow, 2016-09-19:
Work-around: `catkin config --append-args --cmake-args -DWITH_QT=4 -DUseQt5=OFF`

pcarleton, 2016-10-06:
@mryellow Thanks for that workaround.  For it to work, do I need to rebuild rviz using catkin? If so, are there instructions somewhere on how to do that? The build instructions on http://wiki.ros.org/rviz/UserGuide#Install_or_build_rviz give rosmake commands which do not take in to account the catkin config.

mryellow, 2016-10-06:
@pcarleton Yeah that's with a source install where you can force everything (other packages using those flags) to use QT4.

Here is a semi-unattended install script I use (on Debian), along with the ROS source docs it might help show how the whole process works. Manually you don't need to specify dirs and everything like that script does as you're running from those paths already. Note the overlay workspace is a specific robot, replace packages in `assets/rosinstall/kinetic-kulbu.rosinstall` with whatever you want in your workspace. RVIZ is installed to root workspace, specified in `assets/rosinstall/kinetic-desktop-custom.rosinstall` along with a bunch of other deps.
 
https://github.com/mryellow/debian-scripts/blob/master/ros-kinetic.sh

http://wiki.ros.org/kinetic/Installation/Source  
http://wiki.ros.org/catkin/Tutorials/workspace_overlaying

mryellow, 2016-10-07:
@pcarleton Oh I probably have a fork or two coded into those dependencies in `root_ws`. There were a few issues with packages half-way through transitioning to new versions of libraries. Those issues are all closed now and anything you see with `mryellow` as the user can be replaced with the official repos again.

Best bet is probably using the generator to create your own desktop rosinstall file.

jdbug1, 2016-10-18:
@wjwwood Can you tell us when the release will be done? I did a fresh Debian 8.6 and ROS installation (from the official repositories, not from source) today and it's still not working

wjwwood, 2016-10-18:
No, but I was working on it yesterday and I'll probably work on it today too. By "official repositories" you mean packages.ros.org or the official Debian repositories?

jdbug1, 2016-10-18:
I installed following the instructions on http://wiki.ros.org/kinetic/Installation/Debian, so the official Debian repositories. I installed the ros-kinetic-desktop-full package.

wjwwood, 2016-10-18:
@jdbug1 no, actually that wiki uses the packages.ros.org repository. I think we agree, but I would consider the "official Debian repositories" to be packages.debian.org. If you installed `ros-kinetic-desktop-full` then you are using the ROS repositories, i.e. packages.ros.org.

wjwwood, 2017-05-02:
Closing since the released version of rviz no longer has the OpenCV dependency.

closed 2017-05-02 by wjwwood